WouterJD/FortiusANT #2: Python dependencies

mtbiker22, 2020-01-17:
I noticed your project on the antifier GitHub page... I am very eager to get your solution working.

When trying to run your software on the same software as I was using antifier (Python 2.7) in run into the following error

`  File "FortiusAnt.py", line 865
    clv.print()
            ^
SyntaxError: invalid syntax`

Right now I am trying to satisfy requirements in Python 3.5 but run into problems with wxPython 

`Error running configure
    ERROR: failed building wxWidgets
    Traceback (most recent call last):
      File "build.py", line 1468, in cmd_build_wx
        wxbuild.main(wxDir(), build_options)
      File "/tmp/pip-build-mmr8jqj1/wxPython/buildtools/build_wxwidgets.py", line 372, in main
        "Error running configure")
      File "/tmp/pip-build-mmr8jqj1/wxPython/buildtools/build_wxwidgets.py", line 85, in exitIfError
        raise builder.BuildError(msg)
    buildtools.builder.BuildError: Error running configure`

and Pyinput

`Invalid requirement: 'python-xlib>=0.17; linux in "sys_platform"'
Traceback (most recent call last):
`

I will look into this in more detail, guess the error is more related to my local installation.

mtbiker22, 2020-01-17:
Pyinput solved, now working on WxPython

WouterJD, 2020-01-17:
FortiusANT is deveoped with python 3, which might explain the clv.print() syntax error.
FortiusANT is my learning-project for python and I have no prior experience with earlier releases or projects.

You might start off testing individual modules; debug.py, logfile.py and FortiusAntCommand.py are indepentantly runnable to test the module itself.

Succes!

mtbiker22, 2020-01-17:
I am getting close.. have to install some GTK+ libraries to get WxPython running...

WouterJD, 2020-01-17:
Note that FortiusAntGui.py can also be run independently to test the module

mtbiker22, 2020-01-18:
After a night of compiling (WxPython) all dependencies are met... It was complaining about serial so I also installed pyserial. Without the ANT+ stick the GUI boots (and mentions it cannot find the ANT+ device). With the ANT+ stick installed booting stalls at "-----Linux not tested------", no GUI shows up.

When I run the program without the ANT+ stick the GUI boots, upon inserting the ANT stick and pushing Locate HW I get the error "read failed: device reports readiness to read but returned no data (device disconnect or multiple access on port?"..

WouterJD, 2020-01-18:
You're quite close.
Now check the Ubuntu specifics to get it to work.
I have no Ubuntu, so cannot provide info. Sorry.

WouterJD, 2020-01-18:
Try to use -d 127 to create logfile

mtbiker22, 2020-01-18:
`11:14:05,276: FortiusANT started
11:14:05,276: -a
11:14:05,277: -b 2.096,46/34,32/11
11:14:05,277: -d 127 (0b1111111)
11:14:05,277: ------------------
11:14:05,278: Scan for hardware
11:14:05,278: GetDongle()
***** Linux not tested *****
11:14:05,282: ResetDongle()
11:14:05,282: msg4A_ResetSystem()
11:14:05,283: SendToDongle(["a4 01 4a 00 ef"],(Reset dongle))
11:14:05,283: Dongle    send   : synch=164, len= 1, id=74, check=239, info="00" + (Reset dongle)
11:14:05,284: ReadFromDongle()
11:14:05,384: Dongle    drop   : synch=164, len= 1, id=111, check=234, info="20" + `

WouterJD, 2020-01-18:
So ANT+ dongle works.
Will check in detail later

WouterJD, 2020-01-18:
"Also logfile.py gives me errors:"; this is an error in the test-route and not impacting operation. New file uploaded.

WouterJD, 2020-01-18:
Regarding the -d flag:
Application = 0x01
Function = 0x02
antDongle = 0x04
usbTrainer = 0x08

So if you want to trace the antDongle only -d4 would be enough.

It would be helpful to send the logfile in the "read failed: device reports readiness to read but returned no data (device disconnect or multiple access on port?" situation.
Where is that message displayed?

mtbiker22, 2020-01-18:
With dongle after ctrl-c

sudo python3 FortiusAnt.py -a -b=2.096,46/34,32/11 -d 6
14:31:32,392: FortiusANT started
14:31:32,393: -a
14:31:32,393: -b 2.096,46/34,32/11
14:31:32,393: -d 6 (0b110)
14:31:32,393: ------------------
14:31:32,394: GetDongle()
***** Linux not tested *****
14:31:32,397: ResetDongle()
14:31:32,397: msg4A_ResetSystem()
14:31:32,398: SendToDongle(["a4 01 4a 00 ef"],(Reset dongle))
14:31:32,398: Dongle    send   : synch=164, len= 1, id=74, check=239, info="00" + (Reset dongle)
14:31:32,399: ReadFromDongle()
14:31:32,500: Dongle    drop   : synch=164, len= 1, id=111, check=234, info="20" + 
^CTraceback (most recent call last):
  File "FortiusAnt.py", line 885, in <module>
    if LocateHW(x):
  File "FortiusAnt.py", line 224, in LocateHW
    devAntDongle, msg = ant.GetDongle()
  File "/home/bart/Downloads/FortiusANT-master/antDongle.py", line 224, in GetDongle
    reply = ResetDongle(devAntDongle)
  File "/home/bart/Downloads/FortiusANT-master/antDongle.py", line 141, in ResetDongle
    return SendToDongle(messages, devAntDongle, "(Reset dongle)")
  File "/home/bart/Downloads/FortiusANT-master/antDongle.py", line 295, in SendToDongle
    data = ReadFromDongle(devAntDongle, drop)
  File "/home/bart/Downloads/FortiusANT-master/antDongle.py", line 328, in ReadFromDongle
    trv = devAntDongle.read(size=256)
  File "/usr/local/lib/python3.5/dist-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt

mtbiker22, 2020-01-18:
This is with starting the GUI without ANT stick inserted (Motorbrake not powered)

14:38:28,016: FortiusANT started
14:38:28,016: -a
14:38:28,017: -b 2.096,46/34,32/11
14:38:28,017: -d 127 (0b1111111)
14:38:28,017: -g
14:38:28,018: ------------------
14:38:30,148: SetValues() done in 528 ms
14:38:30,150: Scan for hardware
14:38:30,151: GetDongle()
14:38:30,154: GetDongle() returns: 
14:38:30,155: Get USB trainer
14:38:30,155: GetTrainer()
14:38:30,238: GetTrainer - Trainer found
14:38:30,286: GetTrainer() returns, trainertype=0x1932
14:38:30,287: InitialiseTrainer data="02 00 00 00" (len=4)
14:38:30,291: Scan for hardware - end
14:38:30,292: ReceiveFromTrainer()
14:38:30,296: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00 03 13 02 00
 00 00 00 00 00 00 b8 61 00 00 00 00 10 04 00 00 00 00 02 a8 2a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00" (len=6
4)
14:38:30,297: ReceiveFromTrainer: TargetResistance=1040 hr=0 sp=0 CurrentResistance=0 pe=0x0 cad=0 axis=0 2573 0 0
14:38:30,297: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:30,298: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:30,298:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:30,876: ReceiveFromTrainer()
14:38:30,879: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:30,879: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:30,880: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:30,880:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:31,127: ReceiveFromTrainer()
14:38:31,130: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:31,130: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:31,131: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:31,131:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:31,377: ReceiveFromTrainer()
14:38:31,380: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:31,380: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:31,381: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:31,381:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:31,628: ReceiveFromTrainer()
14:38:31,631: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:31,631: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:31,632: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:31,632:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:31,879: ReceiveFromTrainer()
14:38:31,882: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:31,882: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:31,883: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:31,883:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:32,130: ReceiveFromTrainer()
14:38:32,133: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:32,133: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:32,134: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:32,134:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:32,381: ReceiveFromTrainer()
14:38:32,384: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:32,384: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:32,385: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:32,385:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:32,632: ReceiveFromTrainer()
14:38:32,635: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:32,635: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:32,636: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:32,636:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:32,883: ReceiveFromTrainer()
14:38:32,886: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:32,886: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:32,887: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
[... 274 lines cut ...]
14:38:46,687: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:46,687:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:46,934: ReceiveFromTrainer()
14:38:46,937: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
14:38:46,937: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:46,938: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:46,938:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:47,185: ReceiveFromTrainer()
14:38:47,187: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)
bart@Atom ~/Downloads/FortiusANT-master $ tail FortiusANT.2020-01-18\ 14-38-28.log 
14:38:46,686: SendToTrainer(1, 100, False, -1, 0, 0, 0)
14:38:46,687: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:46,687:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:46,934: ReceiveFromTrainer()
[... cut ...]
14:38:46,938: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
14:38:46,938:                   target=1040 pe=0 mode=2 weight=10 cal=1040
14:38:47,185: ReceiveFromTrainer()
14:38:47,187: Trainer recv data="12 99 12 00 00 02 00 00 0a 00 00 00 00 00 00 00 00 00 0d 0a 00 00 00 00" (len=24)

Upon the ant dongle and pushing the locate hardware button I get the .. in the terminal:
"read failed: device reports readiness to read but returned no data (device disconnect or multiple access on port?"

This evening I have more time to look into this in detail.

WouterJD, 2020-01-18:
I'm not at home now, will check Tonite.
Good info!

closed 2020-01-24 by WouterJD