WouterJD/FortiusANT #3: Linux Ant+ Serial

mtbiker22, 2020-01-18:
I have trying to run FortiusANT on Linux. After satisfying all python dependencies (Python 3.5) I run into problems with the ANT+ stick. I have been debugging the code a bit and it seems the code gets stuck at reply = ResetDongle(devAntDongle).

I think the code ends up in an infinite loop in the ReadFromDongle subroutine. Once trv=b'\xa4\x01o \xea', afterwards it is empty each time. Probably some difference in the serial behavior in Windows and Linux?

WouterJD, 2020-01-18:
Like in the other thread, the reset system message is sent to the dongle ("a4 01 4a 00 ef")
then a message is received (164,1,111,234 = 0xa4 0x01 0xea)
and then nothing.

In the other thread you interrupted causing a program abort.

"Probably some difference in the serial behavior in Windows and Linux?" is probably right.

Please look at lines 324... in module antDongle.py
>             if platform.system() == 'Linux':
>                 # ***** Linux not tested *****
>                 devAntDongle.timeout = 0.1
>                 try:
>                     trv = devAntDongle.read(size=256)
>                 except Exception as e:
>                     trv = ""
>                     logfile.Write ( str(e) )
>             else:
>                 trv = devAntDongle.read(0x81,1000,20)   # input:   endpoint address, length, timeout
>                                                         # returns: an array of bytes

which is as antifier coded it.
The comment in the non-Linux part explains the input-parameters and as a first attempt I would try that for Linux as well. (In fact I would hope .read() would be equal over the platforms).

Easiest  is to change 'Linux' into '***Linux***' (which is never true) and hence the windows-branche is executed.

I assume you can modify the antDongle.py module, right?
Please let me know!

WouterJD, 2020-01-18:
Sorry that I do not know exactly how it all works (being new @ python) 

I use 'import usb.core' and the dongle is found with
devAntDongle = usb.core.find(idVendor=0x0fcf, idProduct=ant_pid)

If I read pyusb documentation at
https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst

it states it should be platform independant, so the "if Linux" should not be necessary.

Looking down that page (dev.read(0x81, len(msg), 100)) I see another coding than what is used for Linux.

Based upon that info, I would remove the linux-branch at all; I'm looking forward for your analysis.

mtbiker22, 2020-01-18:
Thanks, will look into this tomorrow, thank you for the support, I will keep you updated!

mtbiker22, 2020-01-19:
I basically removed all Linux specific code parts and usbserial.ko and usb-serial-simple.ko from the kernel drivers (to prevent blocking of the USB ANT+ device) and now the script works.

I have some problems with the GUI. A couple of times it just crashes after starting..... Now I can get into the GUI, but when I push on "run down" or "start" its quits... segmentation fault. Will have to look into debugging/logging to provide you with some additional information.

WouterJD, 2020-01-19:
Wauw! So no changes to the python code untill now?

In fact you do not need the GUI, start FortiusANT without the -g flag.

mtbiker22, 2020-01-19:
I kept

`if platform.system() == 'Linux':
    import serial`

I removed all Linux specific parts throughout antDongle.py.

Yes I know that I do not need the GUI. I was just wondering which additional Information the GUI provides.

WouterJD, 2020-01-19:
Could you please upload the current antDongle.py?

GUI? Antifier had a GUI which is quite simple.
FortiusANT is my learn-python-on-the-job project and I made something visual.

I start FortiusANT with -g -a (automatic start) and expect to use without GUI as soon as it is really operational.

WouterJD, 2020-01-20:
You keep
> if platform.system() == 'Linux': import serial
but is that necessary? Serial seems not referenced.

WouterJD, 2020-01-21:
Hi how are you doing with the Fortius? Check the wiki; it's about finished now.
Curious to learn from your experiences.

mtbiker22, 2020-01-21:
I have been a bit busy, I will upload the code ASAP. I am planning to use the tacx tomorrow evening.

WouterJD, 2020-01-21:
Echt top! cu on Strava!

mtbiker22, 2020-01-21:
Hi Wouter, I am not very familiar with GitHub, can you please provide me with a short description how to upload te file (and not ruin the structure of your project)? Thanks

WouterJD, 2020-01-21:
I think you can attach to a comment. ... 
From iPhone so don't see link here

mtbiker22, 2020-01-21:
Actually I think it is more difficult.... probably have to create a pull request.

WouterJD, 2020-01-21:
Below the text I type (now) I see the box: "Attach files by dragging & dropping, selecting or pasting them.". If I hoover the mouse over that text it allows to select files.

I dropped the requirements.txt there which causes the follow link to be createdm for the file being uploaded...

[requirements.txt](https://github.com/WouterJD/FortiusANT/files/4093833/requirements.txt)

mtbiker22, 2020-01-22:
Ok, I changed the extension to txt, otherwise GitHub would not allow me to upload...
[antDongle.txt](https://github.com/WouterJD/FortiusANT/files/4099788/antDongle.txt)

mtbiker22, 2020-01-22:
Just to be clear, I do not really have a lot of experience with Tacx, neither with Zwift. I primarily decided to buy the Fortius because I stumbled upon antifier and through antifier upon your solution.

I used the original Tacx software (3.5) once and did a video bike ride (Amstel Gold). During that ride I really noticed that the wheel speed is similar to the virtual speed. When I climbed a mountain, I really had to change gears to cope with the increased resistance.... downhill I really could stop pedalling because the rear wheel was driven bij the Fortius motor. I just finished my first real Zwift ride and noticed that it basically measures my power delivered and depending on the slope it changes my virtual bike speed (which really does not correspond with my real wheel speed). I have to read in detail through your wiki, but is this the behaviour you suspect? Apparently my Zwift trial period is ending... I think I will extend to continue optimising your solution....

WouterJD, 2020-01-22:
Thanks, let's close this issue "Linux Ant+ serial" since that one is solved

closed 2020-01-22 by WouterJD

WouterJD, 2020-01-22:
> Ok, I changed the extension to txt, otherwise GitHub would not allow me to upload...
> [antDongle.txt](https://github.com/WouterJD/FortiusANT/files/4099788/antDongle.txt)

As expected, I will update my code accordingly :-)

mtbiker22, 2020-01-25:
To ensure access to the usb device I followed these instructions from the ant bridge project, I propose to add these to your Linux instructions:

Remove usb-serial-simple.ko and usbserial.ko. These drivers will grab the dynastream usb device denying access to it from our application. LOC=/lib/modules/uname -r/kernel/drivers/usb/serial/ sudo mv $LOC/usb-serial-simple.ko ~/Documents sudo mv $LOC/usbserial.ko ~/Documents You may also need to rmmod the modules from running. sudo rmmod usb_serial_simple usbserial

WouterJD, 2020-01-25:
Made a reference

mattipee, 2020-04-21:
Just because this issue exists, I can confirm I get various modes of crash running the GUI on Linux. I think this is to do with GUI interaction on another thread. Very similar to https://stackoverflow.com/questions/14320836/wxpython-pango-error-when-using-a-while-true-loop-in-a-thread

I have tried guarding calls to SetValues() and SetMessages() in FortiusAnt.py with wx.CallAfter(...), as per the link above, but either I've missed one or more, or this doesn't solve anything.

Again, just adding this here because GUI crash was mentioned above. Possibly open a new issue for this, if I (or anyone else) can get it fixed. I don't want to run headless, because I would like to extend the GUI, and want to ditch Windows.

WouterJD, 2020-04-21:
Thanks for checking

I'm 100% caught up with the i-Vortex integration. So will postpone all other suggestions till after