WouterJD/FortiusANT #4: More information on FortiusANT's powercurve

WouterJD, 2020-01-19:
Let's talk about the basics first:

Resistance
------------
It is important to understand that a trainer (FE=Fitness Equipment) like Tacx Fortius only knows 'resistance' which is the torque to turn the braking axle.
From physics we know that power (Watt) = torque (Nm) * speed (km/hr).
So, for a given resistance, the power required is lineair with the speed (=cadence), provided you do not change gears.
Also, for a given resistance and cadence, the power required is lineair with the gear-ratio.

Power- or Ergo-mode
------------------------
In Powermode the CTP (Cycling Training Program) sends the required power to the FE and regardless gear-ratio or cadence, the requested power is constant.
FortiusANT calculates the required resistance = power/speed (with some constants applied).
Note that, if you change gears and/or cadence, the required power will remain equal because the resistance is adjusted.

Resistance mode
-------------------
In resistance mode, FortuisANT receives the required grade from CTP. Grade may vary from -20% (downhill) to +20% (uphill).
FortiusANT must calculate the resistance to be sent to the FE.

This issue describes how this works

WouterJD, 2020-01-19:
FortiusANT uses a power curve that works with two reference points:
- When riding +20% (uphill=heavy), we assume a cadence of 50/minute and an applied power of 1.5 fte.
- When riding -20% (downhill = light), we assume a cadence of 100/minute and an applied power of 1 fte.
During a resistance mode exercise (e.g. using Zwift); the resistance is calculated lineairly between the two reference points which are expressed as Resistance (not power).

The reference points are calculated as follows:
- ReferenceUp = PowerUp / SpeedUp = 1.5 ftp / ( CadenceUp * Tyre * FrontS / CassetteL ) * constant
- ReferenceDown = PowerDown / SpeedDown = 1.5 ftp / ( CadenceDown * Tyre * FrontL / CassetteS ) * constant
Where Tyre is the circumference of the bicycle-wheel, FrontS is the small chainring and CassetteL is the large cassette sprocket.

WouterJD, 2020-01-19:
To be able to calculate the powercurve the following inputs are required:

- ftp; the functional threshold power of the rider
- the tyre circumference
- the largest and smalles chainring sizes
- the largest and smallest cassette sprockets
- the power when riding uphill and downhill

WouterJD, 2020-01-19:
Let's start with a simple example: -b 2.096,50/50,15/15 -f 200 -r 150/100
Which means we are riding a single-speed bicycle

When riding uphill (+20%) at 50rpm we have to produce 200Watt and
when riding downhill (-20%) at 100rpm we have to produce 100Watt.

As explained, the Resistance is set as interpolation between those points.
The Resistance is fixed and hence the power varies with the cadence.

Note that, if your bicycle is not a fixy really but has a 11-35 cassette, you can also vary the required power by changing gears.

WouterJD, 2020-01-19:
You could also define: -b 2.096,50/34,15/25 -f 200 -r 150/100, having a compact with a real cassette.

Required powers and resistance are equal to the example above, but now gear-changing is **required** to be able to achieve the required targets.

WouterJD, 2020-01-19:
My suggestion: experiment with this powercurve (-b -f and -r parameters).
Tip 1: define -b and -f according to reality
Tip 2: start with a "fixy" to see what happens to Zwift without changing gears
Tip 3: experiment with -r to see what happens when the powercurve is tilted
Post your experiences; I'm curious what all happens.

mtbiker22, 2020-01-19:
Interesting stuff... I wonder, how does this relate to the standard Tacx software? What algorithm/calibration do they use to provide realistic resistance values (e.g. in the video training)?

WouterJD, 2020-01-19:
Algorithms for Tacx, TrainerRoad or Zwift software are not published.
For example: if you ride at 234 Watt, how does Zwift calculate your speed?
The USB-interface is not published by Tacx, but https://github.com/totalreverse/ttyT1941 provides 98%.

closed 2020-01-20 by WouterJD