WouterJD/FortiusANT #7: Defining powercurve for Fortius...

WouterJD, 2020-01-22:
Just to be clear, I do not really have a lot of experience with Tacx, neither with Zwift. I primarily decided to buy the Fortius because I stumbled upon antifier and through antifier upon your solution.

I used the original Tacx software (3.5) once and did a video bike ride (Amstel Gold). During that ride I really noticed that the wheel speed is similar to the virtual speed. When I climbed a mountain, I really had to change gears to cope with the increased resistance.... downhill I really could stop pedalling because the rear wheel was driven bij the Fortius motor. I just finished my first real Zwift ride and noticed that it basically measures my power delivered and depending on the slope it changes my virtual bike speed (which really does not correspond with my real wheel speed). I have to read in detail through your wiki, but is this the behaviour you suspect? Apparently my Zwift trial period is ending... I think I will extend to continue optimising your solution....

_Originally posted by @mtbiker22 in https://github.com/WouterJD/FortiusANT/issues/3#issuecomment-577374874_

WouterJD, 2020-01-22:
Hi!
It's always important to tell what -b -f -r parameters are specified and indeed [wiki page](https://github.com/WouterJD/FortiusANT/wiki/The-FortiusANT's-powercurve) should explain in more detail.

If you want "more shifting" uphill adjust -r 150/100 to 250/100 and uphill must get heavier, compared to downhill.

To make downhill even lighter, change -r into 150/-150 and see what happens. I did not try with a negative value and I'm curious to your experiences.

Check "usbTrainer.py" function Grade2Resistance() shows that there has been quite some experimenting to get to an acceptable algorithm. option1 implements the formula from antifier and you could try that by changing the code "option=3" into "option=1". The comment in the code gives some more info.

mtbiker22, 2020-01-22:
Ok, I will have a closer look at it.... at first sight looks very structured and the algorithms are well documented in code!

WouterJD, 2020-01-24:
As you probably know, FortiusANT can be run in manual mode. In that case you do not need Zwift or TrainerRoad, but the required power can be selected from the headunit.

I have implemented the -M flag, for manual grade mode. Now you can test, setting the target grade.

Start FortiusANT as follows:
    FortiusAnt.py -g -a -M -r 150/-50
the target grade can become negative, as well as the target power and you will experience the expected behaviour.

Of course: download the actual python files! (You linux updates have been incorporated)

closed 2020-02-09 by WouterJD