WouterJD/FortiusANT #8: Must remove ANT+ dongle every time program is restarted

Alanimus, 2020-01-26:
Hi, I think this is great and exactly what I was looking for. I am using it to control a servo on a dumb trainer to interface with Zwift so not exactly the intended use, but I noticed one issue, every almost time I completely stop the app to alter the code for instance, I must unplug and reconnect the ANT+ dongle or it can't see the dongles. I have two dongles in as well and neither can be found till I reconnect.

WouterJD, 2020-01-26:
It rarely happens in my environment too, for some reason the dongle itself hangs and needs hard reset.
Do you stop FortiusANT through the interface or by closing the console (kill python). In the latter case no dongle reset is done!

WouterJD, 2020-01-26:
Succes, like to hear more!

Alanimus, 2020-01-26:
This is pretty dumb, but I figured it out. I am using PyCharm for the first time, and it was putting up multiple Python consoles every time I ran the program. If one of these consoles is left open when re-running, it marks the dongle as "in use".

WouterJD, 2020-01-26:
HI Alana, reading your feedback.
When the dongle is "in use" with the "old" console open, the dongle is not freed, it's freed by your OS when the console is closed.

How do you close FortiusANT?
Try to use FortiusANT with the -d4 flag to create an logfile with the ANT-interface.

What OS do you use?

Alanimus, 2020-02-08:
Hi, not sure if I should post this here or in https://github.com/WouterJD/FortiusANT/issues/3

Well I sort of figured out and fixed the replugging thing so I no longer have to replug it every time, I think the program is not shutting down properly. I fixed it by initiating a reset when it times out, and I've progressed passed that to get this working on a Raspberry Pi Zero, thanks to the above closed thread. 

What I am trying to do now is get your code to acknowledge a regular speed+cadence sensor. 
Speed/cadence only displays on data page 0, but since I am new to Ant+ I am unsure how to read this and either bypass the trainer check and put it in the simulated trainer, or add it as a trainer type. I made this crudely to see if I can get it displaying (also added the other bits like ChannelConfig etc.)

```
def msgPage0_CombinedSpeedCadence(Channel, Spec1, Spec2, Spec3, Spec4, BikeSpeedEventTime):

    DataPageNumber = 0
    Spec1 = int(min(0xff, Spec1))
    Spec2 = int(min(0xff, Spec2))
    Spec3 = int(min(0xff, Spec3))
    BikeSpeedEventTime = int(
        min(0xffff, BikeSpeedEventTime * 1000 / 1024))  # Convert seconds into 1024seconds (since ever)
    fChannel = sc.unsigned_char  # First byte of the ANT+ message content
    fDataPageNumber = sc.unsigned_char  # First byte of the ANT+ datapage (payload)
    fSpec1 = sc.unsigned_char
    fSpec2 = sc.unsigned_char
    fSpec3 = sc.unsigned_char
    fBikeSpeedEventTime = sc.unsigned_short

    format = sc.no_alignment + fChannel + fDataPageNumber + fSpec1 + fSpec2 + fSpec3 + fBikeSpeedEventTime
    info = struct.pack(format, Channel, DataPageNumber, Spec1, Spec2, Spec3, BikeSpeedEventTime)

    return info
```

What would I need to do to actually open the channel and read from the new sensor? I was thinking of adding an option in GetTrainer(). However, I'm a bit confused as to what to use for idpl if the sensor is just a generic speed/cadence sensor (I have the sensor ID that Simulant+ picks up, would I use that?)

WouterJD, 2020-02-08:
Hi thanks for posting.
I will check later.

Alanimus, 2020-02-08:
I think I've got it to open a channel, but I am unsure of how to handle the incoming data. Here is the setup:

```
18:35:46,525: msg42_AssignChannel(channel=2, ChannelType=0, NetworkNumber=0)
18:35:46,527: msg51_ChannelID(channel=2,DeviceNumber=23018, DeviceTypeID=121, TransmissionType=0)
18:35:46,530: msg45_ChannelRfFrequency(channel=2, RfFrequency=57)
18:35:46,532: msg43_ChannelPeriod(channel=2, ChannelPeriod=8086)
18:35:46,535: msg60_ChannelTransmitPower(channel=2, TransmitPower=3)
18:35:46,537: msg4B_OpenChannel(channel=2)
18:35:46,540: Dongle    send   : synch=164, len= 3, id=66, check=231, info="02 00 00" + (Speed and Cadence channel config)
18:35:46,545: Dongle    drop   : synch=164, len= 3, id=64, check=167, info="02 42 00" + 
18:35:46,569: Dongle    send   : synch=164, len= 5, id=81, check=56, info="02 ea 59 79 00" + (Speed and Cadence channel config)
18:35:46,573: Dongle    drop   : synch=164, len= 3, id=64, check=180, info="02 51 00" + 
18:35:46,576: Dongle    drop   : synch=164, len= 3, id=64, check=229, info="00 01 03" + 
18:35:46,601: Dongle    send   : synch=164, len= 2, id=69, check=216, info="02 39" + (Speed and Cadence channel config)
18:35:46,605: Dongle    drop   : synch=164, len= 3, id=64, check=160, info="02 45 00" + 
18:35:46,630: Dongle    send   : synch=164, len= 3, id=67, check=111, info="02 96 1f" + (Speed and Cadence channel config)
18:35:46,635: Dongle    drop   : synch=164, len= 3, id=64, check=166, info="02 43 00" + 
18:35:46,661: Dongle    send   : synch=164, len= 2, id=96, check=199, info="02 03" + (Speed and Cadence channel config)
18:35:46,666: Dongle    drop   : synch=164, len= 3, id=64, check=133, info="02 60 00" + 
18:35:46,693: Dongle    send   : synch=164, len= 1, id=75, check=236, info="02" + (Speed and Cadence channel config)
18:35:46,698: Dongle    drop   : synch=164, len= 3, id=64, check=174, info="02 4b 00" + 
```

Then I seem to be managing to receive this when I engage the sensor (I cut out the other debugging info)
`18:35:49,778: Dongle error:Unknown message ID: synch=164, len= 9, id=78, check=145, channel=2, page=156(0x9c) info="02 9c a8 19 00 e6 11 af 05"
18:43:54,618: Dongle error:Unknown message ID: synch=164, len= 9, id=78, check=136, channel=2, page=224(0xe0) info="02 e0 ec 25 00 58 fa e7 05"
18:43:54,622: Dongle error:Unknown message ID: synch=164, len= 9, id=78, check=81, channel=2, page=42(0x2a) info="02 2a fc 26 00 58 fa e7 05"
`

Strange it is sending page 156, 224 and 42 when the speed & cadence device profile says it should only send page 0? I am assuming that is not actually page number data. Thank you so much for any help!

WouterJD, 2020-02-09:
I created new issue #9 to discuss this

closed 2020-02-27 by WouterJD