WouterJD/FortiusANT #9: Use FortiusANT with "Raspberry Pi Zero"

WouterJD, 2020-02-09:
Hi, not sure if I should post this here or in https://github.com/WouterJD/FortiusANT/issues/3

Well I sort of figured out and fixed the replugging thing so I no longer have to replug it every time, I think the program is not shutting down properly. I fixed it by initiating a reset when it times out, and I've progressed passed that to get this working on a Raspberry Pi Zero, thanks to the above closed thread. 

What I am trying to do now is get your code to acknowledge a regular speed+cadence sensor. 
Speed/cadence only displays on data page 0, but since I am new to Ant+ I am unsure how to read this and either bypass the trainer check and put it in the simulated trainer, or add it as a trainer type. I made this crudely to see if I can get it displaying (also added the other bits like ChannelConfig etc.)

```
def msgPage0_CombinedSpeedCadence(Channel, Spec1, Spec2, Spec3, Spec4, BikeSpeedEventTime):

    DataPageNumber = 0
    Spec1 = int(min(0xff, Spec1))
    Spec2 = int(min(0xff, Spec2))
    Spec3 = int(min(0xff, Spec3))
    BikeSpeedEventTime = int(
        min(0xffff, BikeSpeedEventTime * 1000 / 1024))  # Convert seconds into 1024seconds (since ever)
    fChannel = sc.unsigned_char  # First byte of the ANT+ message content
    fDataPageNumber = sc.unsigned_char  # First byte of the ANT+ datapage (payload)
    fSpec1 = sc.unsigned_char
    fSpec2 = sc.unsigned_char
    fSpec3 = sc.unsigned_char
    fBikeSpeedEventTime = sc.unsigned_short

    format = sc.no_alignment + fChannel + fDataPageNumber + fSpec1 + fSpec2 + fSpec3 + fBikeSpeedEventTime
    info = struct.pack(format, Channel, DataPageNumber, Spec1, Spec2, Spec3, BikeSpeedEventTime)

    return info
```

What would I need to do to actually open the channel and read from the new sensor? I was thinking of adding an option in GetTrainer(). However, I'm a bit confused as to what to use for idVendor and idpl if the sensor is just a generic speed/cadence sensor (I have the sensor ID that Simulant+ picks up, would I use that?)

_Originally posted by @Alanimus in https://github.com/WouterJD/FortiusANT/issues/8#issuecomment-583700746_

WouterJD, 2020-02-09:
I try to understand what you want to achieve, but cannot figure it out.
To understand, please explain what a "Raspberry Pi Zero" is.

I create [wiki page Fortius ANT Configurations](https://github.com/WouterJD/FortiusANT/wiki/FortiusANT-Configurations---ANT--Master-and-Slave-devices) and perhaps that gives a hint to draw what configuration you want to achieve.

The 4th configuration (with the independant HRM strap) is an option I'm building these days.
The -H flag makes FortiusANT to listen to an ANT+ master device (a heart rate monitor strap) to pick up the heartrate when the Tacx Fortius does not supply it, in the case you would use a new ANT+ HRM that is not recognized by the Tacx Fortius. This is not essential, but makes the heartbeat display work again.

In case somebody would use a bicicle with a power-meter and cadence sensor, need would arise for a -P and -C flag, so that FortiusANT would pick it up (listen to ANT+ master devices) and send it as one integrated ANT+ FE-C master device to the training software (Trainer Road, Zwift).
This would only be usefull, if you have a trainer brake without power/cadence/speed sensor, integrate them with the independant devices as one ANT+ FE-C device. 

An explanation what you want to achieve would be helpfull; **Adding options to GetTrainer() would NOT be usefull, since that is the USB-interface to Tacx Fortius**

Alanimus, 2020-02-09:
Yes I see now, sorry I was probably asking too many questions at once. I
was really asking how to actually read from the separate speed + cadence
sensors but I managed to figure it out. The code I posted was backwards, I
was mixed up between the "page" and "unpage" functions.

I'm using this code as a base because it already has all the calculations
to convert speed and cadence to simulated power and reads track resistance
from Zwift.

WouterJD, 2020-02-09:
What device do you connect?

WouterJD, 2020-02-09:
Also, are you on Strava? Would like to see FortiusANT results :-)

mtbiker22, 2020-02-09:
Raspberry Pi Zero (5 euro) is a open source single board computer running Linux with a very small form factor.

WouterJD, 2020-02-09:
> Raspberry Pi Zero (5 euro) is a open source single board computer running Linux with a very small form factor.
Sure, but I'm curious how this relates to FortiusANT and/or Controllable Fitness Equipment :-)

mtbiker22, 2020-02-09:
It can be a cheap platform for ‘converting’ the USB to ANT+ connection. It could run headless and be ‘integrated’

Alanimus, 2020-02-09:
Yes exactly. I am just messing around with making a servo work with a
non-connected dumb trainer to integrate with Zwift. I am using a generic
speed + cadence sensor (but using SimulANT+ for testing). Before I began
this I did not realise so many options existed for turning USB trainers
into smart trainers, but it is still fun to work on this. I just had/have a
big learning curve with how Ant+ signals work and the calculations.

I see someone has done something similar with an Arduino and a linear
actuator
<https://www.instructables.com/id/Open-Bicycle-Grade-Simulator-OpenGradeSIM/>
but not on a Pi, though there is a project for integrating HR with fans and
lights <https://github.com/jludwig75/trainer_tools>.

WouterJD, 2020-02-09:
I guessed something like this was going on.

New to ANT+? I hope you are familiar with www.thisisant.com  and
D00001163_-_ANT+_Device_Profile_-_Bicycle_Speed_and_Cadence_2.1.pdf?
It took me a while to get from the antifier source to that documentation, so ask the dumb question.

If you want to use FortiusANT and read data from cadense/speed sensor:
- in antDongle.py add channel_CSC as a unique channel ID
- in FortiusAnt.py Tacx2Dongle()
-- open a slave channel to the sensor
-- expect messages from the sensor (add elif Channel == ant.channel_CSC:) where all incoming messages are handled
-- handle message 0 (default page) with a msgUnpage0_CombinedSpeedCadence(info) function according the information at page 44 returning BikeCadenceEventTime, CumulativeCadenceRevolutionCount, BikeSpeedEventTime, CumulativeSpeedRevolutionCount

I have created an ExplorANT program to check what dongles are in my computer, what devices are on the network and what happens on the devices. I will publish shortly, after doing some final tests.
I expect this to be helpful for you since it provides a template to create slave devices, listening to master devices; where the BikeCadenceSensor is the one to be implemented by you.

Like to hear from you and your achievements.

WouterJD, 2020-02-10:
Alana, new version upload with SCS code example added. Check antDongle.py and FortiusANT.py for "scs". Tested with https://www.strava.com/activities/3088842976

WouterJD, 2020-02-27:
If more info needed, please reopen the issue or create a new one.

closed 2020-02-27 by WouterJD

WouterJD, 2021-02-27:
@Alanimus you will like to see the RPI developments!