WouterJD/FortiusANT #10: Ant+ stick don't detect in Windows OS.

yegorvin, 2020-02-23:
Ant stick don't detect in Windows OS.
Tested in PyCharm and with your compiled exe files.

For example:
`
#
 ------------------------------------------------------------------------------
# C o m p o s e   A N T   M e s s a g e
# ------------------------------------------------------------------------------
def ComposeMessage(id, info):
    fSynch      = sc.unsigned_char
    fLength     = sc.unsigned_char
    fId         = sc.unsigned_char
    fInfo       = str(len(info)) + sc.char_array  # 9 character string

    format  =    sc.no_alignment + fSynch + fLength + fId + fInfo
    data    = struct.pack (format, 0xa4,    len(info), id,  info)
    #---------------------------------------------------------------------------
    # Add the checksum
    # (antifier added \00\00 after each message for unknown reason)
    #---------------------------------------------------------------------------
    data += calc_checksum(data)    
    
    return data`

data    = struct.pack (format, 0xa4,    len(info), id,  info)
**data - string with raw data here**

`#-------------------------------------------------------------------------------
def calc_checksum(message):
    return CalcChecksum (message)           # alias for compatibility
    
def CalcChecksum (message):
    xor_value = 0
    length    = message[1]                  # byte 1; length of info
    length   += 3                           # Add synch, len, id
    for i in range (0, length):             # Process bytes as defined in length
        xor_value = xor_value ^ message[i]

#   print('checksum', logfile.HexSpace(message), xor_value, bytes([xor_value]))

    return bytes([xor_value])`

    length    = message[1]
**lenght  -  char with symbol from index 1 in string variable message**

length    = ord(message[1])
retturn correct int val

How does it work for you??

WouterJD, 2020-02-23:
Please send error-message and logfile (-d flag)
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

yegorvin, 2020-02-23:
1. start your code without changes:
```
FortiusAnt.py -d 255
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\FortiusAnt.py", line 1008
    clv.print()
            ^
SyntaxError: invalid syntax
```

2. rename method **print** in **FortiusAntCommand.py** and all this function calls
-def print(self):
+def printEx(self):
```
FortiusAnt.py -d 255
Traceback (most recent call last):
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\FortiusAnt.py", line 112, in <module>
    import antDongle         as ant
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\antDongle.py", line 28, in <module>
    import structConstants      as sc
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\structConstants.py", line 21
SyntaxError: Non-ASCII character '\xc2' in file D:\Install\Tacx\FortiusANT-master\pythoncode\structConstants.py on line 21, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
```

3. mark file **structConstants.py** as **UTF-8**
`+# -*- coding: utf-8 -*-`
```
FortiusAnt.py -d 255
12:39:07,358: FortiusANT started
12:39:07,358: -d 255 (0b11111111)
12:39:07,359: ------------------
12:39:07,359: Version info for the components
12:39:07,359:         FortiusAnt = 2020-02-10
12:39:07,359:          antDongle = 2020-02-10
12:39:07,359:              debug = 2020-01-25
12:39:07,359:  FortiusAntCommand = 2020-02-09
12:39:07,359:      FortiusAntGui = 2020-02-09
12:39:07,360:            logfile = 2020-02-02
12:39:07,360:    structConstants = 2020-01-25
12:39:07,360:         usbTrainer = 2020-01-25
12:39:07,360:           argparse = 1.1
12:39:07,360:              numpy = 1.16.6
12:39:07,360:             pickle = 2.0
12:39:07,360:           platform = 1.0.7
12:39:07,360:       sys (python) = 2.7.15 (v2.7.15:ca079a3ea3, Apr 30 2018, 16:22:17) [MSC v.1500 32 bit (Intel)]
12:39:07,361:                usb = 1.0.2
12:39:07,361:                 wx = 4.0.7.post2
12:39:07,361: Scan for hardware
12:39:07,361: Get Dongle
12:39:07,361: GetDongle - Check for dongle 4100 Older
12:39:07,372: GetDongle - Check for dongle 4104 Suunto
12:39:07,375: GetDongle - Dongle found
12:39:07,375: GetDongle - Check for dongle 4105 Garmin
12:39:07,378: GetDongle() returns: Could not find ANT-dongle
12:39:07,379: Dongle - Could not find ANT-dongle
12:39:07,379: Get USB trainer
12:39:07,379: GetTrainer()
12:39:07,379: GetTrainer - Check for trainer 0x1932
12:39:07,383: GetTrainer - Check for trainer 0x1942
12:39:07,386: GetTrainer - Check for trainer 0xe6be
12:39:07,389: GetTrainer - No trainer found
12:39:07,389: GetTrainer() returns, trainertype=0x0
12:39:07,390: Tacx   - GetTrainer - No trainer found
12:39:07,391: Scan for hardware - end
```
In debug mode in IDE, I see that the error occurs in the line 130, antDongle.Py
with error message: cannot concatenate 'str' and 'int' objects
See image:
[image]

yegorvin, 2020-02-23:
What version python you use?

WouterJD, 2020-02-23:
21:36:27,967: ------------------
21:36:27,967: Version info for the components
21:36:27,968:         FortiusAnt = 2020-02-12
21:36:27,968:          antDongle = 2020-02-13
21:36:27,969:             antHRM = 2020-02-18
21:36:27,969:              antFE = 2020-02-18
21:36:27,969:              debug = 2020-01-25
21:36:27,970:  FortiusAntCommand = 2020-02-10
21:36:27,970:      FortiusAntGui = 2020-02-12
21:36:27,971:            logfile = 2020-02-12
21:36:27,971:    structConstants = 2020-01-25
21:36:27,971:         usbTrainer = 2020-01-25
21:36:27,972:           argparse = 1.1
21:36:27,972:              numpy = 1.17.4
21:36:27,973:             pickle = 4.0
21:36:27,973:           platform = 1.0.8
21:36:27,973:       sys (python) = 3.8.0 (tags/v3.8.0:fa919fd, Oct 14 2019, 19:21:23) [MSC v.1916 32 bit (Intel)]
21:36:27,974:                usb = 1.0.2
21:36:27,974:                 wx = 4.0.7.post2

WouterJD, 2020-02-23:
def CalcChecksum (message):
    xor_value = 0
    length    = message[1]                  .... second byte in ANT string is length, implicit int assumed
                                                          ... int(message[1]) would make explicit would is meant
                                                          ... definitely a python version issue, perhaps due to incorrect coding?

    length   += 3                           # Add synch, len, id
    for i in range (0, length):             # Process bytes as defined in length
        xor_value = xor_value ^ message[i]

yegorvin, 2020-02-23:
After install python 3.8 it worked.
Specify min python version in the requirements for compiling the project.

I will adapt the program to IMagic 1902.
I will have more questions :)

Thanks!

closed 2020-02-23 by yegorvin

WouterJD, 2020-02-23:
Thinking about this issue
Python 3 is definitely different from python 2 in this area and I have modified the code for python 3, using byte-arrays for data to be sent to ANT or trainer.

Since python 3 a byte is not a one-byte ASCII character, which you already discovered where you added "+# -*- coding: utf-8 -*-" to the code.

I think the better option is to move to python 3, as I see you now did.

WouterJD, 2020-02-23:
Succes!