WouterJD/FortiusANT #11: Tacx IMagic (1902) Support

fritz-hh, 2020-02-23:
Congratulation for your amazing work, and for the amazing work of [totalreverse](https://github.com/totalreverse) in reverse engineering the tacx communication protocol.
It gives me the hope that I may finally be able to get my Tacx Imagic 1902 back in service,

I noticed that `FortiusANT` does not support 1902 yet, and I spent a few hours at trying to understand what it woud mean to support it. I am not yet sure to completely grasp all what it would require, but I would like to explain how far I came within my understanding (I am willing to put some work in this, but I am definitely not a developer...):

Platform: Windows 10
Trainer: Tacx Imagic (1902 headend device)

* Using your wiki, I (successfully) installed `libusb-win32` (using `Zadig`)
* The device is visible in the _device manager_, and reports itself with the expected ID 
[image]
* A learned in the [totalreverse/ttyT1941 github repository wiki](https://github.com/totalreverse/ttyT1941/wiki#tacx-products), that 1902 requires the firmware to be loaded each time the fitness equipment powers-up.
* I managed to rip the iMagic-Firmware using [totalreverse/cutFirmwareIMagic](https://github.com/totalreverse/cutFirmwareIMagic) from the Imagic driver (`I-magic.sys`) located in the original CD
* Then I could install the firmware onto the fitness device using [totalreverse/fxload](https://github.com/totalreverse/fxload) (I did not yet do extensive tests, but I could at least see the Heart rate LED blinking as expected on the fitness device). So, the 1902 requires the same firmware loading procedure as the 1942 each time the connection is initialized: see [here](https://github.com/WouterJD/FortiusANT/blob/442164669c7c6a02057b5da038b256b22f11a23f/pythoncode/usbTrainer.py#L367). The usbTrainer.py file could be extended to make that possible for the 1902 too.
* Next I added the the ID of my fitness device in your [device list](https://github.com/WouterJD/FortiusANT/blob/442164669c7c6a02057b5da038b256b22f11a23f/pythoncode/usbTrainer.py#L335), and ran `python FortiusAnt.py -d 127 -m`. The good news is that it detected the 1902. But then FortiusANT crashed (See output below). Do you have an idea what the reason for this crash could be?

This is how far I could go until now.
I also learned from the _totalreverse/ttyT1941 wiki_ that the 1902 uses another [protocol](https://github.com/totalreverse/ttyT1941/wiki#t1902-usb-legacy-protocol-6-bytes-out-21-bytes-in) than most of the other tacx trainers. So there is clearly some few lines of codes to be written until 1902 can be supported. I will try to see how far I can come with my limited skills in the next weeks.

````
C:\Users\julien\Documents\GitHub\FortiusANT\pythoncode>python FortiusAnt.py -d 127 -m
17:59:45,538: FortiusANT started
17:59:45,538: -d 127 (0b1111111)
17:59:45,540: -m
17:59:45,540: ------------------
17:59:45,540: Version info for the components
17:59:45,540:         FortiusAnt = 2020-02-10
17:59:45,540:          antDongle = 2020-02-10
17:59:45,540:              debug = 2020-01-25
17:59:45,540:  FortiusAntCommand = 2020-02-09
17:59:45,540:      FortiusAntGui = 2020-02-09
17:59:45,540:            logfile = 2020-02-02
17:59:45,540:    structConstants = 2020-01-25
17:59:45,541:         usbTrainer = 2020-01-25
17:59:45,541:           argparse = 1.1
17:59:45,541:              numpy = 1.18.1
17:59:45,541:             pickle = 4.0
17:59:45,541:           platform = 1.0.8
17:59:45,541:       sys (python) = 3.8.1 (tags/v3.8.1:1b293b6, Dec 18 2019, 22:39:24) [MSC v.1916 32 bit (Intel)]
17:59:45,541:                usb = 1.0.2
17:59:45,541:                 wx = 4.0.7.post2
17:59:45,541: Scan for hardware
17:59:45,541: Get Dongle
17:59:45,541: Dongle - Simulated Dongle (manual mode)
17:59:45,542: Get USB trainer
17:59:45,542: GetTrainer()
17:59:45,542: GetTrainer - Check for trainer 0x1902
DEVICE ID 3561:1902 on Bus 000 Address 001 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x100 USB 1.0
 bDeviceClass           :   0xff Vendor-specific
 bDeviceSubClass        :   0xff
 bDeviceProtocol        :   0xff
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x3561
 idProduct              : 0x1902
 bcdDevice              : 0x2001 Device 32.01
 iManufacturer          :    0x0
 iProduct               :    0x0
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0xda (218 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0x80 Bus Powered
[... 210 lines cut ...]
       bmAttributes     :    0x1 Isochronous
       wMaxPacketSize   :   0x10 (16 bytes)
       bInterval        :    0x1
17:59:45,591: GetTrainer - Trainer found: 0x1902
17:59:45,591: GetTrainer() returns, trainertype=0x1902
17:59:45,591: Tacx   - GetTrainer - Trainer found: 0x1902
17:59:45,591: Tacx   - GetTrainer - Trainer found: 0x1902
17:59:45,591: InitialiseTrainer data="02 00 00 00" (len=4)
Traceback (most recent call last):
  File "FortiusAnt.py", line 1059, in <module>
    if LocateHW(x):
  File "FortiusAnt.py", line 259, in LocateHW
    usbTrainer.InitialiseTrainer(devTrainer)     #initialise trainer
  File "C:\Users\julien\Documents\GitHub\FortiusANT\pythoncode\usbTrainer.py", line 408, in InitialiseTrainer
    dev.write(0x02,data)
  File "C:\Users\julien\AppData\Local\Programs\Python\Python38-32\lib\site-packages\usb\core.py", line 943, in write
    return fn(
  File "C:\Users\julien\AppData\Local\Programs\Python\Python38-32\lib\site-packages\usb\backend\libusb0.py", line 529, in bulk_write
    return self.__write(_lib.usb_bulk_write,
  File "C:\Users\julien\AppData\Local\Programs\Python\Python38-32\lib\site-packages\usb\backend\libusb0.py", line 611, in __write
    return int(_check(fn(
  File "C:\Users\julien\AppData\Local\Programs\Python\Python38-32\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: Falscher Parameter.\r\n'
````

yegorvin, 2020-02-23:
After dev.set_configuration() insert:
`dev.set_interface_altsetting(0, 1)`
I also have 1902. Together we will definitely do it.

WouterJD, 2020-02-23:
Great fun to get iMagci work!

WouterJD, 2020-02-23:
What I see on a first glance that the trainer is found
And then in function InitializeTrainer() wants to send four bytes initialization string to dev:
dev.write(0x02, 0x20000000)

The function raises an exception (raise USBError(errmsg, ret))
saying that 'the parameter is incorrect' (Thank you for not telling which parameter is wrong)

I would first check that dev is actually filled (you never know) -- print(dev). I expect that to be OK.

And then check the function write, which is also used in function SendToTrainer(). There the write-function is tried (try/except) and returns a proper error when failing.

It never crashed in my case and I never investigate the interface.

Note the behaviour of getTrainer for the 'unitialized' Fortius;
- first it gets the trainer_type == 0xe6be,
- then it loads the firmware and
- then it does "dev = usb.core.find(idVendor=0x3561, idProduct=0x1942)"

But since you succesfully execute dev.set_configuration() etcetera, this must  have gone well.

So we get to write()
Please read
https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst
... If we know the endpoint address in advance, we could just call the write function from the device object: ...

FortiusAnt blundly writes to endpoint 0x02; which could easily be the invalid parameter
The Fortius device returns.
So I would start checking from that point onwards.
Well got to go now

DEVICE ID 3561:1932 on Bus 000 Address 001 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x100 USB 1.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x3561
 idProduct              : 0x1932
 bcdDevice              :    0x1 Device 0.01
 iManufacturer          :    0x4 Tacx?VR
 iProduct               :    0xe VR-Interface
 iSerialNumber          :    0x0
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 0 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x20 (32 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Vendor Specific ===========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x82: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0
      ENDPOINT 0x2: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x0

WouterJD, 2020-02-23:
PS. jegorvin == fritz-hh ?

yegorvin, 2020-02-23:
We are different people

WouterJD, 2020-02-23:
What a coincidence; you both raise simultaneously an issue to make imagic work😊

yegorvin, 2020-02-23:
I myself am surprised :)

fritz-hh, 2020-02-23:
> After dev.set_configuration() insert:
> `dev.set_interface_altsetting(0, 1)`
> I also have 1902. Together we will definitely do it.

Thanks a lot.
That solved the issue.
I am now able to see read packets and see the message content change when I push a button on the fitness device.

> Note the behaviour of getTrainer for the 'unitialized' Fortius;
> 
> * first it gets the trainer_type == 0xe6be,
> * then it loads the firmware and
> * then it does "dev = usb.core.find(idVendor=0x3561, idProduct=0x1942)"

For the 1902, it is a bit different, as the 1902 already has the right ID before uploading the firmware.
But other than that the FW upload works the same.

I will try to better understand the 1902 specific protocol (and especially how it differs from the fortius) in the next days.

We will make it 😊

yegorvin, 2020-02-23:
The 1902 protocol is much simpler than in 1942.

WouterJD, 2020-02-24:
Hi, does uploading firmware require same code?Did you see that yegorvin is doing same job, coincidentally at the same moment?
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

WouterJD, 2020-02-24:
> The 1902 protocol is much simpler than in 1942. One byte will be enough for us to control the load.
> The main difficulty is to calculate the power of pedaling

Ref TotalReverse, T1902 USB 'Legacy' prodtocol, 6 bytes OUT, 21 bytes IN.
If you have time to wait, I will modify usbTrainer.py to support this. Expect to do this week.

fritz-hh, 2020-02-24:
> Hi, does uploading firmware require same code?

I need to check that in detail.
Yesterday I loaded the FW using the fxload tool directly (not using FortiusANT)
What I can say until now is:
* that 1902 and 1942 use different FW (I could provide the FW for 1902)
* that the[ fxload tool](https://github.com/totalreverse/fxload) is suitable to upload the FW

> Did you see that yegorvin is doing same job, coincidentally at the same moment?

Yes I saw that. That's great.
Together we will get it work.

fritz-hh, 2020-02-24:
> The main difficulty is to calculate the power of pedaling

Yes, I think so too. Unfortunately I do not have a power meter on my bike for the calibration.

> If you have time to wait, I will modify usbTrainer.py to support this. Expect to do this week.

That would be great.

> Hi, does uploading firmware require same code?

I can try to provide you a pull request with the needed change to support both 1902 and 1942 FW upload.

WouterJD, 2020-02-24:
check usbTrainer.py; code adjusted according your info and TotalReverse interface definitions, a step in the right direction (I hope) but to be completed.

I'm curious what the correct values for DesiredForceValue and ResistanceValueOfBrake will be...

yegorvin, 2020-02-24:
os.system("fxload -I tacximagic.hex -d 3561:1902 -t fx")#load firmware 
I Use fxload.exe from libusb-1.0.22 lib
There I can set a vid:pid.

fxload-libusb every time asks me to choose a USB device
```
fxload-libusb.exe -I tacximagic.hex -t fx -vv
0: VendId: 0FCF ProdId: 1008
1: VendId: 3561 ProdId: 1902
Please select device to configure:
```

yegorvin, 2020-02-24:
> Yes, I think so too. Unfortunately I do not have a power meter on my bike for the calibration.
> @yegorvin: Do you have one?
Yes I have a Quarq power meter.

WouterJD, 2020-02-24:
> os.system("fxload -I tacximagic.hex -d 3561:1902 -t fx")#load firmware
> I Use fxload.exe from libusb-1.0.22 lib
> There I can set a vid:pid.
>
> fxload-libusb every time asks me to choose a USB device [...]

What is the exact command in os.system please.
I see three options
```
os.system("fxload -I tacximagic.hex -d 3561:1902 -t fx")
os.system("fxload -I tacximagic.hex -t fx -vv")
os.system("fxload -I tacximagic.hex -d 3561:1902 -t fx -vv")
```

fritz-hh, 2020-02-24:
> check usbTrainer.py; code adjusted according your info and TotalReverse interface definitions, a step in the right direction (I hope) but to be completed.

Looking at it right now.
I will provide feedback later this evening.

fritz-hh, 2020-02-24:
I provided various fixes to the IMagic protocol in #12.
There is still various issues to solve.
Among them:

* Speed reading wrong
* Calibration
* Crash in run off
* And probably more

fritz-hh, 2020-02-24:
By the way: thanks a lot @WouterJD .
It is really incredible how quick you provided the protocol implementation.

WouterJD, 2020-02-24:
Thanks.
Creating structure took time and now pays off😊

fritz-hh, 2020-02-25:
I will continue to work on this during the weekend.
(I am quite busy until the end of the week)

fritz-hh, 2020-02-25:
> Thanks.
> Creating structure took time and now pays off😊

Also, great to see such a well documented source!

WouterJD, 2020-02-25:
> I will continue to work on this during the weekend.
> (I am quite busy until the end of the week)

Thanks and success with the next steps; I will assist where I can.

WouterJD, 2020-02-26:
@fritz-hh ....
I do not like constants in code and therefore defined tt_ variables in usbTrainer.py.
I messed up the names (Fortius, iMagic, iFlow) and got stuck today when using it for the new function CalibrateSupported(). Luckily.

Therefore I went back to the documentation  @totalreverse provided
https://github.com/totalreverse/ttyT1941/wiki#communication-protocols-of-cable-based-trainers

and it contains a phrase which may be very interesting for you:
`You can load the firmware of the "solid blue" device (FortiusSWPID1942Renum.hex) also to the solid green/yellow/white. The solid green head unit 'magically' converts to a 'solid blue' one. This works at least on the devices I own.`

May be interesting for you and @yegorvin as well.
I do not have an iMagic and cannot try, but will upload new code where calibration is interface dependant, I will hear from you what path you go (with/without magic conversion).

yegorvin, 2020-02-27:
Load FortiusSWPID1942Renum.hex firmware.
The device detected as 1942.
Install a driver for it.
Program detected 1942 device.
А green light began to flash rapidly on it.
But when turning the wheel, no parameters are displayed in the program.

From totalreverse wiki:
> BUT: You have to control the brake different, because the PCB layout between the CPU (PortC) and the serial cable is a little different. The green unit uses PINC.7 to controlled the magnetic field, whereas the blue unit uses the serial interface on PINC.0 / PINC.1.

yegorvin, 2020-02-27:
1902 device is implemented in the project GoldenCheetah and well  commented.
See https://github.com/GoldenCheetah/GoldenCheetah/blob/master/src/Train/Imagic.cpp
Method Imagic::run()

```
            //--------------------------------------------------------------
            // HR, Cadence and steering angle are reported as 8-bit integers
            //--------------------------------------------------------------
            curSteering = buf[12];
            curCadence = buf[3];
            curHeartRate = buf[4];

            // ----------------------------------------------------------------
            // Speed seems to be in some mysterious internal units
            // Divide by 11.9 to convert to km/h
            // ----------------------------------------------------------------
            curSpeedInternal = (double)qFromLittleEndian<quint16>(&buf[1]);
            curSpeed = curSpeedInternal / (1.19f * 10.0f);

            // Power
            // Need to calculate this based on resistance setting and speed.
            // From measurements, it seems that Watts = (0.0036 * reported resistance setting + 0.2) * reported speed
            curResistance = buf[9];
            curPower = ((curResistance * 0.0036f) + 0.2f) * curSpeedInternal;
            if (curPower < 0) curPower = 0;

            // average power over last second (ish)
            powertot += curPower;
            powertot -= powerhist[powerindex];
            powerhist[powerindex] = curPower;

            curPower = powertot / 10;
            powerindex = (powerindex == 9) ? 0 : powerindex+1;

            curPower *= powerScaleFactor; // apply scale factor
```

WouterJD, 2020-02-27:
@yegorvin thanks for this excellent reference and @GoldenCheetah thanks for yuor work.
I will implement these formula's in usbTrainer and publish; then it's ready for you to test.
Will try to publish tonite.

yegorvin, 2020-02-27:
Thanks for your great work.
Initially, I wanted to finalize the antifier project, but there everything is too confusing for me.

WouterJD, 2020-02-27:
Well, formula's from Golden Cheetah are implemented and there is a fair chance that this works.
The approach is in line with the "New interface", just some minor changes regarding constants.

Regarding antifier: I tried to get it work (as more people, if I may believe posts on the subject) but do not understand the basic principles of the code. After that I got it working; reverse engineering and removing the table-driven code, (using @TotalReverse interface definitions) I went my way and use FortiusANT to learn ANT and python :-)

Curious whether this works and what errors/typo's will be found; I cannot test because have no iMagic.

mtbiker22, 2020-02-27:
I did not follow the thread in detail... what do we expect for the Tacx iFlow... the control unit is the same as for the fortius, however it employs an electromagnetic brake.

WouterJD, 2020-02-27:
The interface for Fortius and iMagic is different; the command/response buffers are different in structure and length.
The logic is pretty much the same and can therefore he realized with limited effort.

The iFlow might be another one in the family; as soon as somebody has an iFlow, I will assist in getting it work as well.

mtbiker22, 2020-02-27:
I have an iFlow...

WouterJD, 2020-02-27:
Well...connect an check what happens.
As soon as... create a new issue and we will see.
The iFlow is defined in the code...

mtbiker22, 2020-02-27:
Ok, will check... one thing that will not work is the callibration phase in which FortiusANT drives the wheel at 20 km/h, iFlow can not drive the wheel.

WouterJD, 2020-02-27:
The new code supporting iMagic resolves that problem; only Fortius does calibrate

mtbiker22, 2020-02-27:
Ok, I will try in the weekend... hopefully also use the Fortius again. ;-)

fritz-hh, 2020-02-27:
> Well, formula's from Golden Cheetah are implemented and there is a fair chance that this works.
> The approach is in line with the "New interface", just some minor changes regarding constants. [...]

This is really great. I am so impatient to test this during the week-end!

WouterJD, 2020-02-27:
The iFlow is NOT  defined in the code, so small modifications are required at least.

fritz-hh, 2020-02-27:
One small question: 

In the trainer code, you use fxload (https://github.com/WouterJD/FortiusANT/blob/d6f9eb7bc0d0322cb21113d04b2c61fcdce9f589/pythoncode/usbTrainer.py#L434).
This is an external dependency (no python code).
Did you already consider, using https://github.com/fritz-hh/fxload instead.
That is a pure python solution that could be included in FortiusAnt (and would make it simpler to install)

WouterJD, 2020-02-27:
> Did you already consider, using https://github.com/totalreverse/fxload instead.
> That is a pure python solution that could be included in FortiusANT (and would make it simpler to install)

Since I cannot test it, I will leave implementing this suggestion to others.

fritz-hh, 2020-02-27:
That make sense.

I will test it this week-end.

fritz-hh, 2020-02-29:
@WouterJD : thanks a lot. With all your improvements in the last week, I can now control the target power (in increments of 50 Watt) from the fitness device. In addition I can see the regulation of the brake happening (I did not spend much time checking this though)
The computation of the speed still needs some work (see #17).
When this is done, it would be cool to compare the computed power with the actual measure power (@yegorvin has a power meter:-) )

WouterJD, 2020-03-02:
I have accepted and merged all pull-requests. Will regression-test functioning of FortiusANT on FortiusANT asap. Code looks good.

WouterJD, 2020-03-02:
@yegorvin would be nice to hear how FortiusANT works on your iMagic
@mtbiker22 did you do any steps on your iFlow?

WouterJD, 2020-03-02:
@fritz-hh I doubt whether all merges (involving same files) have gone correctly; I see that usbTrainer.py does not refer fxload.py now!

Backup the version you tested before uploading from github again!

fritz-hh, 2020-03-02:
> @fritz-hh I doubt whether all merges (involving same files) have gone correctly; I see that usbTrainer.py does not refer fxload.py now!
> 
> Backup the version you tested before uploading from github again!

Thanks a lot for checking and merging the PRs.

I saw that indeed, the merge of #23 (see: https://github.com/WouterJD/FortiusANT/pull/25/files?utf8=%E2%9C%93&diff=unified&w=1) removed (at least) the contributions from #16 and #22.
That likely happened when you tried to resolve the conflicts before merging #23.

This can happen and is part of the learning process when starting to work with Github.

What would probably help to make the merges easier, is:
* that you (and all of us) activate automatic trailing spaces removal in your ID (I use atom at it is possible in this IDE). and commit all the *.py files with normalized trailing spaces. That will reduce the number of modified lines in a PR to the actual changes.
* Another thing that always helps, is to trying to keep the PR small.
* When somebody (like me) provides many PR at the same time and after merging a few of them the next one requires too many conflicts to resolve, do not hesitate to ask the contributor to resolve them for you, so that you can merge them easily (that the way I proceed at work)

WouterJD, 2020-03-02:
Correct. I would suggest you propose the modified code as a whole.
There was only one correction: where w is replaced by a the unit in the comment must be km/he instead of mm/sec.

And as suggestion, the version info at the top should be updated as well. Further I think all updates are 👌

fritz-hh, 2020-03-02:
I will send you the delta from the current state

WouterJD, 2020-03-02:
👍

fritz-hh, 2020-03-02:
see #26.

fritz-hh, 2020-03-02:
The imagic should work fine now.
I already tested power regulation (-m option): success.
Grade mode still needs to be tested.
@yegorvin : does it work on your end too?

yegorvin, 2020-03-03:
[image]

A few seconds after the start of the race in Zwift, the error always crashes:
```
11:51:45,476: Trainer send data="97 00 00 00 00 00" (len=6)
11:51:45,476:                   target=151 pe=0 mode=2 weight=10 cal=1040
11:51:45,476: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0x4f, info="02 00 ff ff ff 05 00 08 5c" [ch=2 p=0(Default data page)]
11:51:45,477: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe7, info="02 01 03" msg=0x1
11:51:45,477: Dongle    receive: synch=0xa4, len= 3, id=0x40 Channel Response     , check=0xe4, info="01 01 03" msg=0x1
11:51:45,477: Dongle    receive: synch=0xa4, len= 9, id=0x4f Acknowledged Data    , check=0x2b, info="01 fc 00 00 64 00 00 50 00" [ch=1 p=252(??)]
11:51:45,497: Dongle    timeout
11:51:45,497: ReadFromDongle() returns: ["a4 03 40 02 01 03 e7", "a4 03 40 01 01 03 e4", "a4 09 4f 01 fc 00 00 64 00 00 50 00 2b"]
11:51:45,497: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xd1, info="01 10 19 23 3b 49 04 5c 33" [ch=1 p=16(General FE data)]
11:51:45,518: Dongle    timeout
11:51:45,518: ReadFromDongle() returns: []
11:51:45,518: SendToDongle() returns: ["a4 03 40 02 01 03 e7", "a4 03 40 01 01 03 e4", "a4 09 4f 01 fc 00 00 64 00 00 50 00 2b"]
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python38-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\FortiusAntGui.py", line 826, in OnClick_btnStart_Thread
    self.callTacx2Dongle()
  File "D:/Install/Tacx/FortiusANT-master/pythoncode/FortiusAnt.py", line 156, in callTacx2Dongle
    return Tacx2Dongle(self)
  File "D:/Install/Tacx/FortiusANT-master/pythoncode/FortiusAnt.py", line 700, in Tacx2Dongle
    logfile.Write('FE data page 252 ignored. info=%' % logfile.HexSpace(info))
ValueError: incomplete format
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python38-32\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Program Files (x86)\Python38-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "D:\Install\Tacx\FortiusANT-master\pythoncode\FortiusAntGui.py", line 826, in OnClick_btnStart_Thread
    self.callTacx2Dongle()
  File "D:/Install/Tacx/FortiusANT-master/pythoncode/FortiusAnt.py", line 156, in callTacx2Dongle
    return Tacx2Dongle(self)
  File "D:/Install/Tacx/FortiusANT-master/pythoncode/FortiusAnt.py", line 700, in Tacx2Dongle
    logfile.Write('FE data page 252 ignored. info=%' % logfile.HexSpace(info))
ValueError: incomplete format
11:53:14,714: SetValues() done in 55 ms
11:53:14,883: ReceiveFromTrainer()
11:53:14,883: Trainer recv data="ff 1a 00 00 5c 00 00 00 00 97 00 bc 00 00 00 5a f7 06 3a e5 1a" (len=21)
11:53:14,883: ReceiveFromTrainer: hr=92 sp=2.18 CurrentResistance=151 pe=0x0 cad=0 axis=188 0 0 0
11:53:14,883: SendToTrainer(1, 100, False, -1, 0, 2.18, 0)
11:53:14,884: Trainer send data="e2 00 00 00 00 00" (len=6)
11:53:14,884:                   target=226 pe=0 mode=2 weight=10 cal=1040
11:53:15,130: ReceiveFromTrainer()
11:53:15,130: Trainer recv data="ff 00 00 00 00 00 00 00 00 80 10 bd 00 00 00 fa ff 06 3a e5 1a" (len=21)
11:53:15,130: ReceiveFromTrainer: hr=0 sp=0.0 CurrentResistance=128 pe=0x10 cad=0 axis=189 0 0 0
```

yegorvin, 2020-03-03:
IMagic + Zwift
[image]

PowerMeter + Garmin
[image]

Сurrent issues:
1. The power coincides with some error, but it starts to differ greatly when the gradient changes.
2. The brake force does not match the gradient. When riding horizontally or downhill, I ride as if constantly uphill.

WouterJD, 2020-03-03:
> Some kind of error in FortiusAnt.py", line 700:
> `logfile.Write('FE data page 252 ignored. info=%' % logfile.HexSpace(info))`
Solved and uploaded; typo.

WouterJD, 2020-03-03:
Closed, since iMagic works; let's raise independant issues to address separate topics.

closed 2020-03-03 by WouterJD

fritz-hh, 2020-03-03:
> Range is 0x00...0xFF
> 0x80 is 'neutral' (electrical magnetic field switched off).
> Values less than 0x80 reduce brake force (electrical magnetic field reduces the static magnetic fields) and values greater than 0x80 increase the brake force (electrical magnetic field increase the static magnetic fields).

I saw this in the totalreverse wiki a few days ago too.
To me this explanatation in the wiki is not correct.
Especially `0x80` does not correspond to electrical field switched off.
From what I observed `0x00` = electrical field switch of,f and the resistance increases gradually wenn the value increases.
This is how it was implemented in FortiusANT, with the particularity that we never go below 0x30 (see: https://github.com/WouterJD/FortiusANT/blob/05e1f5bed495407771c5bee08b08ca407cf83b89/pythoncode/usbTrainer.py#L139). I do not know why 30 is set as a minimum, but this is how GoldenCheetah does it. 

I think we still have some tuning to do...

> Closed, since iMagic works; let's raise independant issues to address separate topics.

Good idea!