WouterJD/FortiusANT #14: Who is using FortiusANT? Just for my curiosity.

WouterJD, 2020-02-27:
At the moment FortiusANT is used by Fortius users, in a Raspberry context and two iMagic trainers.
Of cources I'm curious what trainers actually use the software and a post is appreciated, perhaps with picture and/or reference to strava or else...

My strava experiences can be found on: https://www.strava.com/athletes/2885978
Riding, using Trainer Road and Zwift interfacing to FortiusANT simultaneously:
[image]

fritz-hh, 2020-03-01:
I plan to use it as soon as the tacx imagic is supported by FortiusANT. 
And luckily that should be very soon :-)

yegorvin, 2020-03-10:
Thank you @WouterJD and others for great work and the opportunity to give a second life to the old Tacx trainer.
It is very bad when the manufacturer does not support old devices in any way, stimulating excess debris on our planet.

Using FortiusANT the next winter season I will be able to train with more efficiency and interest.

My profile in Strava: https://www.strava.com/athletes/4806561

[image]

RamonFigueras, 2020-03-24:
Thank you Wouter for your excel·lent Job and even a better support  to have FortiusAnt up and running. I'm currently using it under Zwift and Rouvy.
Please extend my thanks to all contributors.
Ramon

vincentb2709, 2020-03-27:
Hello. 
Thank you for this initiative, it permits me to use old solution with Zwift ! I use Tacx Imagic 1904 with win10 and it works well.
However my Tacx is not detected when I launch FortiusANT.exe (prompt opens up and down immediatly with message "Tacx not found"). However, if I launch from command prompt FortiusAnt.py, it detects Tacx, open the status windows in prompt, works and Zwift recognize it.
Can you explain this behaviour ? 
Thank you again for this amazing job!

WouterJD, 2020-03-27:
yes; the .exe is old. I will upload new asap

WouterJD, 2020-03-27:
Where are you from?

vincentb2709, 2020-03-27:
I'm from France! We cannot go out or cycle so I got an old Tacx and were frustrated of its program.  Thank you for this information, I understand now

fritz-hh, 2020-03-28:
> yes; the .exe is old. I will upload new asap

Before you create the new exe, the following lines (https://github.com/WouterJD/FortiusANT/blob/7d4e1c060c5c2a683e051a6de36324fc9db10f8d/pythoncode/usbTrainer.py#L76 and https://github.com/WouterJD/FortiusANT/blob/7d4e1c060c5c2a683e051a6de36324fc9db10f8d/pythoncode/usbTrainer.py#L77) need to be corrected to take into account the path to the firmware in the exe file 

I guess that the path needs to be set similarly to how its done in FortiusAntGui.py: https://github.com/WouterJD/FortiusANT/blob/7d4e1c060c5c2a683e051a6de36324fc9db10f8d/pythoncode/FortiusAntGui.py#L75

WouterJD, 2020-03-28:
Thanks for heads-up!

WouterJD, 2020-03-28:
> yes; the .exe is old. I will upload new asap
I have uploaded new executables and created a new pull-request:
https://github.com/WouterJD/FortiusANT/pull/41

If you confirm that the .exe's work I will merge into the master branch.

@fritz-hh Julian is this the way to go?

oflahero, 2020-03-28:
I have an old Fortius, and run WIndows 7. I'll certainly be trying this when Aliexpress sent me the two ANT+ dongles I ordered this evening! Country has just locked down so lots of indoors training for a while :)

WouterJD, 2020-03-29:
Keep me informed, I'm curious what USBid those dongles will use

E-NINA, 2020-03-29:
Hello, due to this situation about Corona I was also checking my old trainner. I just relaised that Tacx Trainner Software isn't supported anymore. I've try fortiusAnt and I get working parcially as I have some issues.
I'm from Portugal, Evora and I've a Fortius T1932.
I've 2 ANT dongles but I realised one is very old (GARMIN USB) and as only 4 channels so no detected as a ANT dongle by ExplorANT.exe (I don't know if this is soposly to be like that but left here for reference).
The other dongle works correct and is dected by ExplorANT.exe.
I'm able to go ahead as I've a tablet with ANT integrated.
So running fortiusant windows exe file I got the following:
 - on ZWIFT detects HR but not the trainner.
 - on ROUVY detects HR and also the trainer (but as soon I start pedal - fortiusant crashes)
Runing PY files 
 - on ZWIFT detects HR but not the trainner.
 - on ROUVY detects HR and also the trainer and app runs smothly.

Any hit why ZWIFT can't detect trainner?

WouterJD, 2020-03-29:
> I'm from France! We cannot go out or cycle so I got an old Tacx and were frustrated of its program. Thank you for this information, I understand now

If you like, give me a ping on [strava](https://www.strava.com/athletes/2885978) thanks!

WouterJD, 2020-03-29:
> I have an old Fortius, and run WIndows 7. I'll certainly be trying this when Aliexpress send me the two ANT+ dongles I ordered this evening! Country has just locked down so lots of indoors training for a while :)

If you like, give me a ping on [strava](https://www.strava.com/athletes/2885978) thanks!

WouterJD, 2020-03-29:
> I'm from Portugal, Evora and I've a Fortius T1932.

@E-NINA If you like, give me a ping on [strava](https://www.strava.com/athletes/2885978) thanks!

vincentb2709, 2020-04-04:
> Keep me informed, I'm curious what USBid those dongles will use

I tested it this week and it works well with .exe on win10! Thank you!
The soft is just quite slow to start (20 seconds) but do the job!
Just one point, is it normal to not have possibility to have GUI? 
Thanks again for your job and help that is saving my Mont Ventoux training! ;)

WouterJD, 2020-04-04:
Buon jour, Vincent

The exe unzipped itself into a temp folder and then executes; try his causes a slower start. The python version has a quicker start

No GUI is the default, -g flag provides GUI. For more info check wiki.

Succes with your moUnt Ventoux training; when do you go?

Please meet on Strava: https://www.strava.com/athletes/2885978

iepuzaur, 2020-04-10:
hi, I own a Tacx i-vortex (the non-smart one). Would this work with it? I might give it a try :)

Smieszm, 2020-04-14:
Hi Wouter,
I have just found your software that hopefully will clean my I-Flow Multiplayer from dust. Greeting from Poland. :) It would be great at some point if the calibration for the I-Flow would be available as it is for the Fortius. :). I will check how (and if) it works with ZWIFT - I have two ANT+ dongles which seem to be recognized by the soft. Anyway amazing job!

WouterJD, 2020-04-15:
> It would be great at some point if the calibration for the I-Flow would be available as it is for the Fortius. :). 

Please check #30 where @yegorvin has done the meassurements for iMagic, if you have a powermeter then you could do the tests for iFlow. As soon as you have your iFlow working and you start with calibration, start a new issue please.

mattipee, 2020-04-16:
Just tried out FortiusANT on Windows this morning. I have a Fortius T1941 motor brake w/ blue T1932 head unit.

Looking forward to using it more and having a play with source. Good work!

WouterJD, 2020-04-16:
> Just tried out FortiusANT on Windows this morning. I have a Fortius T1941 motor brake w/ blue T1932 head unit.
> Looking forward to using it more and having a play with source. Good work!

Great and welcome in the FortiusANT community; show the results on strava: https://www.strava.com/athletes/4806561

iainhay, 2020-04-17:
@WouterJD my Strava link is 
https://www.strava.com/athletes/iainhay and I'm based in London not Cambridge FYI

WouterJD, 2020-04-17:
> @WouterJD my Strava link is
> https://www.strava.com/athletes/iainhay and I'm based in London not Cambridge FYI

Thanks for adjusting; the list of users is longer than I can maintain and even know of; shortly I may limit to the first two people in a country or so - and then add a pin on each country on the map :-)

brianrow, 2020-04-21:
Hi, this is a great facility for us who don't have or can't get smart trainers. (Cost!) Thank you for you efforts. I've connected my Fortius (0x1932) and can see output data for all parameters, however in 'Run mode' the Target stays fixed at 100W. I'm using GoldenCheetah (GC) as the trainer app, and it seems that either GC is not sending data to Fortius, or Fortius is not accepting. Even if I use manual ERG mode and increase or decrease load, there is no change on the trainer.

1. Where does the Target come from?
2. Does FortiAnt send info back to the trainer?

Thanks again for this utility.

B.

M4M4R2, 2020-04-21:
Hi, 
According to me no data is sent back to the trainer. At least in my case (Tacx Flow + 1932 head) when in use with Zwift the resistance seems to be fixed. The difficulty is increased (slope simulation) by slowing down the speed of my character for the same pace on the bike.

Barh007, 2020-04-23:
Hi, everebody

I try start with check this new App...
But no succes today.
I have good PC win8/х64 with Tacx Fortius and extra old Netbook with XP, х086 (sure),
and two Garmin Ant+ sticks.

FortiusANT.exe  blink for 2 seconds, ExplorANT.exe also ....
(At both pc)

https://www.strava.com/athletes/1160339

BMIL534, 2020-05-02:
Thank you WouterJD for a very quick response.
My name is Brian from South Africa.
RLV rides are real life videos for the Tacx trainer.

aegisdesign, 2020-05-03:
Hi Wouter, I'm in the UK. I picked up a Tacx i-Magic t1904 a few weeks back to keep cycling during lock down. At the moment, stuck with the Fortius software as the ANT+ dongles I bought are the currently unreliable CYCPLUS type. Running the software on a Mac.

Smieszm, 2020-05-03:
Hi Shaun .I am using Cycplus dongles. No issues with them so far. Installed
easy. Maybe reinstall The driver?

aegisdesign, 2020-05-03:
@Smieszm I'm using a Mac. ANT+ support is built in to the OS. There's no extra driver like on Windows. Interesting that your dongle works though as they're unreliable for other people.

Smieszm, 2020-05-03:
@Shaun , I am using Windows 10.The only issue is The trainer i-Flow
Multiplayer that cannot be properly calibrated like Fortius. I am only able
to manualy adjust the resistance instead of Zwift. Hope this can be fixed
at some point ;) with The New release od Fortius Ant

RamonFigueras, 2020-05-03:
@Wouter.
Just in case you are not yet checked, since the addition of Page 54 support, FortiusAnt 2.6 is now compatible with Tacx Desktop Application. I'm had rode several sessions under this TAD application.
Very Good job Wouter 👍

Ramon

mattipee, 2020-05-03:
> Just in case you are not yet checked, since the addition of Page 54 support, FortiusAnt 2.6 is now compatible with Tacx Desktop Application. I'm had rode several sessions under this TAD application.

Excellent news! :D @RamonFigueras @WouterJD

WouterJD, 2020-05-03:
> Hi Shaun .I am using Cycplus dongles. No issues with them so far. Installed easy. Maybe reinstall The driver? niedz., 3 maj 2020, 10:48 użytkownik Shaun Murray <notifications@github.com> napisał:

@Smieszm welcome on board assisting with these CYCLPLUS dongles! Where are you from? I like to know where the FortiusANT users are and you might add a country to the list!

PS. Please exchange info on the issue-pages, this page is to introduce ourselves - as far as we like to of course.

Great to see that so many Tacx-ers are working together!

Endres-to, 2020-05-20:
I have been using the FortiusANT on WIN7 with BigRingVR on Win10. The trainer is Tacx multiplayer T2200 (Brake T1901 and head T1932 Fortius)

I do not think the reported Watt is correct – at least it's only 2/3 of what I get when I am using the original Taxc software.

It this combination supposed to work and be autodetected by the software? Do I have to run the Python code and manipulate anything (Legacy power?) or option 1,2,3 grade to power?

WouterJD, 2020-05-20:
Hi @Endres-to How are you today and welcome on the FortiusANT team. Always curious to know where you come from.

The power reported is the formula as fitting to the T1932 headunit; that's the device I know that is being connected. If you have additional information, such as a real-life measurement (power meter) together with the data the trainer reports, it's always possible to investigate the rootcause.

Suggestion to do: run FortiusANT in manual power mode (-m flag), record the test that is done and measure the "real" power with a power-meter. Analysis can then be done. If you might need legacy power, that would be an option to implement.

In release 3.0 the options have been removed; these affect the Grade2POwer conversion and have no impact on the reported power however. Please have a test and let me know.

Frenske8, 2020-05-21:
Hi, thank you for developing FortiusAnt! I'm running it on Win10 since a week. After changing an Ant-stick (branded Anself, but manufacturer CYCPLUS.) it works great. Now I can use  nu 10 year  old Fortius-trainer with Veloreality. BTW I'm from Eindhoven.

WouterJD, 2020-05-21:
Frenske dank voor de reactie; Wouter uit Elst in de Betuwe. Groeten!

BikeBeppe64, 2020-05-22:
Hi WJD, I'm Giuseppe, from Milan, Italy. Got a tacx T2250 (I-Flow T1932 hand + 1901 motor reconized by Edge 820 as 57591). Started Using FortiusAnt 2.6 with W10 x64, a Garmin USB had with Vector 2s (Power Meter) and a Antifer USB. 2 Garmin Edge 820 & 800.

WouterJD, 2020-05-22:
Benvenuto nel gruppo FortiusANT, Giuseppe. FortiusANT 3.0 é disponibile. il numero 57591 é il deviceID data da FortiusANT al trainer-interface FE-C.

danielopez96, 2020-05-29:
Hi Wouter, we have talked before on Zwift Froums. 

I'm Daniel from, Bogotá D.C, Colombia (Southamerica); Your program has been awesome for me, literally changed the way I train.

Thanks for the passionate effort you put on this.

WouterJD, 2020-05-30:
Thanks and welcome to FortiusANT team

Langsett, 2020-06-13:
Hi

I have now 2 dongles so will try and  rebuild my Fortius/imagic for Zwift 

Which trainer on Zwift do we  pair with /check on their set up once working?

How much tension should be on the resistance roller?

I am near Leeds, Yorkshire England

https://www.strava.com/athletes/304867

Langsett, 2020-06-13:
Oh dear, the FortiusAnt says no trainer detected, and I have swapped leads from brake to head unit,  changed head units and still not detected brake how can I check it is getting power

WouterJD, 2020-06-13:
@Langsett welcome in FortiusANT team. I have created issue #96 for you

EvoxRider365, 2020-06-16:
I am near Ottawa Ontario Canada and looking forward to using FortiusAnt soon.  I just picked up some used ANT USB dongles on the weekend.  I just need to know if TTS4 and FortiusAnt can coexist on the same windows 10 computer.  I still want to use TTS4 for as long as I can.  Thanks

oflahero, 2020-06-27:
> I have an old Fortius, and run WIndows 7. I'll certainly be trying this when Aliexpress send me the two ANT+ dongles I ordered this evening! Country has just locked down so lots of indoors training for a while :)

Almost 3 months since I posted this! After 3 attempts to purchase dongles and a lot of bad online delivery luck I currently have 5 (5!), none of which work :( Unfortunately they're all the evil CYCPLUS type, though some are branded differently (ANSELF, for example, but are functionally identical.) End of the road for me unfortunately!

My advice is to just buy an expensive, proven dongle (Suunto?), and don't cheap out like I did! If I find one, I may yet end up using FortiusANT some day!

BikeBeppe64, 2020-06-27:
> Almost 3 months since I posted this! After 3 attempts to purchase dongles and a lot of bad online delivery luck I currently have 5 (5!), none of which work :( Unfortunately they're all the evil [...]

Just my 2 cent: Anself works perfectly with FortiusAnt and Win 10.  Try to consider that the problem could be the OS. Anyway elwait for Wouter JD

oflahero, 2020-06-27:
> Just my 2 cent: Anself works perfectly with FortiusAnt and Win 10. Try to consider that the problem could be the OS. Anyway elwait for Wouter JD

The ANSELF dongle identifies in Device Manager identically to the CYCPLUS (Dynastream); FortiusANT also identifies it as CYCPLUS. It also arrived in a in packaging quoting a '@cycplus.com' email address despite being printed with ANSELF on the unit - I think there's a comment on another thread referencing this - https://github.com/WouterJD/FortiusANT/issues/65#issuecomment-623399932

So just be aware when purchasing your cheap dongles that counterfeiting appears to be an issue.

BikeBeppe64, 2020-06-27:
[image]

That's what I use, honestly I have a Garmin and an Anself

WouterJD, 2020-06-29:
Well, a lot is said under #65 #76 #77 and error-recovery is implemented to the max, perhaps your dongles have other (erratic) behaviour. I do not know what price the CYCPLYS/Anself dongles are sold for. If you cannot purchase them, contact me 1:1 to see how I can help.

WouterJD, 2020-07-30:
@msjnaessens thanks for the info; I plan to create a proper set of documentation, but that will be a winter-time activity :-)

Sigfreid74, 2020-08-01:
Hi Wouter, from Melbourne, Australia 
- I have a Genius T2021 Brake (Non-smart TACX proprietary ANT+), and an original Tacx 2018 Dongle. I am currently running TTS 4.22.5.
I am interested in liberating my trainer a little to experiment with some of the newer software around. Surely if the Vortex can work, then the Genius must be possible too at some point. I would be happy to try and test any ideas you have. 
Interestingly, I decided to jump in and uninstall the "Jungo" TACX USB drivers, and install the "libusb-win32-1.2.6" drivers instead. TTS4 still works for me! (Windows 10)

FortiusANT is not registering the trainer yet though.
Any tips to try and get it working?

Thanks

rgprints, 2020-09-07:
Hello Wouter, I'm Rob from Blue Bell, Pennsylvania, USA, near Philadelphia. 

Ik ben oorspronkelijk van Leiden, maar woon in de USA vanaf 1956. Omdat mijn Hollands niet meer zo prima is, sla ik nu weer terug naar Engels.

This is terrific work you have done, so I really hope I can get this all to work for me.

I have 64 bit Windows 10 and have done the libusb stuff. After I do that the i-Magic shows under "libusb-win32 devices" and not under USB devices. Is that correct?
[image]

I would like to test the connection to my iMagic before I acquire any ANT dongles. Is that possible? I thought I could do so by using -m or -M on the command line but that seems to have no effect (I'm using the Windows *.exe file).

Here is what I see in the GUI:
[image]
When I commence pedaling the trainer, none of the dials move. The only interaction that seems to work is that if I press the Cancel button on the i-Magic, then the GUI closes.

So I'm not sure what's going on unless nothing works until ANT dongles are available?

One other piece of information - I don't have a working heart rate monitor. Is that an issue? I've tried putting -H-1 on the command line.

I look forward to hearing from you.

Regard,
-Rob

rgprints, 2020-09-10:
Hello Wouter, I'm Rob from Blue Bell, Pennsylvania, USA, near Philadelphia.

Ik ben oorspronkelijk van Leiden, maar woon in de USA vanaf 1956. Omdat mijn Hollands niet meer zo prima is, sla ik nu weer terug naar Engels.

I have an  i-Magic T1902 unit. Windows 10 laptop and two Anself ANT units.

This is terrific work you have done. 

I had some trouble getting set up, but the cause was in my cable connections. Once I figured out that was the issue, I got it all to work. Did my first Zwift just now!

Thank you for all of the effort. It really paid off for me.

-Rob

Langsett, 2020-09-11:
> Hello Wouter, I'm Rob from Blue Bell, Pennsylvania, USA, near Philadelphia.
>
> Ik ben oorspronkelijk van Leiden, maar woon in de USA vanaf 1956. Omdat mijn Hollands niet meer zo prima is, sla ik nu [...]

Hi Rob,  I have the same hardware, and didnt manage to get set up last time  due to IT and connection problems , do i need  TTS software if so what version ?

What trainer do you set it as on Zwift   and how true are the power readings?

With advice i hope to get set up for this winter 

Martyn , Yorkshire , England

rgprints, 2020-09-11:
> Hi Rob, I have the same hardware, and didnt manage to get set up last time due to IT and connection problems , do i need TTS software if so what version ?
>
> What trainer do you set it as on Zwift and [...]

Hi Martyn,
You do not need TTS software. My Tacx software is on an old Vista computer, and I set everything up on a different computer running Windows 10, 64 bit. 

I've never used Zwift before, so I just sort of guessed. For trainer I chose Other, although I toyed with choosing one of the Tacx options. However, since none of the Tacx options available in Zwift were i-Magic or Fortius, I thought it better to choose Other.

I'm not sure how good Zwift's power readings are, but I suspect they are high. It is an area I plan to dig deeper into. I also have not done any calibration, so my input on this is almost purely conjecture. My suspicion that they are high is based on a comparison to my power ratings on a very different trainer, an Expresso trainer that I used at my gym. (Which went bankrupt due to Covid-19.) On that trainer, my sustained power level (20 minutes or longer) when in shape and working hard was between 180-200 watts. Yesterday on my first and so far only ride on Zwift it was 230. I'm somewhat out of shape due to 30 days of injury recovery and I wasn't pushing myself much. Of course I have no idea of the Expresso accuracy, to begin with.

Since I just did this, let me run you through the steps. that worked for me:

1.  Set up an i-Magic driver. Download and unzip libusb-win32-bin-1.2.6.0.zip.
2.  Drill down in the unzipped libusb stuff to the bin folder and right-click inf-wizard.exe and choose Run as Administrator (very important).
3.  Click the next button (contrary to the instructions in the window do not have your device connected, yet. If you do, it may not be obvious how your device is being labelled by your system. Instead, as you will see in the next step, connecting and disconnecting the device will help you recognize the device to use.)
4.  Examine the list of USB devices, connect the i-Magic to a USB port and hopefully you will see the device added to the list. Mine was labelled as "undefined" or "unknown" (I don't recall right now and cannot reproduce) and not as i-Magic or Fortius or Tacx. It may also show x1902 under product ID. Select the device and click Next.
5.  Now you have an opportunity to indicate the manufacturer and name of your device. Doesn't matter what name you choose, I used "i-Magic" and specified "Tacx" as the manufacturer. Click Next.
6.  This will open the standard Save As dialog box. The default name for the file you are about to create will be derived from the name you used in the prior step. Use the dialog box to select a folder where you want the file saved. Does not matter what folder you use, just remember where you saved it. Click Save.
7.  This brings you back to the Inf-Wizard and click Install Now. If you get the error "system policy has been modified to reject unsigned drivers" it means that you didn't run (step 2) as an Administrator. Hopefully you will be rewarded with "Installation Successful".
8.  Run FortiusAnt.exe (this worked fine for me, no need to set up to run with Python, and no need at this point to mess with command line switches). This will open both a DOS (CMD) window and a Fortius Antifier V3.2.1 window. If it correctly found the i-Magic unit, you should see "T1902 head unit initialized (iMagic)" about halfway down the screen. If you already have an ANT dongle connected, you should also see that it was found. If you don't have an ANT dongle connected, and don't have one to test with, then about the only thing you can do to confirm that the sotware is communicating with the i-Magic is to press the Cancel button on the i-Magic and it should close both the Fortius window and the DOS window. I expected to be able to use the -m command line switch to see if I could see speed, power and cadence in a stand-alone mode without an ANT device, but was not able to do that. To go further, you need to have an ANT device connected.
9.  So, connect your ANT device. If Antifier is already running, just click on the Locate HW button, and the Antifier interface hopefully finds the ANT and indicates that (for me it says "Using Dynastream Innovations Dongle" - I have the inexpensive ANSELF dongles). In addition, the Runoff and Start buttons should now also no longer be grayed out. If Antifier was not already running, just start it with the default switches and it should automatically find the i-Magic and the ANT devices, and start (as if you had pressed the Start button).  Note that if you want to use the Runoff, and you've started Antifier with default switch settings, you'll need to click on Stop to enable the Runoff and Start buttons. 
10.  Once connected to both the i-Magic and an ANT device, and when Antifier itself has been started (indicated by having only the Stop button not grayed out), you can now test speed, cadence and power meters by getting on your bike. 
11.  Finally, to use Zwift, connect your second ANT device. Make sure that Antifier has been started (only the Stop button is available), and start the Zwift app. Click on Let's Go and Zwift will open the full screen window and you can log in. Now you may need to establish various connections, but that should be pretty self explanatory. I've found that it was only necessary to do once. Now when I start Zwift it automatically connects to the i-Magic unit and to the second ANT dongle. (Never having used an ANT dongle I had expected that I might need to do some setup to ensure that the two ANT dongles were on different frequencies or otherwise were set up to stay out of each others way, but was pleasantly surprised to find that all was taken care of automatically.)

So, I hope this helps and good luck and good riding!

-Rob

stedec, 2020-09-18:
Hello Wouter, I'm Steph from Belgium.

I realised I still had an old Tacx Flow stored in my attic. I saw the software project you developped and I'm triggered to give it a try as I would like to use Zwift or Rouvy during the winter  :-). 

I would like to know if my tackx flow is compatible with the software. I have a tackx flow T1680 but I realised it doesn't have the possiblity to connect it to a laptop through USB. I am able to buy a second hand Tacx Upgrade Virtual reality kit with a T1932 interface. I bought 2 ant+ dongles from [amazon](https://www.amazon.co.uk/Anself-Stick-Garmin-Forerunner-310XT/dp/B01M4NFENM?th=1). 

Do you think this setup will be able to work with FortiusAnt?

I added some pictures about the devices:
[image]
[image]
[image]
[image]

Thanks in advance for the information!

[453 more replies]