WouterJD/FortiusANT #17: Crash if (non bundled) application called from another directory

fritz-hh, 2020-02-29:
A crash occurs, if the non bundled application is started from within another directory (see example below)

```
C:\Users\julien\Documents\GitHub\FortiusANT>python pythoncode\FortiusAnt.py -m -g
13:14:39,093: FortiusANT started
13:14:39,093: -g
13:14:39,094: -m
13:14:39,094: ------------------
Traceback (most recent call last):
  File "C:\Users\julien\Documents\GitHub\FortiusANT\pythoncode\FortiusAntGui.py", line 704, in OnPaint
    dc.DrawBitmap(self.BackgroundBitmap, 0, 0)          # LeftTop in pixels
wx._core.wxAssertionError: C++ assertion "bmp.IsOk()" failed at ..\..\src\msw\dc.cpp(1313) in wxMSWDCImpl::DoDrawBitmap(): invalid bitmap in wxMSWDCImpl::DrawBitmap
```

fritz-hh, 2020-02-29:
I will propose a PR

WouterJD, 2020-03-01:
Current directory is expected to be the directory where FortiusANT.py is located, to load FortiusANT.jpg. Should not crash when absent but did not investigate to solve.
Just let me know what code-suggestion you have

fritz-hh, 2020-03-01:
My suggestion is in pull request #18.

PR #18 (2020-03-02): Issue #17 - Fix crash when app is started from another dir
Fixes #17

Fix a crash that occurs if the non bundled app is started from another directory then the `pythoncode` directory.
commit ac6db936 (2020-03-02): Issue #17 - Fix crash when started from another dir

Fixes #17

Fix a crash that occurs if the non bundled app is started from another directory then the `pythoncode` directory.
closed 2020-03-02 by WouterJD