WouterJD/FortiusANT #23: Magic Speed computation issue

fritz-hh, 2020-02-29:
It seem that the raw speed (called wheelspeed in the app) values (byte 1-2) returned by the Imagic (legacy protocol) differ from those of the Fortius (new protocol). This as the consequence that the speed reading are wronging displayed (much to low in case of imagic)

fritz-hh, 2020-02-29:
Solving this is not as easy as one might think.

Indeed the `WheelSpeed` appears at many different places in the code.
I.e. the `ReceiveFromTrainer` function returns the `WheelSpeed` and not only the speed in engineering units (in km/h). As a consquence, the code that calls `ReceiveFromTrainer` has to deal with the Fortius specific WheelSpeed values. 

My proposal would be to modify `ReceiveFromTrainer` so that it provides an interface (returns values) that are not protocol specific (So `ReceiveFromTrainer` would returns the speed in km/h). (and also remove the dependencies to `WheelSpeed` at other places in the FortiusANT code)

@WouterJD : What is your opinion on this?

fritz-hh, 2020-02-29:
I will now be offline for the rest of the day.

fritz-hh, 2020-02-29:
I have thought about it, and could provide you a proposal for review :-)

WouterJD, 2020-03-01:
WheelSpeed is mm/s, so ReceiveFromTrainer could translate the iMagic value to that unit as well.
SPeed (km/hr) and WheelSpeed(mm/s) are maintain simultaneously; Speed is for the user and ANT, WheelSpeed is for the power-calculations.

Two main questions:
- What does the iMagic return for speed
- What should the formula be

Just quickly checking your posts. Curious on your answers.

fritz-hh, 2020-03-01:
Here is what I found out in goldencheetah (confirmed by my own testing too) (see `SpeedScale_Legacy`)

```
SpeedScale = 301							        # TotalReverse: 289.75
SpeedScale_Legacy = 11.9                            # GoldenCheetah: curSpeed = curSpeedInternal / (1.19f * 10.0f);
```

fritz-hh, 2020-03-01:
> WheelSpeed is mm/s, so ReceiveFromTrainer could translate the iMagic value to that unit as well.
> SPeed (km/hr) and WheelSpeed(mm/s) are maintain simultaneously; Speed is for the user and ANT, WheelSpeed is for the power-calculations.

Thanks for the clarification.
As `WheelSpeed` and `SpeedKmh` are somehow redundant, but `WheelSpeed` has strange units (even not exactly mm/s)
I thought, that I could make sense to change most of the interface to use `SpeedKmh` only.
I did work on this yesterday evening (see [here](https://github.com/fritz-hh/FortiusANT/commit/87a081c30c4e699c092b9d586f1292e03fe6099f))

I did this change very carefully and paid attention at not changing anything else (especially the formulas in power-calculations)
I did a 1st workout with imagic and these changes applied and it worked very well.

If you are interested, I can make a PR out of this.

WouterJD, 2020-03-01:
It would be an improvement that only Speed is returned.
I also hate that "Not found" is returned in Speed when an error occurs, test should never be put in a numeric field (even though an object has a dynamic type); it's bad programming as remaining from Antifier.
I suggest you make a PR for the iMagic porting you have done and I will validate and accept.

WouterJD, 2020-03-01:
Main question: does it work now? Did you test without ANTdongle till now? (using -m flag)?

WouterJD, 2020-03-01:
When standardizing on Speed, check that Speed is rounded to one digit. This may reduce precision.
I would be OK that Speed is not rounded; rounding can be done on displaying the figure.

fritz-hh, 2020-03-01:
> Main question: does it work now? Did you test without ANTdongle till now? (using -m flag)?

Yes it works for me. I did a 1 hour workout with -m (I do not have the ANT dongles yet)
* speed computation is as expected
* power / resistance regulation work well

> When standardizing on Speed, check that Speed is rounded to one digit. This may reduce precision.
> I would be OK that Speed is not rounded; rounding can be done on displaying the figure.

Indeed, in my proposal speedkmh is still rounded to 1 digit after the comma (which does not reduce precision for imagic, as the scale is around 10. But reduces precision for fortius quite a lot)
Yes, it would make sense indeed to avoid rounding or round to 2 digits (and round to 1 digit for displaying).
I will adapt my proposal with this regard.

WouterJD, 2020-03-01:
Julian, I'm very happy that it works; it's an extra reward on the work invested in FortiusANT.
Would like to see you on Strava or so

fritz-hh, 2020-03-01:
I just increase the precision to 2 digits in `Wheel2Speed()`.
The gui and log message seam to already round the speed to 1 digit.
So no update required there.
Pull request is coming.

fritz-hh, 2020-03-01:
See pull request #25 
Do not hesitate to ask me if you need some help with the pull requests.
I will do my best.

fritz-hh, 2020-03-01:
> Julian, I'm very happy that it works; it's an extra reward on the work invested in FortiusANT.
> Would like to see you on Strava or so

I saw that you subscribed my profile!
Once I have the dongles, I will post the workout with FortiusANT on strava.

WouterJD, 2020-03-01:
👍
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

PR #25 (2020-03-02): Issue #23 - Imagic - Refactor & Fix SpeedKmh computation
Fixes #23

- Refactor the code to use `SpeedKmh` instead of `WheelSpeed` wherever possible, without changing the structure of the code. Especially, ExplorAnt.py and FortiusAnt.py were modified so that they do not need to deal with and understand `WheelSpeed` anymore (Rationale: WheelSpeed is Trainer specific).
- For imagic: Correct the calculation of the Speed in Kmh based on internal wheelspeed data
- Increase precision of SpeedKmH to 2 digits

Code tested with Imagic using the `-m` option
commit 87a081c3 (2020-03-02): Issue #23 - Fix imagic speed computation

Fixes #23

- Refactor the code to use `SpeedKmh` instead of `WheelSpeed` whereever possible, without changing the structure of the code. Especially, ExplorAnt.py and FortiusAnt.py were modified so that they do not need to deal with and understand `WheelSpeed` anymore (Rationale: WheelSpeed is Trainer specific).
- Correct the calculation of the Speed in Kmh based on internal imagic wheelspeed data

Code tested with Imagic (factors tuning might still be required)
closed 2020-03-02 by WouterJD