WouterJD/FortiusANT #24: ANT+ dongle compatibility

fritz-hh, 2020-03-01:
As the imagic will be fully useable soon on FortiusANT, I am now looking for a cheap compatible ANT+ dongle.
I found this one (https://www.amazon.co.uk/Anself-Stick-Garmin-Forerunner-310XT/dp/B01M4NFENM?th=1), and I wonder if it is likely to work with FortiusANT.
I plan to buy 2 of these.

WouterJD, 2020-03-01:
I use two dongles (one for FortiusANT, the other one for TrainerRoad on the same system or other system)
- manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
- manufacturer=Suunto, product=Movestick mini, vendor= 0xfcf, product=0x1008(4104)

I paid EUR 35,= @ https://www.mantel.com

I would think an ANT+ dongle would work, but of course cannot guarantee.
It's a good price and the description (Anself USB ANT+Stick for Garmin Forerunner 310XT 405 610 Compatible With CycleOps Virtual Trainer Sunnto Watch Zwift TrainerRoad PerfPRO Studio) suggests it would work for Fortius/TrainerRoad as well.

fritz-hh, 2020-03-01:
Thanks. 
I will try it and report.

fritz-hh, 2020-03-10:
The dongles (both connected to the same PC) worked perfektly with the latest version of FortiusANT.

closed 2020-03-10 by fritz-hh