WouterJD/FortiusANT #33: requirements.txt

WouterJD, 2020-03-03:
@fritz-hh FortiusANT is depending on usb (import usb.core) and fxload.
Also, on python version 3

Is this something that can be defined in requirements.txt?

Thanks in advance for info!

fritz-hh, 2020-03-03:
The short answer, is: I have no experience with python...

The longer answer:
* I did not find any module called `usb` in pip, and I had to remove it from the requirement file (installing the depencies lead to a crashes before I removed `usb`).
From what I understood, `usb.core` is provided by the `pyusb` module, that you already have in the requirements file.
* I do not think that you have an external dependency to fxload, as `fxload.py` is part of your code: https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/fxload.py
* For python 3, I do not know.

WouterJD, 2020-03-04:
Thanks
usb- agree
fxload - I would want to be dependant on an install and not copy code from totalreverse
python 3 - check sys.version_info at runtime

closed 2020-03-04 by WouterJD

fritz-hh, 2020-03-04:
Completely agree.
Did you ask totalreverse, if he would be willing to publish fxload as a python module in npm?