WouterJD/FortiusANT #38: Get struct error:  unpack requires a buffer of 64 bytes

RamonFigueras, 2020-03-22:
Hello Wouter,

As per our chat on Strava I'm reporting an error than stops me from use FortiusAnt.
Let me first identify the configuration  I'm using:
I'm running FortiusAnt on Windows10 and I've downloaded Python 3.8.2
As Usb ANT+ Dongle I'm using is Garmin. This is the one that connects to Fortius trainer
As to connect with tha App, I'm currently using Rouvy as Zwift doesn't recognizes my trainer, I'm currently using an Android phone with ANT+ services support.
I'm running FortiusAnt from a CMD window using the following arguments:
FortiusAnt -d DEBUG -b 2096,55/34,11/34 -g 
The program starts identifying both the USBTrainer and the ANT+ Dongle and opens the GUI. Once the Start Button on the GUI is clicked the FE-C trainer is recognized on the app. 
But at the same time, just after the Start button is clicked the following error appears on the Debug Window:

struct.error: unpack requires a buffer of 64 bytes
Traceback (most recent call last):
  File "FortiusAntGui.py", line 738, in OnTimer
    self.callIdleFunction()
  File "FortiusAnt.py", line 131, in callIdleFunction
    import antHRM            as hrm
  File "FortiusAnt.py", line 208, in IdleFunction
    logfile.Write (msg)
  File "usbTrainer.py", line 619, in ReceiveFromTrainer
    data   = False

I tried to run FortiusAnt.py on the Python Shell and the same error occurs.

I'm not familiar at all with Python so no idea of what I can do to solve the issue.

Do you have any clue on this?

Thanks in advance for your support

WouterJD, 2020-03-22:
Q1. FortiusAnt -d DEBUG -b 2096,55/34,11/34 -g

PLease specify FortiusAnt -d 127 -b 2096,55/34,11/34 -g and send the created logfile, so I can see what data is received from the trainer.

WouterJD, 2020-03-22:
Q2. PLease check whether you use the most recent code; the logfile as generated under Q2 will provide the versions to me; it's always good to use the latest code when reporting errors.

WouterJD, 2020-03-22:
Q3. Getting to the point; usbTrainer.py basically does two important operations: **send** commands to the trainer and **receive** response back. The response is believed to be 64 bytes (if larger than 40) but no check is done (structure as provided by original antifier code) and you are the first example where another buffer-size is returned.
The logfile from Q1 will provide more information for further analysis.

WouterJD, 2020-03-22:
Q4. "Zwift doesn't recognizes my trainer"; this may be caused because FortiusANT crashes before ZWIFT has the chance to recognize. Let's start with the logfile and see where we get. The logfile may also reveal what trainer you excatly use

Thanks for posting the issue.

RamonFigueras, 2020-03-22:
I'm sending the log file produced. Let me kmow is this is what you expected.

Thanks for taking care,

Ramon
[FortiusANT.2020-03-22 21-37-21.log](https://github.com/WouterJD/FortiusANT/files/4366413/FortiusANT.2020-03-22.21-37-21.log)

WouterJD, 2020-03-22:
Q1. Logfile received, so you have a 1942 trainer that requires 12/48 bytes interface ([see](https://github.com/totalreverse/ttyT1941/wiki))
Q2. You do not have the most recent code
Q3. I will study to implement the 12/48 interface as described by TotalReverse
As soon as I have implemented, I will upload. Give me some time please

WouterJD, 2020-03-23:
The solution seems simple; padd the 1942's buffer to 64 bytes with zeroes.
The solution is uploaded; download the whole codeset and let me know the results please

RamonFigueras, 2020-03-23:
Hi Wouter,

Sorry for the delay I've been Zwifting this morning (using my Vector2 powermeter). 
Now I'm already tested FortiusAnt, but I suspect that I'm not running the downloaded versión ¿? andI'm very sorry to say that I got the same results as always.
I proceed to download the entire zip file from de Code page in GitHUB. Uncompressed the zip file in a directory called in C:\ and I ran the FortiusAnt executable from that last directory.
Should I download the whole codeset from any other place?
For you to analyze I'm attaching the log file I obtained according the procedure you gave yesterday.
[FortiusANT.2020-03-23 11-41-43.log](https://github.com/WouterJD/FortiusANT/files/4369125/FortiusANT.2020-03-23.11-41-43.log)

RamonFigueras, 2020-03-23:
Hello again,

I realized that on the last comment I closed mistakenly this issue, I'm already reopening it.
The last test I did I believe that Will put some light on what the current issue is:
This time I ran the FortiusAnt.py from the Python shell instead and got the following error message, stating that an array couldn't be expanded with bytes but with another array. See the message as it appears:
Traceback (most recent call last):
  File "C:\FortiusAnt-master1\FortiusANT-master\pythoncode\FortiusAnt.py", line 973, in <module>
    Tacx2Dongle(x)
  File "C:\FortiusAnt-master1\FortiusANT-master\pythoncode\FortiusAnt.py", line 520, in Tacx2Dongle
    usbTrainer.ReceiveFromTrainer(devTrainer)
  File "C:\FortiusAnt-master1\FortiusANT-master\pythoncode\usbTrainer.py", line 771, in ReceiveFromTrainer
    data += b'\00' * (64 - len(data))
TypeError: can only extend array with array (not "bytes")

RamonFigueras, 2020-03-23:
Hi Wouter,

I think I got it. I replaced the statement you added on usbtrainer.py 
 data += b'\00' * (64 - len(data))

with this one
for v in range(64-(len(data))):
        data.append(0)

Now running the FortiusAnt on the Shell doesn't give any error.
TODO running the Shell with the aproppiate arguments. I Will do after lunch. I Will let you know the results

WouterJD, 2020-03-23:
> I'm not familiar at all with Python so no idea of what I can do to solve the issue.
Well; you're more familiar than you thought you are :-) Gracefully accepted this code-improvement and curious for your experiences.

What trainers do you use? I see Zwift-results, so there must be more!

RamonFigueras, 2020-03-23:
Hello Wouter,

As I told you I'm not familiar with Python, in fact this is my first experience with that programing environment, anyway I have some basic principles in software programing. Hopefully with your help unveiling where the structure error came from, the Shell giving the clue on the new error after your module modification and the invaluable help of the Python tutorials in Google.

RamonFigueras, 2020-03-23:
Some results from the modified usbtrainer.py module:

Ran from Python Shell with -d DEBUG -g -p 0.6 and when running without being connected to the APP it seems to run correctly, but once I connect to the App, currently Rouvy as Zwift still doesn't recognize the trainer, the GUI stops to show Speed, Cadence and Power. 

In order to give you with some more information I ran the script with de -d 127 argument to produce a log file. By the end of the file you Will see the data send and received when connected to Rouvy. From the point that I started the application the GUI indicators went all to 0 and the speed and power was not received by the app. 
[FortiusANT.2020-03-23 17-34-15.log](https://github.com/WouterJD/FortiusANT/files/4371301/FortiusANT.2020-03-23.17-34-15.log)

WouterJD, 2020-03-23:
Well, it seems you are quite an end in getting the system work! Compliments!

FortiusANT does NOT go into calibration mode, because that is supported for 1932 only, although your trainer might support it. This is as expected.

I have been looking through the logfile and it seems OK.

17:34:27,161 The trainer is stopped (mode=0)
17:34:27,208 The trainer has default resistance 1040 (mode=2)
17:34:37,836
17:34:46,398 A message is received from a trainer, going into grademode

17:34:47,032: Target=4.0%(106W) Speed=15.9kmh hr=  0 Current= 96W Cad= 63 r=1617  60%
This is the -once in a second- logrecord that shows the vaues as displayed on screen

17:34:47,230: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xfa, info="01 19 4a 3f 32 0f 79 00 30" [ch=1 p=25(Trainer Data)]
This is the ANT message containing Cadence and power.

Also the other expected messages
17:34:49,132: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0x7c, info="01 10 19 06 50 e2 10 00 33" [ch=1 p=16(General FE data)]
17:34:52,635: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0xda, info="01 50 ff ff 01 59 00 3b 0b" [ch=1 p=80(Manufacturer Info)]
are sent

The last message shows speed and cadence, so I do not understand " the GUI indicators went all to 0". Are you sure that FortiusANT did not crash and that messages are displayed in "the console"?
17:34:53,576: Target=4.0%(105W) Speed=16.0kmh hr=  0 Current=184W Cad= 63 r=2869  60%
** Please check ** That FortiusANT is still running...

And another questions remains "currently Rouvy as Zwift still don't recognize the trainer". You  know how ZWIFT works, so that is not the issue. Perhaps the distance from your "APP" and the computer is too far away; I have noted that some dongles require a quite short distance. Give a try.
** Please check ** Dongle distance

In case you might have two ANTdongles available, try ExplorANT.py, which "explores" the ANT network.
** perhaps ** it helps to debug

Thanks for using FortiusANT, I'm curious on next steps.

RamonFigueras, 2020-03-24:
Regarding the FortiusAnt, yes it crashes, I'm sending the image of the GUI while I was pedaling. 
The issue was:
 File "C:\FortiusAnt-master1\FortiusANT-master\pythoncode\logfile.py", line 86, in HexSpace
    str = binascii.hexlify(info).decode("utf-8")
TypeError: a bytes-like object is required, not 'bool'

I already went through it modifying the code in logfile.py in line 86 this way:

if info == False:
        return

Now it seems that everything is working well, I'm very grateful for your support Wouter, thousands thanks.

 The only exception is Zwift not recognizing my Tacx trainer. It seems that in your case you have no issue with that. Are you using Fortius or Imagic? Perhaps is the device type sent to the App what makes de difference.

WouterJD, 2020-03-24:
Thanks for resoving this issue. I published a new logfile.py

FortiusANT does not publish Fortius or iMagic on the ANT-interface, because those devices are not defined in ANT (they are not ANT-trainers!)

The device published is determined in antDongle.py
> ModelNumber_FE      = 2875     # short antifier-value=0x8385, Tacx Neo=2875
> SerialNumber_FE     = 19590705 # int   1959-7-5
> HWrevision_FE       = 1        # char
> SWrevisionMain_FE   = 1        # char
> SWrevisionSupp_FE   = 1        # char

So FortiusANT pretends to be a Tacx NEO. Antifier used 0x8385 and I do not know what device that is supposed to be. ZWIFT recognizes FortiusANT without issues in my case, but also some other users have no issues in that area.

Please check carefully again; it's worth while some extra investigation!

WouterJD, 2020-03-24:
I will do a check with Rouvy; looks promising.

RamonFigueras, 2020-03-24:
Bingoooo!!!! Last issue solved.

Today I got a second Dongle and guess what ....

Zwift recognized my Tacx Fortius as a Tacx Neo as you told me. So the issue was the Ant service on my Android that identified the trainer as a Generic FE-C device.

A very big thanks Wouter and all the contributors to this excellent piece of software.

I believe that this issue can be closed

WouterJD, 2020-03-24:
Hurray; another happy Zwifter💪

closed 2020-03-24 by WouterJD