WouterJD/FortiusANT #39: tacx app

andritre, 2020-03-24:
Hi, I wanted to know if you could use FortiusAnt for Tacx app descktop windows. I have tried but it seems that it cannot connect. Thanks AR

WouterJD, 2020-03-24:
Tacx seems to do utmost best to close software. The desktop app does some checks that Zwift, TrainerRoad and others not do. Perhaps even hidden functions in the ANTinterface.

Who has a bright idea, please assist to get it Work🤔

WouterJD, 2020-03-26:
In addition to previous; @RamonFigueras has tested with rouvy. That may be a good replacement for Tacx desktop app.

RamonFigueras, 2020-03-26:
Is shown by the bycicle icon. See picture.

Best regards

Ramon

RamonFigueras, 2020-03-26:
Regarding the behavior of the current configuration of FortiusAnt, If I'm not mistaking the actual setup is for the use of Imagic trainer and, as I own an old Fortius with motor, not magnets, the algorithms to calculate Power and resistance had to be twin. I'm currently working on this. 
Nevertheless the program is usable with both Zwift and Rouvy and Fullgaz even though the resistance is not responding linear to Grade. I replied here to complement the Wouter's one but I believe that this comment deserves an specific issue, so if you don't mind I will open a new issue and copy the comment there

WouterJD, 2020-03-27:
With having https://fulgaz.com/ mentioned, topic can be closed

closed 2020-03-27 by WouterJD

RamonFigueras, 2020-04-11:
Hello Wouter,

Any idea on how to broadcast the FortiusAnt data via Bluetooth FTMS?

Regards,

Ramon

Enviat des del meu dispositiu mòbil

WouterJD, 2020-04-11:
Yes: replace the ANT+ interface by Bluetooth.
The advantage would be that a mobile could be used to run Zwift.

Purpose of FortiusANT is to make an old Tacx usable and mission accomplished.

Implementing Bluetooth is not on my bucket-list. Next step will be to support Vortex and Bushido, as soon as owners will present themselves.