WouterJD/FortiusANT #40: Tacx Fortius #1941 with motor break doesn't show the correct Grade and Power

RamonFigueras, 2020-03-26:
Regarding the behavior of the current configuration of FortiusAnt, If I'm not mistaking the actual setup is for the use of Imagic trainer and, as I own an old Fortius with motor not magnets, the algorithms to calculate Power and resistance had to be twik. I'm currently working on this.
Nevertheless the program is usable with both Zwift and Rouvy and Fullgaz even though the resistance is not responding linear to Grade.

WouterJD, 2020-03-27:
The setup you use is identical to mine, the two trainers have minimal interface differences (48 versus 64 bytes remember).
The power- and grade- to resistance algorithm is well documented in wiki, source code and accompanying Excel sheet.
See also issue #30 even where it mentions iMagic.

Please note that we use good old Fortius and try to equalize behaviour to Tacx Neo; of course Tacx' developments are not limited to the ANTinterface!

Study issue #30 and wiki and report your experiences; do not forget to experiment with the Commando-Line variables (see wiki)

WouterJD, 2020-03-27:
@RamonFigueras I have added [The FortiusANT's powercurve (2)](https://github.com/WouterJD/FortiusANT/wiki/The-FortiusANT's-powercurve-(2)) to the wiki and hope it helps.

WouterJD, 2020-04-06:
@RamonFigueras is training and shows results on Strava

closed 2020-04-06 by WouterJD

WouterJD, 2020-05-13:
@RamonFigueras The powercurve is modified and ready for test.

Beta-version available in
https://github.com/WouterJD/FortiusANT/tree/54-usbTrainer

Comments please add to
https://github.com/WouterJD/FortiusANT/pull/84