WouterJD/FortiusANT #42: Trainer not correctly detected by ZWIFT, Dongle not supported

WouterJD, 2020-03-29:
Hello, due to this situation about Corona I was also checking my old trainner. I just relaised that Tacx Trainner Software isn't supported anymore. I've try fortiusAnt and I get working parcially as I have some issues.
I'm from Portugal, Evora and I've a Fortius T1932.
I've 2 ANT dongles but I realised one is very old (GARMIN USB) and as only 4 channels so no detected as a ANT dongle by ExplorANT.exe (I don't know if this is soposly to be like that but left here for reference).
The other dongle works correct and is dected by ExplorANT.exe.
I'm able to go ahead as I've a tablet with ANT integrated.
So running fortiusant windows exe file I got the following:
 - on ZWIFT detects HR but not the trainner.
 - on ROUVY detects HR and also the trainer (but as soon I start pedal - fortiusant crashes)
Runing PY files 
 - on ZWIFT detects HR but not the trainner.
 - on ROUVY detects HR and also the trainer and app runs smothly.

Any hit why ZWIFT can't detect trainner?

_Originally posted by @E-NINA in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-605647331_

WouterJD, 2020-03-29:
Hi @E-NINA
Thanks for leaving your message, I have copied into an issue to be able to handle analysis here.
multiple issues...

"very old (GARMIN USB)" ExplorANT first lists all USB devices in the system. It would help to tell what output you get there. I expect a table like "manufacturer=, product=, vendor=, product=()". Please copy.

The software only expects (4104, "Suunto"), (4105, "Garmin"), (4100, "Older") but that table can be simply extended if needed.

If you start FortiusANT with -a -g flags; it will start with calibration. The display shows "* * * * * C A L I B R A T I N G * * * * *". You need to start to pedal so that the calibration actually starts (this is how 1932 Fortius works).

After calibration, FortiusANT will be in operational mode and ZWIFT / ROUVY will detect the trainer.
If you want to skip the calibration, use the -n flag.

It seems something with the calibration.

If it keeps crashing, please send logfile (-d 127)

E-NINA, 2020-03-29:
Hello,
The OLD USB is an Garmin stick that I have from a forerunner watch.

[image]

(if you need linux "lsusb" I can send only tomorrow)

Regarding the crash, I did make the Calibration and wait to run till the end (also try without make the calibration) with windows exes crashed with python run normaly. I attach the logs:

[FortiusANT.2020-03-29 20-32-53.log](https://github.com/WouterJD/FortiusANT/files/4399799/FortiusANT.2020-03-29.20-32-53.log)

I thinks this is the error mensage:

20:34:27,815: ReadFromDongle() returns: ["a4 03 40 02 01 03 e7", "a4 03 40 01 01 03 e4", "a4 09 4f 01 46 ff ff ff ff 02 36 01 90"]
20:34:27,820: SendToDongle() returns: ["a4 03 40 02 01 03 e7", "a4 03 40 01 01 03 e4", "a4 09 4f 01 46 ff ff ff ff 02 36 01 90"]
Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "FortiusAntGui.py", line 826, in OnClick_btnStart_Thread
  File "FortiusAnt.py", line 140, in callTacx2Dongle
  File "FortiusAnt.py", line 830, in Tacx2Dongle
  File "logfile.py", line 79, in HexSpace
TypeError: a bytes-like object is required, not 'bool'

Also for information the info for the working dongle:

20:30:28,853: ExplorANT started
20:30:28,853: --------------------
20:30:28,853: Dongles in the system:
20:30:28,876: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
20:30:28,877: manufacturer=Tacx̚VR, product=   VR-Interface, vendor=0x3561, product=0x1932(6450)
20:30:28,878: --------------------
20:30:29,407: Using Dynastream Innovations dongle
20:30:31,294: Pairing, press Ctrl-C to exit
. . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .

[image]

E-NINA, 2020-03-29:
LINUX info for the non working ANT dongle:

Bus 001 Device 007: ID 0fcf:1004 Dynastream Innovations, Inc. ANTUSB Stick

[  215.107857] usb_serial_simple 1-6:1.0: device disconnected
[  219.500895] usb 1-6: new full-speed USB device number 7 using xhci_hcd
[  219.651312] usb 1-6: New USB device found, idVendor=0fcf, idProduct=1004, bcdDevice= 3.00
[  219.651322] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  219.651328] usb 1-6: Product: Dynastream ANT2USB
[  219.651335] usb 1-6: Manufacturer: Silicon Labs
[  219.651340] usb 1-6: SerialNumber: 10034
[  219.701958] usbcore: registered new interface driver cp210x
[  219.701979] usbserial: USB Serial support registered for cp210x
[  219.702017] cp210x 1-6:1.0: cp210x converter detected
[  219.704433] usb 1-6: cp210x converter now attached to ttyUSB0

WouterJD, 2020-03-29:
I have good news: the crash is resolved in logfile.py
# 2020-03-24    Resolve crash when non-bytes input to HexSpace()

So download the newest software and that should work.

WouterJD, 2020-03-29:
In antDongle.py you find the list:
>       dongles = { (4104, "Suunto"), (4105, "Garmin"), (4100, "Older") }
4100 = 0x1004 and hence, should be found...

The list is derived from antifier so I never tested it (I do not have that dongle) but I believe antifier did.
So it must be possible to get that working as well.

You show ExplorANT's output "for the working dongle:" but I would be interested to see in the situation of the non-working dongle.
Please explain:
- your configuration
- number of computers
- which dongle where
- where do you run FortiusANT and where do you run ZWIFT

BUT first ... please download newest files!
Goodluck!

E-NINA, 2020-03-30:
Hello,

I've tested new windows version and can confirm that now I've no crash running ROUVY.
Regarding Zwift trainner still don't get recognized.

My configuration, for FortiusANT:
Laptop running windows 10 Pro
USB dongle:

PS C:\Users\sv028268\Documents\FortiusANT-master\pythoncode> py explorant.py
20:30:28,853: ExplorANT started
20:30:28,853: --------------------
20:30:28,853: Dongles in the system:
20:30:28,876: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
20:30:28,877: manufacturer=Tacx̚VR, product=   VR-Interface, vendor=0x3561, product=0x1932(6450)
20:30:28,878: --------------------
20:30:29,407: Using Dynastream Innovations dongle
20:30:31,294: Pairing, press Ctrl-C to exit
. . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .

For Zwift and ROUVY
Tablet Samsung S4 (embed ANT inside)

For the non working dongle I try investigate a litle bit more on that, so far I saw that that old ant versions works with serial UART -> cp210x. I think this is the part o the "antDongle.py" code lines 432...470. Ist this part of code active still?

, I don't thin

WouterJD, 2020-03-30:
Hi E-NINA

#1 Computer; Windows 10, ANT-Dongle, FortiusANT - works
#2 Tablet, Zwift and Rouvy - works ??
This should be a complete setup... (assuming #2 works)

What do you need the second dongle for and in what system?

antDongle line 433 "elif False" is kept only for reference; it will never be executed.
Might it be that somebody has a system that would require this code (as inherited from antifier) it is easy to re-activate.

E-NINA, 2020-03-30:
Hello,

#1 Computer; Windows 10, ANT-Dongle, FortiusANT - works  -> YES

#2 Tablet, Zwift and Rouvy - works ??  -> on Zwift don't detect trainer (but HR 57592 yes)
                                                                   on Rouvy detect Trainer (57591) and also the HR 57592

Regarding the second dongle I'm try with that just to make zwift work

WouterJD, 2020-03-30:
Well that's special; if Zwift detects one device but not the other and Rouvy detects both ... it's something with Zwift.

Perhaps the distance from tablet to PC?

E-NINA, 2020-03-30:
Isn't distance tablet and laptop are less than 10 cm away.
I was loking on zwift and in fact seems android version ANT isn't yet quite wll supported.
I've order a new usb ant dongle, anyway many thanks for your great support.
For now on I will ride on ROUVY :)

WouterJD, 2020-03-31:
Hi E-nina, I hope that new USB dongle will help. Success with ride on Rouvy.

I am trying to get rouvy work on my windows pc but no luck yet. Under investigation why rouvy does not detect ANT-dongle.
Do you link rouvy to strava? see you there:  https://www.strava.com/athletes/2885978

E-NINA, 2020-03-31:
I've just linked rouvy to strava. Ride 3 consecutive day without any issue b)

WouterJD, 2020-03-31:
Great. Strava upload from Rouvy seems slow, but let's see tomorrow!👍

WouterJD, 2020-04-06:
@E-Nina is succesfully training with results on strava

closed 2020-04-06 by WouterJD