WouterJD/FortiusANT #44: the gui does not start

V10l4t0r, 2020-04-07:
[Nuovo Documento di Microsoft Office Word.docx](https://github.com/WouterJD/FortiusANT/files/4442459/Nuovo.Documento.di.Microsoft.Office.Word.docx)

hello I managed to run Antifier which is an old version, with Fortius ANT I'm stuck here.

WouterJD, 2020-04-07:
Good day!

For some reason, I cannot download the attached pdf.
If the GUI does not start, you probably did not specify the -g flag.
Please refer to https://github.com/WouterJD/FortiusANT/wiki#command-line

Welcome and thanks for using FortiusANT; where are you from?

V10l4t0r, 2020-04-07:
Thanks, I'm Italian, I'll try to take a look

WouterJD, 2020-04-07:
Benvenuti sul progetto!

V10l4t0r, 2020-04-07:
Hi I was lucky because the roller was given to me only because they could not use it with zwift etc. Thanks to you and this project I can use my bike in this period of quarantine at home.

WouterJD, 2020-04-07:
I hope everything is going well with you and family; if you use strava -  please give me a ping there: https://www.strava.com/athletes/2885978

Buona pedalata!

WouterJD, 2020-04-07:
What Tacx do you use? Fortius/iMagic? 1902, 1904, 1932, 1942?

V10l4t0r, 2020-04-07:
> What Tacx do you use? Fortius/iMagic? 1902, 1904, 1932, 1942?

Tacx I-flow T1932 .
ps: 
thanks for adding me on strava

WouterJD, 2020-04-07:
Thanks. Great fun to see you using FortiusANT.
What training-software do you use?

V10l4t0r, 2020-04-07:
> Thanks. Great fun to see you using FortiusANT.
> What training-software do you use?

I'm trying zwift, in addition to what I wrote above I add that as sensors I use the cadence sensor and HR sensor of the garmin 530, today I will also put the speed sensor on the rear wheel to get more precise data

V10l4t0r, 2020-04-07:
> Good day!
> 
> If the GUI does not start, you probably did not specify the -g flag.
> Please refer to https://github.com/WouterJD/FortiusANT/wiki#command-line
> 
> Welcome and thanks for using FortiusANT; where are you from?

I'm looking at your guide, but under windows I can't put those commands, the problem is that by starting FortiusANT.exe the calibration never ends.

WouterJD, 2020-04-07:
Please check https://github.com/WouterJD/FortiusANT/blob/master/supportfiles/FortiusAnt.bat

Download that file and put it in the same folder where FortiusANT.exe is located.
The file contains some examples of a command-line.

For you "FortiusAnt.exe -g -a" will do the job.

When calibrating starts, you must move the pedal and then calibration actually starts.
Sorry that this is not guided well on the user-interface, I could improve there.

The Tacx then starts calibrating automatically and will stop when the calibration has become stable.

You can also stop calibration with the Cancel button on the console.

WouterJD, 2020-04-07:
If you do NOT want calibration, specify the -n flag on the command line

FortiusAnt.exe -g -a -n

V10l4t0r, 2020-04-07:
[image]

hello I managed to start the program in gui mode, the problem is to do the calibration otherwise during the zwift session the data is very altered.I just finished a session but I had to use the old Antifier version.I have attached the screen of the problema.Ciao

ps: I tried to do the calibration but the roller goes too hard

WouterJD, 2020-04-07:
I will check

WouterJD, 2020-04-07:
The special thing is that you have an iFlow which returns the same ID (1932) as the Fortius.
The Fortius supports calibration, but the iFlow seems not.

You can disable calibration by changeing the FortiusANT.bat file and specify the followig command:
`FortiusAnt.exe -g -a -n`
where -n stands for "no calibration"

V10l4t0r, 2020-04-08:
> The special thing is that you have an iFlow which returns the same ID (1932) as the Fortius.
> The Fortius supports calibration, but the iFlow seems not.
>
> You can disable calibration by changeing the [...]

Just to ask, how long does the calibration usually take? With the Antifier program it takes a few seconds; with this I cycled for about 5 minutes but it was very difficult to keep the watts fixed, already at 19/20 km / h it expressed more than 200 watts

WouterJD, 2020-04-08:
Hi gianluca, good morning.

As far as I know, the antifier program does not do a technical calibration but provides the runoff option.
The calibration as implemented in FortiusANT sends the calibrate command to the roller. The roller then starts spinning (at 20 km/h) and measures the resistance. As soon as the resistance is stable the calibration stops.
The steps are:
- start clibration by moving the pedals down (you do not need to cycle)
- the roller rotates autonomously (you do not need to do anything)
- FortiusANT stops and goes to normal mode

Please execute a test and send the logfile, adding the -d 127 option to the command line.

It may be that your setup (iFlow/1932) does not support this method and then the -n flag can be used to skip the calibration.

I am working on the crash in the runoff part of FortiusANT (already solved).
I am working on improvement on the messaging around calibration (under test now).

Please allow some time and I will upload a new FortiusANT.exe, to be expected these days.

Thanks for working together to improve FortiusANT for iFlow users.

WouterJD, 2020-04-08:
Version 2.5 is uploaded (exe only); please test

V10l4t0r, 2020-04-08:
> Hi gianluca, good morning.
>
> As far as I know, the antifier program does not do a technical calibration but provides the runoff option.
> The calibration as implemented in FortiusANT sends the [...]

thank you so much for dedicating your time to this project, I will try the new executable soon. Thanks for the explanation on the calibration.

V10l4t0r, 2020-04-08:
[FortiusANT.2020-04-08 14-34-17.log](https://github.com/WouterJD/FortiusANT/files/4451123/FortiusANT.2020-04-08.14-34-17.log)
Good afternoon, I performed the test with the new executable regarding the calibration without subsess. Now I will try to do a session on zwift skypping the calibration to see how it goes with the new exe

WouterJD, 2020-04-08:
From the logfile it seems that calibration does not work in your situation; the returned resistance is constant.

The calibration stops when the resistance remains constant for some time and will stop after 8 minutes. My Fortius returns a negative value during the calibration and it seems your system returns a positive value.
I will check this and detect the situation.

You may keep this issue open untill it's really solved; I prefer that the software fits your iFlow as well as the others.

Success with zwift tonite!

V10l4t0r, 2020-04-08:
[FortiusANT.2020-04-08 16-17-08.log](https://github.com/WouterJD/FortiusANT/files/4451639/FortiusANT.2020-04-08.16-17-08.log)
hello, just to be exact I did the calibration test again waiting 10 min. The roller does not move by itself. Probably this function is not there; in this regard, I remember that not even in the original tacx software did an automatic calibration. It was necessary to make a sort of runoff and depending on the data that came out on a graph it was necessary to tighten or loosen a lever. Now what I think is the real problem is that in the zwift program the power does not work well.

closed 2020-04-08 by V10l4t0r

BikeBeppe64, 2020-05-07:
Hi WJD, 
Beppe from Italy just to confirm GUI v2.6 works with T1932 as picture shows. As soon as possible I will do other tests.
Regards

[image]

WouterJD, 2020-05-07:
Buon "giorno"! Benvenuto. Da dove sei?
Saluti della Olanda!

BikeBeppe64, 2020-05-07:
Milano, Lombardia, Italy
Windows 10, Antifier & Garmin USB used
I-Flow VR T2250 new
Also Garmin Vector 2s (powermeter)

WouterJD, 2020-05-07:
Greate Beppe, see you later