WouterJD/FortiusANT #45: Supported for mac?

iainhay, 2020-04-10:
I'm looking to set up my tacx trainer on my mac - is there something similar for mac? Or can this be run on mac?

WouterJD, 2020-04-10:
Hi Ian, welcome in the FortiusANT community!

I think FortiusANT will run on Mac and I willl help you in case there are issues, even adjust code if needed. 

1. Install python 3
2. Download the code
3. Run the code and see where we get

What Tacx do you have?

iainhay, 2020-04-10:
Hey - awesome thanks. 

I have a i-Flow with a T1932 Head unit. I was setting up after many years put away and found out the tacx software has been discontinued and mine doesn't work with the desktop app.

So I have x2 ANT+ sticks which arrived so trying to get this set up (ideally with Zwift)

I'll install python 3 and see how I go - not used Python before, it's the FortiusAnt.py  I need to run once downloaded correct?

Huge thanks in advance!

WouterJD, 2020-04-10:
Indeed, read wiki for instructions
Succee

iainhay, 2020-04-10:
Hey!

So got the app working, it tries to calibrate, and I see this error message a few times in the terminal:

`18:40:13,839: ReadFromDongle read error: [Errno 60] Operation timed out`

And then the error it stops. I see all the dials as if it is sending/receiving.

[image]

Is calibration meant to automatically turn the trainer wheel to 20km p/hr? Been so long since I've used it couldn't remember if that was possible?

iainhay, 2020-04-10:
Also - if I start without calibrating I get the same error repeatedly (it doesn't stop)

`18:40:13,839: ReadFromDongle read error: [Errno 60] Operation timed out`

WouterJD, 2020-04-10:
There someone else with an i-Flow/1932 and it seems that calibration does not work
Use the -n flag (no calibration) and see what happens

iainhay, 2020-04-10:
Yes if I start without calibrating I get the same error repeatedly (it doesn't stop)

`18:40:13,839: ReadFromDongle read error: [Errno 60] Operation timed out`

[image]

WouterJD, 2020-04-10:
run with -d8 flag and send logfile please

iainhay, 2020-04-10:
Sure thing! I noticed someone had a similar issue on mac with antifier but no solution was every posted.

[FortiusANT.2020-04-10 21-06-08.log](https://github.com/WouterJD/FortiusANT/files/4463434/FortiusANT.2020-04-10.21-06-08.log)

WouterJD, 2020-04-11:
I have checked the logfile:

From 21:06:09 ... 21:06:11 message "[Errno 60] Operation timed out"
Then commands are sent (you probably pressed Start)
From 21:06:11 ... 21:06:33 als message "[Errno 60] Operation timed out"

But then it appears normal operation; FortiusANT sends command and receives an answer.
From 21:06:33 ... 21:06:53

Note that it is all within a minute.

Some questions:
1. Do the messages continue to appear?
2. What happens if you start cycling
3. Could it be that your system needs some time to start?

Sorry that I cannot check myself...please try ans see what happens!
Succes

iainhay, 2020-04-12:
1. Do the messages continue to appear? - Yes, repeatedly
2. What happens if you start cycling - with the app open I see the speed and power dials (I have a cadence too but not connected)
3. Could it be that your system needs some time to start? - Not sure what you mean?

I have x2 ANT+ dongles plugged into the mac, along with the USB from the trainer

WouterJD, 2020-04-12:
The provided logfile shows that the messages stop (see my message from yesterday) 
3. It may be that yoour Tacx needs some time to start and that messages disappear later-on

Provide logfile over more time than just one minute, please. Let's say 10 minutes at least.

iainhay, 2020-04-12:
I can see the speed etc immediately when I start pedaling, so the trainer sending the data doesn't seem to be the issue.

10min log file attached - the previous one was a minute because I only ran it for a minute (as it's just repeating over and over) - it just repeats, to end I clck stop (the errors still come through and the close the app and it exist in the terminal.

Thanks!

[FortiusANT.2020-04-12 11-34-58.log](https://github.com/WouterJD/FortiusANT/files/4466246/FortiusANT.2020-04-12.11-34-58.log)

WouterJD, 2020-04-12:
Sorry * 100

ReadFromDongle has nothing to do with the Tacx trainer, but is generated at the ANT usbDongle side. (As the error message says).

Check in antDongle.py
```
    except Exception as e:
        if "timeout error" in str(e):
            if debug.on(debug.Data1):
                logfile.Write ("Dongle    timeout")
            pass
        else:
            logfile.Write ("ReadFromDongle read error: " + str(e))

```

The code expects "timeout error" in the error-message, but (on your system)  the message is "ReadFromDongle read error: [Errno 60] Operation timed out" and hence the error is given and not ignored as it should be. Note that the exception handling does not alter any data but only logs the error message.

You may simple ignore the message or replace "timeout error" with "timeout" and see what happens; behaviour should be the same in both cases. 

I do not like this construction (cheling text in message) and will check how to improve.

WouterJD, 2020-04-12:
The correct code would be:
    except TimeoutError:
        pass
    except Exception as e:
        logfile.Write ("ReadFromDongle read error: " + str(e))

If you don't mind checking that, then it's confirmed on your platform and I will implement in the baseline. Thanks

iainhay, 2020-04-12:
So replacing this:

```
except Exception as e:
        if "timeout error" in str(e):
            if debug.on(debug.Data1):
                logfile.Write ("Dongle    timeout")
            pass
        else:
            logfile.Write ("ReadFromDongle read error: " + str(e))
```

with:

```
except TimeoutError:
      pass
    except Exception as e:
      logfile.Write ("ReadFromDongle read error: " + str(e))
```

Still gives me:

`13:06:22,863: ReadFromDongle read error: [Errno 60] Operation timed out`

Repeated.

I also open Zwift with the app running, I can connect via Power Source and ANT+, but it receives no data (e.g. I start peddling but don't move or see any power). In the app I still see Power and Speed - but this is coming via the USB presumably.

iainhay, 2020-04-12:
If I change to this:

```
except Exception as e:
        if "timed out" in str(e):
            if debug.on(debug.Data1):
                logfile.Write ("Dongle    timeout")
            pass
        else:
            logfile.Write ("ReadFromDongle read error: " + str(e))
```

And just check against 'timed out' then the error goes, but I'm still not receiving any data in Swift - no power etc. So it is like the receiving ANT+ dongle is not receiving the data?

WouterJD, 2020-04-12:
OK, you're doing good.

Please run FortiusANT with -d4 this will provide the ANT logging.
Sorry for keeping you budy on this sunny easter day :-)

https://github.com/WouterJD/FortiusANT/wiki#values

iainhay, 2020-04-12:
Okay cool - attached is the log file (kept on for 10mins).

No worries - thanks for helping me with this!

[FortiusANT.2020-04-12 13-54-21.log](https://github.com/WouterJD/FortiusANT/files/4466448/FortiusANT.2020-04-12.13-54-21.log)

WouterJD, 2020-04-12:
OK logfile looks good, but indeed - no commands are received from Zwift.

Did you pair Zwift with FortiusANT?
Note that a master (the trainer, FortiusANT) is unaware whether a slave (Zwift, ...) is looking at or pairing [with the master]..

Please send screen-shot of pairing-page, something like:

[image]

iainhay, 2020-04-12:
Okay great.

So here is what I am doing - just listing in case I'm missing something:

1. Plugging in the USB from my trainer
2. Plugging in both ANT+ dongles
3. Running the FortiusANT app (skipping calibration)
4. Opening Zwift

Here is what I see in Zwift:

[image]

(This is the only available device)

WouterJD, 2020-04-12:
Checking the screen-print from FortiusANT (Trainer detected, Using Dynastream...) the hardware is found correctly and I would expect things to work. The logfile shows that data is transmitted.

What happens when you press SEARCH on cadence and controllable?
Both should should be paired with FortiusANT.

iainhay, 2020-04-12:
No nothing - I just get the searching screen..

Do I need to do anything to the ANT+ dongles? I read about having master/slave etc?

iainhay, 2020-04-12:
For some reason the Heart Rate one picks up an ANT+ device and connects, but not Cadence or Controllable. Power Source connects but receives not data

iainhay, 2020-04-12:
Also, if I simulate (with -s) I get the same results

WouterJD, 2020-04-12:
> No nothing - I just get the searching screen..
> 
> Do I need to do anything to the ANT+ dongles? I read about having master/slave etc?

master/slave is decided by software; a trainer is master, Zwift is slave

WouterJD, 2020-04-12:
> Also, if I simulate (with -s) I get the same results

-s simulates a trainer, this comes handy in this case - you do not need to cycle to test Zwift.
Good idea to use it

WouterJD, 2020-04-12:
> For some reason the Heart Rate one picks up an ANT+ device and connects, but not Cadence or Controllable. Power Source connects but receives not data

What is the deviceID of the heart rate monitor?

iainhay, 2020-04-12:
So it's simulating now:

[image]

And I can only connect via ANT+ to a power source and HRM in Zwift:

[image]

HRM ID:

[image]

iainhay, 2020-04-12:
But no watts or BPM received either

WouterJD, 2020-04-12:
plse read https://github.com/WouterJD/FortiusANT/issues/42

It sounds like similar and that case was never explained. It seems another dongle solved the problem. Anyway: FortiusANT was not changed to resolve...

What dongles / configuration do you use?
Where do you run Zwift?
Perhaps try Trainer Road or Rouvy

I have confidence it will work

iainhay, 2020-04-12:
It does seem to be something with Zwift now - I will try Rouvy tomorrow and let you know.

Thank you so much for all your help!

WouterJD, 2020-04-13:
👍

iainhay, 2020-04-13:
So in Rouvy I get the same, when searching for a trainer I see only a HRM :(

[image]

I had to try on my phone because ANT+ is not supported on mac by Rouvy. But it seems that it's not just Zwift.

Here is what I get in ExplorAnt.py:

```
06:29:28,501: ExplorANT started
06:29:28,502: --------------------
06:29:28,502: Dongles in the system:
06:29:28,514: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
06:29:28,515: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
06:29:28,516: manufacturer=GenesysLogic, product=     USB2.1 Hub, vendor= 0x5e3, product= 0x610(1552)
06:29:28,516: manufacturer=Realtek, product=USB 10/100/1000 LAN, vendor= 0xbda, product=0x8153(33107)
06:29:28,517: manufacturer=GenesysLogic, product=     USB3.1 Hub, vendor= 0x5e3, product= 0x626(1574)
06:29:28,519: manufacturer=Apple Inc., product=Apple Internal Keyboard / Trackpad, vendor= 0x5ac, product= 0x340(832)
06:29:28,520: manufacturer=  Apple, product=           None, vendor= 0x5ac, product=0x8104(33028)
06:29:28,520: --------------------
06:29:29,060: Using Dynastream Innovations dongle
06:29:31,157: Pairing, press Ctrl-C to exit
. . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . 
06:29:41,268: Pairing stopped
06:29:41,269: --------------------
06:29:42,609: Listening, press Ctrl-C to exit
06:29:43,628: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:44,646: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:45,658: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:46,678: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:47,693: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:48,701: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:49,719: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:50,731: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:51,750: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:52,762: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
06:29:53,777: HRM#= 0 hr= -1 FE-C#= 0 Speed=  -1 Cadence= -1 Power= -1 hr= -1
```
Does this look right?

My set up is:

Macbook Pro 2019
x2 ANT+ Dongles
Tacx iFlow w/ 1932 Head Unit

WouterJD, 2020-04-13:
Is FortiusANT running at the same time?

iainhay, 2020-04-13:
So when running `python3 FortiusAnt.py -g -a -n -s` and then running ExplorAnt.py I get:

```
07:31:16,155: ExplorANT started
07:31:16,155: --------------------
07:31:16,155: Dongles in the system:
07:31:16,170: manufacturer=Realtek, product=USB 10/100/1000 LAN, vendor= 0xbda, product=0x8153(33107)
07:31:16,171: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
07:31:16,172: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
07:31:16,173: manufacturer=GenesysLogic, product=     USB2.1 Hub, vendor= 0x5e3, product= 0x610(1552)
07:31:16,174: manufacturer=GenesysLogic, product=     USB3.1 Hub, vendor= 0x5e3, product= 0x626(1574)
07:31:16,175: manufacturer=Apple Inc., product=Apple Internal Keyboard / Trackpad, vendor= 0x5ac, product= 0x340(832)
07:31:16,177: manufacturer=  Apple, product=           None, vendor= 0x5ac, product=0x8104(33028)
07:31:16,178: --------------------
Exception: [Errno 19] No such device (it may have been disconnected)
07:31:16,187: GetDongle: [Errno 19] No such device (it may have been disconnected)
07:31:16,187: We're done
07:31:16,187: --------------------
```

WouterJD, 2020-04-13:
ExplorANT "listens" on the ANT-network and shows what master-devices broadcast data.

There are two dongles; one used by FortiusANT and the second by ExplorANT.
No other programs should be running, because then no dongle is free.

If problem persists, run ExplorANT with -d5 flag and send logfile.

iainhay, 2020-04-13:
Here is the log file:

[ExplorANT.2020-04-13 07-48-23.log](https://github.com/WouterJD/FortiusANT/files/4468545/ExplorANT.2020-04-13.07-48-23.log)

WouterJD, 2020-04-13:
For some reason, the dongle cannot be used.
Since 2020-03-09 multiple dongles of the same type should be supported; before always the first dongle was taken, which would be in use by FortiusANT.

To get more info in the logfile, run ExplorANT -d7 so that functions are also logged.

M4M4R2, 2020-04-13:
Hi guys!
I'm happy to see some of you are managing to do what i've been trying to for month if not years!
Ok, i've been trying to install Antifier on my Raspberry pi 1 and it unfortunately didn't work properly as the ANT dongle could not be detected for some reason... Now I discover your work and I really find it great! Much like iainhay I'm a mac user very naive on programming with Python. I downloaded the Python code for FortiusANT but when running the module it says "ModuleNotFoundError: No module named 'numpy'". Would you please describe the installation process you used on mac???
THX

WouterJD, 2020-04-13:
Just to be sure, I ran a `FortiusANT -g -a -s` test on Windows 10, without the Tacx trainer so using simulation mode.

Then `ExplorAnt.py -d1`; minimum logfile which produces the following result, which is as expected and immediatly shows what output is expected:
The dongle is initiated
Then a listening loop is done to check what devices are present (pairing) and HRM and FE is found. 57592 is the deviceID of the HRM like what Zwift shows you.
And then a listening loop is done to listen what the masters say and correct HRM- and FE-values are received.

> 13:17:11,894: ExplorANT started
> 13:17:11,894: -d 1 (0b1)
> 13:17:11,894: -D -1 (-0x1)
> 13:17:11,894: -H -1 (-0x1)
> 13:17:11,894: -F -1 (-0x1)
> 13:17:11,894: --------------------
> 13:17:11,894: Dongles in the system:
> 13:17:11,957: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
> 13:17:11,957: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
> 13:17:11,957: --------------------
> 13:17:12,499: Using Dynastream Innovations dongle
> 13:17:14,594: Pairing, press Ctrl-C to exit
> 13:17:24,784: Pairing stopped
> 13:17:24,784: HRM discovered on channel=0, number=57592 typeID=120 TrType=120
> 13:17:24,785:  FE discovered on channel=1, number=57591 typeID= 17 TrType= 17
> 13:17:24,785: --------------------
> 13:17:26,307: HRM slave channel 5 opened; listening to device 57592
> 13:17:26,495: FE  slave channel 4 opened; listening to device 57591
> 13:17:26,495: Listening, press Ctrl-C to exit
> 13:17:27,503: HRM#= 4 hr= 90 FE-C#= 3 Speed=35.7 Cadence=100 Power= 98 hr= 86
> 13:17:28,508: HRM#= 8 hr= 87 FE-C#= 7 Speed=35.7 Cadence=102 Power=103 hr= 87
> 13:17:29,384: Listening stopped
> 13:17:29,953: We're done
> 13:17:29,953: --------------------
> 

`ExplorAnt.py -d3`; produces the following **additional** result:
> ...
> 13:21:50,994: GetDongle - Check for dongle 4100 Older
> 13:21:50,994: GetDongle - Check for dongle 4105 Garmin
> 13:21:51,010: GetDongle - Dongle found: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
> 13:21:51,010: GetDongle - Send reset string to dongle
> 13:21:51,010: GetDongle - Dongle found: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
> 13:21:51,010: GetDongle - Send reset string to dongle
> 13:21:51,512: GetDongle - Read answer
> 13:21:51,547: ReadFromDongle() returns: ["a4 01 6f 00 ca"]
> 13:21:51,547: GetDongle - Check for an ANT+ reply
> 13:21:51,547: GetDongle() returns: Using Dynastream Innovations dongle
> 13:21:51,547: Using Dynastream Innovations dongle
> ...

The software uses `usb.core.find()` which returns two dongles with type 4105; the first one gives an error on the "Send reset string" (because it's in use by FortiusANT), on the second device the "Send reset string" goes well, an answer is received and the software proceeds.
By the way: FortiusANT and ExplorANT use the exact same loop.

It COULD mean that, when I think I use the second dongle, that actually the first is tried and hence is in use.

iainhay, 2020-04-13:
Log files for -d1, -d3 and -d7

[ExplorANT.2020-04-13 13-45-27.log](https://github.com/WouterJD/FortiusANT/files/4469745/ExplorANT.2020-04-13.13-45-27.log)
[ExplorANT.2020-04-13 13-46-26.log](https://github.com/WouterJD/FortiusANT/files/4469746/ExplorANT.2020-04-13.13-46-26.log)
[ExplorANT.2020-04-13 13-45-55.log](https://github.com/WouterJD/FortiusANT/files/4469747/ExplorANT.2020-04-13.13-45-55.log)

WouterJD, 2020-04-13:
There is another error now, as if there are no dongles in your system `Could not find ANT-dongle` where previously you reported  `GetDongle: [Errno 19] No such device (it may have been disconnected)`.

In this case, I would power-down the system, take out the dongles (so that they are powered down as well), reboot and see whether there is any difference.

WouterJD, 2020-04-13:
Please download logfile.py and antDongle.py and retry test with` ExplorAnt -d3`

The dongle that is tried is printed to the logfile, let's see whether the second attempt indeed tries the second dongle.

WouterJD, 2020-04-13:
> Hi guys!
> I'm happy to see some of you are managing to do what i've been trying to for month if not years!
> Ok, i've been trying to install Antifier on my Raspberry pi 1 and it unfortunately didn't [...]

`pip install -r requirements.txt` should do the job, please try

iainhay, 2020-04-14:
Log file attached. Still seeing a similar error - I've checked and my mac recognises both dongles as being plugged in. I also restarted etc as mentioned.

[ExplorANT.2020-04-14 08-15-35.log](https://github.com/WouterJD/FortiusANT/files/4474005/ExplorANT.2020-04-14.08-15-35.log)

M4M4R2, 2020-04-14:
Hi, 
Python shell doesn't accept pip install : "SyntaxError"! I didn't see that one since coding in Basic in the early eighties!
Here is the log : 
Python 3.8.2 (v3.8.2:7b3ab5921f, Feb 24 2020, 17:52:18) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license()" for more information.
>>> 
======= RESTART: /Applications/FortiusANT-master/pythoncode/FortiusAnt.py ======
Traceback (most recent call last):
  File "/Applications/FortiusANT-master/pythoncode/FortiusAnt.py", line 119, in <module>
    import numpy
ModuleNotFoundError: No module named 'numpy'
>>> pip install -r requirements.txt
SyntaxError: invalid syntax

M4M4R2, 2020-04-14:
I tried "python -m pip install". Didn't work neither! Now I'm confused!

WouterJD, 2020-04-14:
> Log file attached. Still seeing a similar error - I've checked and my mac recognises both dongles as being plugged in. I also restarted etc as mentioned.
> 

Please download antDongle.py and re-test; I have changed the error-handling in the module.
It seems "in use" works different than under Windows/Linux

WouterJD, 2020-04-14:
> I tried "python -m pip install". Didn't work neither! Now I'm confused!

I do not have a Mac, so cannot test for you.
"Didn't work neither" is not very clear, what messages do you get?

Perhaps [google](https://www.google.com/search?rlz=1C1GCEU_nlNL872NL872&sxsrf=ALeKk01xs2vWNR5dpUXJX1CjRlST-ZeHnQ%3A1586853714916&ei=UneVXtunN8KMkgXS-YEQ&q=pip+install+-r+requirements.txt+SyntaxError%3A+invalid+syntax+macintosh&oq=pip+install+-r+requirements.txt+SyntaxError%3A+invalid+syntax+macintosh&gs_lcp=CgZwc3ktYWIQAzoECAAQR0oJCBcSBTEyLTMxSgkIGBIFMTItMTJQhzVY8kNgxkZoAHACeACAASmIAeUCkgECMTCYAQCgAQGqAQdnd3Mtd2l6&sclient=psy-ab&ved=0ahUKEwjbmPPfwufoAhVChqQKHdJ8AAIQ4dUDCAw&uact=5) helps...

M4M4R2, 2020-04-14:
Sorry for that! 
python -m pip install -r requirements.txt
SyntaxError: invalid syntax
>>>

iainhay, 2020-04-14:
@M4M4R2 

Try: `python3 -mpip install` - then what you are trying to install

e.g. `python3 -mpip install pyusb`

Note - I use `python3` to use python v3 - my mac default is 2.7 (I think)

iainhay, 2020-04-14:
@WouterJD 

Log file attached.

[ExplorANT.2020-04-14 09-36-18.log](https://github.com/WouterJD/FortiusANT/files/4474463/ExplorANT.2020-04-14.09-36-18.log)

M4M4R2, 2020-04-14:
@iainhay 2.7 version for MacOs and 3.8 installed for me! The command you provided seems to be understood by the shell at least! I added an unnecessary space between -m and pip! Though  when prompting pip install requirements.txt I get : 
~ % python3 -mpip install requirements.txt
Collecting requirements.txt
  ERROR: Could not find a version that satisfies the requirement requirements.txt (from versions: none)
ERROR: No matching distribution found for requirements.txt
WARNING: You are using pip version 19.2.3, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

iainhay, 2020-04-14:
@M4M4R2 

Don't use the requirements.txt, try installing the things within one by one, e.g.

`python3 -mpip install keyboard`
`python3 -mpip install pyusb`
`python3 -mpip install pygame`
`python3 -mpip install pynput`
`python3 -mpip install wxPython`
`python3 -mpip install wheel`

Note you probably don't need all of these (some will already by installed), I think I only needed the following: 

`python3 -mpip install pyusb`
`python3 -mpip install wxPython`

WouterJD, 2020-04-14:
> python -m pip install -r requirements.txt
> SyntaxError: invalid syntax
> 
Please check whether you have python 3 installed; refer [this](https://github.com/WouterJD/FortiusANT/wiki#python-version)

WouterJD, 2020-04-14:
I have reviewed and tested the error handling in antDongle.py GetDongle() on windows.
Please download and test
Version is 2020-04-14b which should be visible in logfile.

Succes!

iainhay, 2020-04-14:
Attached log file - which I think is working, I think it tries one dongle which doesn't work (because this is being used by FortiusANT) then the next one and then is sending and receiving) - is that correct?

[ExplorANT.2020-04-14 13-19-36.log](https://github.com/WouterJD/FortiusANT/files/4475483/ExplorANT.2020-04-14.13-19-36.log)

I opened Zwift but still only see a HRM - same with Rouvy too :(

WouterJD, 2020-04-14:
Thank you for persistance!

This is what happens on my system:

```
13:51:53,301: ExplorANT started
13:51:53,301: -d 1 (0b1)
13:51:53,302: -D -1 (-0x1)
13:51:53,302: -H -1 (-0x1)
13:51:53,302: -F -1 (-0x1)
13:51:53,303: --------------------
13:51:53,303: Dongles in the system:
13:51:53,351: manufacturer=Suunto, product=Movestick mini, vendor= 0xfcf, product=0x1008(4104)
13:51:53,355: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
13:51:53,358: --------------------
13:51:53,920: Using Dynastream Innovations dongle
13:51:55,921: Pairing, press Ctrl-C to exit
13:52:06,001: Pairing stopped
13:52:06,001: HRM discovered on channel=0, number=57592 typeID=120 TrType=120
13:52:06,002:  FE discovered on channel=1, number=57591 typeID= 17 TrType= 17
13:52:06,002: --------------------
13:52:07,428: HRM slave channel 5 opened; listening to device 57592
13:52:07,588: FE  slave channel 4 opened; listening to device 57591
13:52:07,588: Listening, press Ctrl-C to exit
13:52:08,589: HRM#= 3 hr= 93 FE-C#= 3 Speed=34.3 Cadence= 98 Power= 99 hr= 91
13:52:09,589: HRM#= 7 hr= 91 FE-C#= 7 Speed=34.6 Cadence=101 Power= 98 hr= 91
13:52:10,589: HRM#=11 hr= 91 FE-C#= 9 Speed=34.6 Cadence= 98 Power= 98 hr= 91
13:52:11,458: Listening stopped
13:52:12,001: We're done
13:52:12,001: --------------------

```

It appears that the HRM from FortiusANT uses all channels.
Please restart with -H 0 which causes that no HRM-master is broadcasted and let me know the result.

iainhay, 2020-04-14:
Log attached.

[ExplorANT.2020-04-14 14-01-40.log](https://github.com/WouterJD/FortiusANT/files/4475797/ExplorANT.2020-04-14.14-01-40.log)

Zwift doesn't find the HRM anymore but nothing else. It finds a Power Source as before but no watts coming through.

[image]

[85 more replies]

closed 2020-04-27 by WouterJD