WouterJD/FortiusANT #47: Anyone have this working on Linux?

mikeherriman, 2020-04-15:
I have downloaded all of the repository files and when i try to run the FortiusANT.py program I receive the following:
Traceback (most recent call last):
   File "FortiusAnt.py", line 119, in <module>
       import numpy
ModuleNotFoundError: No module named 'numpy'

I am running Lubuntu 18.04.
I have 1 Ant 2 Stick and 1932 Head Unit.

Thanks
Mike

WouterJD, 2020-04-15:
It sure works under Linux
Check your pythoninstallation: https://github.com/WouterJD/FortiusANT/wiki#python-version

mikeherriman, 2020-04-15:
Ok thanks.  Missed running the pip install requirements.

That is now complete, and I can execute:
‘Python3 FortiusAntGui.py’ successfully
The GUI is loaded, with Tacx Trainer displayed along with the Ant+ USB dongle.

However, upon startup the Power, Speed and Cadence all begin to show numbers automatically....almost like a demo mode?

The Locate HW, Runoff, Start buttons are greyed out, but Stop button is available.

In the terminal window there are a bunch of messages displayed:
dbind-WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ally.Bus was not provided by any .service files

OnClick_btnLocateHW()
callLocateHW not defined by application class
OnClick_btnStart()
OnClick_btnStart_Thread()
callTacx2Dongle not defined by application class
OnClick_btnStop()
callIdleFunction not defined by application class

callIdleFunction not defined by application class

WouterJD, 2020-04-15:
It sure is a demo-mode to test module FortiusAntGui.
The correct program to start is FortiusAnt: https://github.com/WouterJD/FortiusANT/wiki#command-line

mikeherriman, 2020-04-15:
Thanks JD.

I think i am almost there.  When I click the Locate HW button on the GUI I see:
GetTrainer - Trainer found: 0x1932
GetDongle - ANT dongle in use

My terminal window displays:
GetDongle - Exception: [Errno 16] Resource busy
GetTrainer - Trainer found: 0x1932

I am relatively new to linux, but my googling tells me that the USB Ant stick is being used/locked by another service or application? I dont have anything else running in the OS that I know of.  
Any suggestions?

WouterJD, 2020-04-15:
PLease check whether you have the newest software downloaded, some quality improvements were made for MacOS the last days especially on the error-handling around the ANTdongle.

Please load the latest software and retry. Sorry for the inconvenience.

mikeherriman, 2020-04-15:
I just downloaded the latest antDongle.py and logfile.py.

now when i execute:
sudo python3 FortiusAnt.py
I receive:
Traceback (most recent call last):
  File "FortiusAnt.py", line 133, in <module>
    import antDongle         as ant
  File "/home/mherriman/Tacx/FortiusANT/pythoncode/antDongle.py", line 7
    <!DOCTYPE html>

WouterJD, 2020-04-15:
I would expect an error to occur at line 7

Two questions:
- Is this all you get
- Please check antDongly.py with any text-editor and copy what is at line 7 ...

WouterJD, 2020-04-15:
If the problem persists, though, please try to reset the dongle
- Remove from USB-port and re-insert; the easy way
- Power down system and restart; the longer route
Experience shows that now-and-then, especially when testing a lot, the dongle is not released properly; perhaps because the software did not do the right things before quitting.

closed 2020-04-17 by WouterJD