WouterJD/FortiusANT #49: ANT FE-C page 54 not supported

mattipee, 2020-04-17:
I don't have access to protocol specs, but I have an issue.

Trying to perform either ERG or Slope workouts in GoldenCheetah, I am not receiving any power or slope values into FortiusANT, whereas I do see these values from Zwift.

Looking into GoldenCheetah source, it uses some bit-field values in ANT/ANT.h:
#define FITNESS_EQUIPMENT_RESIST_MODE_CAPABILITY    0x01
#define FITNESS_EQUIPMENT_POWER_MODE_CAPABILITY     0x02
#define FITNESS_EQUIPMENT_SIMUL_MODE_CAPABILITY     0x04

from:
#define FITNESS_EQUIPMENT_TRAINER_CAPABILITIES_PAGE 0x36

0x36 is 54 in decimal.

I can see that in FortiusANT I am getting request(s) for this page, but it is unhandled.

Without a response to this request listing the device capabilities, specifically the MODEs above, GoldenCheetah will not send power/slope values to FortiusANT.

The message that FortiusANT should reply with is read by GoldenCheetah as follows:

                case FITNESS_EQUIPMENT_TRAINER_CAPABILITIES_PAGE:
                     fecMaxResistance         = message[9];
                     fecMaxResistance        |= (message[10]) << 8;
                     fecCapabilities          = message[11];
                     break;

WouterJD, 2020-04-17:
> I don't have access to protocol specs, but I have an issue.

It did cost me some time, but here is the link/documents...
Refer:    https://www.thisisant.com/developer/resources/downloads#documents_tab
ANT:     D00000652_ANT_Message_Protocol_and_Usage_Rev_5.1.pdf
trainer: D000001231_-_ANT+_Device_Profile_-_Fitness_Equipment_-_Rev_5.0_(6).pdf
hrm:     D00000693_-_ANT+_Device_Profile_-_Heart_Rate_Rev_2.1.pdf

see trainer page 60.

Indeed, page54 is not implemented.
It should have been done (it is a mandatory page) but no other software issued the request untill now, which makes it difficult to test.

If you are OK to test; I will seek some time to implement.

WouterJD, 2020-04-17:
Could you please send the logfile, preferably using the most recent .py modules.

mattipee, 2020-04-17:
I have the first document, but the other two are ANT+ Device Profiles and requires ANT+ Adopter Account membership. Is that free?

I will submit a PR against this issue soon, good practice and saving you dev effort, just review.

I have it implemented locally, and it works. I have successfully loaded a .gpx into GoldenCheetah and virtually ridden the route in slope mode. I'll test power mode also.

WouterJD, 2020-04-17:
Well, download the newest files and give it a try

mattipee, 2020-04-17:
That was quick. :)

# Capabilities; bit 0 = Basic, bit 1 = Target bit 3 = Simulation
info = ant.msgPage54_FE_Capabilities(ant.channel_FE, 0xff, 0xff, 0xff, 0xff, 1000, 0x03)

#define FITNESS_EQUIPMENT_RESIST_MODE_CAPABILITY 0x01
#define FITNESS_EQUIPMENT_POWER_MODE_CAPABILITY 0x02
#define FITNESS_EQUIPMENT_SIMUL_MODE_CAPABILITY 0x04

Just unpicking a little, and you have the spec so I hope you can answer this.

GoldenCheetah uses bitmasks to parse the capabilities byte... checking for bit0 being set (ie 0x01) = RESIST, bit1 set (ie 0x02) = POWER and bit2 set (ie 0x04) = SIMUL.

The value of 0x03 that you send in your recent change will signal to GoldenCheetah that the trainer supports RESIST_MODE + POWER_MODE, but not SIMUL_MODE, therefore will not work with slope-based workouts.

Is you change, or GoldenCheetah's implementation the correct interpretation? Not being able to see the wording/table of the spec, I'm stuck.

WouterJD, 2020-04-17:
OK, misinterpreted simulation mode; but I understand now.
Please test the module and I will update 0x03 into 0x07

[image]

mattipee, 2020-04-17:
In your comments, should read bit **2** = Simulation
Will test later today. Thanks!

WouterJD, 2020-04-17:
:-)

WouterJD, 2020-04-17:
By the way; give an eye to Rouvy. Upload GPX, then ride on the map. If you like - you can smoothe the trip :-)

mattipee, 2020-04-17:
I can confirm that, with the change in your reply above (not yet committed?), I get both Grade and Power modes working correctly in GoldenCheetah. Therefore page 54 is being responded to correctly.

Thanks! :D

WouterJD, 2020-04-17:
No was not yet committed but will be soon; issue solved

closed 2020-04-17 by WouterJD

mattipee, 2020-05-03:
https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-623156955
"since the addition of Page 54 support, FortiusAnt 2.6 is now compatible with Tacx Desktop Application"