WouterJD/FortiusANT #50: Virtual flywheel weight configurable?

mattipee, 2020-04-17:
I need to do a bit more testing, but I thought I'd raise the issue now for comments.

I noticed in code that when setting resistance in grade mode, the virtual flywheel value you send is 0x0a.
```
            Weight = 0x0a                                   # weight=0x0a is a good fly-wheel value
                                                            # UserAndBikeWeight is not used!        
                                                            #       an 100kg flywheels gives undesired behaviour :-)
```

During a GoldenCheetah test ride this afternoon, there were a number of minor slope changes which felt unnaturally sharp. I changed the value to reflect body weight and the result felt more natural. Interested to know what your undesired behaviour was. I suppose in the end, it may just be preference.

Perhaps a configuration parameter could allow the user to choose? I understand body/bike weight can be provided over ANT too, though perhaps can't assume it will be, or may wish to override it anyway for testing.

Eg:
 - use ANT-provided value
 - set as fixed value (default 0x0a)

WouterJD, 2020-04-17:
it's based upon totalreverse interface specs
Your tests are appreciated

What tacx do you use?

mattipee, 2020-04-17:
I have a Fortius.

It seems like a good candidate for configuration, independent of more subjective testing.

WouterJD, 2020-04-17:
What I understood from it, is that it is the flywheel weight - influencing the responsiveness: a heavyfly wheel would result in low responsiveness.
Implementing bodyweight did not give the desired result, as I see from the comment in the code.

I would be much obliged getting your view on this value.
Please note that I am not a Tacx- or other trainer specialist and it's a journey to get things working; your help is highly appreciated!

As always, curious who I chat with; where are you from, are you on strava as well?

mattipee, 2020-04-17:
I'm not a trainer specialist either. Just an overweight hobby cyclist with some coding experience. I'm in the UK.

While the equations for the power required to maintain a given speed on a slope account for gravity's affect on weight, they don't account for the power required to overcome inertia.

Being rather large, it takes a bit more to get me going, and a bit more to stop me.  So if I hit an incline on the road, I've got my considerable moving mass working in my favour - it will carry me up a bit - the rate of change of resistance is low, I can gradually put on the power as gravity starts to slow me down. If I weighed nothing, then the slope would immediately require me to put more power in to maintain speed - a fairly rapid onset of resistance. This is my intuitive understanding, and what I think I feel on the bike when changing the flywheel weight. I could be wrong.

So being a heavier rider, low responsiveness is what I'm used to?

mattipee, 2020-04-17:
[image]

Test was 0.1km of 0%, 2% and -2%, repeat.

First two cycles were with flywheel turned off (0x0a). Second two cycles were using UserAndBikeWeight (85 - a little light but it'll do for the test).

At first glance, you can see the "sqaure" nature of the first two , and the "slightly rounder" nature of the second two. And you can identify which the 5th is.

With no flywheel, focussing on the speed graph, you get a flat section, then a constant velocity uphill section, then a quick ramp up to terminal velocity on the downhill, then quickly back to the flat, drop speed slightly to uphill, then whizz back to top speed (freewheeling).

With 85kg flywheel, you can see that the uphill speed drops from the flat-speed a little more gradually, as my heft carries me forward. Then when cresting the hill, it takes a little more time to accelerate the mighty me downhill and when I reach the bottom, I'm rolling out onto the flat and slowly decreasing speed.

A wordy explanation, I hope you follow the description - otherwise just interpret the graphs. It's totally as I would expect. The virtual flywheel is important for simulating real-world feel, I think.

WouterJD, 2020-04-17:
Interesting study!
What does Golden Cheetah do with the weigth

mattipee, 2020-04-17:
It doesn't appear that GC is sending the weight through. I think it probably should, or could be made to, I'll look into it.

Zwift has me coming through to FortiusANT at 96.6, though I'm configured in my profile as 95. So a very light virtual bike at 1.1kg!

Note: floating point weights coming frrom ANT will need converted to int() before formatting the message to go to the trainer.

Overall though, I do like the feel of the first-principles power calcs (drag+slope+roll) plus populating weight for the virtual flywheel.

WouterJD, 2020-04-18:
GC?
Thanks for rounding hint
Ride On!👍

WouterJD, 2020-05-01:
Would it be an idea to combine this issue with #51 since it appears related.

mattipee, 2020-05-01:
@WouterJD  - yes.

closed 2020-05-01 by WouterJD