WouterJD/FortiusANT #52: 1932 iFlow and Rouvy

mikeherriman, 2020-04-17:
Hey all,

i gave up on the FortiusANT software on Linux and went back to Windows.
i have everything installed and functioning.
i am able to get the old iFlow recognized and paired to Zwift......it recognized it right away.  i also have a Flow Smart right next to my old iFlow.  Zwift is able to pickup both devices with no problems.  i am a big fan of the RouvyAR application and want to use that.  

Rouvy gives a message that Bluetooth needs to  be enabled and ANT+ is "coming soon". from what i can see on their website ANT+ is already supported.  How does this work?  i also see that there are closed tickets in the FortiusANT archives indicating that it works.

what am i doing wrong?  is there a different rouvy software than the RouvyAR that is on their website?

please advise.

WouterJD, 2020-04-18:
Yeah, had to search a lot to find the link. Will post asap!

mikeherriman, 2020-04-18:
Thanks!
I was able to find all of the “updates” from version 4.1.x all the way up to 6.2.x on the Rouvy changelog pages....tried to install several of these updates, but the software throws an error.  This is likely because they are only updates and not a full installation.

I appreciate all the work you are putting into this!

Thanks again
Mike

mattipee, 2020-04-18:
I had the same confusion. There are two dowloads...

My understanding is that the new "Rouvy AR" in the downloads page is still a beta release and doesn't (yet) support ANT+
 - https://cdn.virtualtraining.eu/update/setup_rouvy_ar.exe
 - Reports as 0.12.0.27 in About dialog

Scroll further down that page and there's "Rouvy Workouts". I'm guessing this is the older version, to be replaced by Rouvy AR, and will have the AR rides removed but still has them just now.
 - https://cdn.virtualtraining.eu/downloads/setup_rouvy.exe
 - reports as 7.0.0. build 0 in the Help > About dialog

It shows the following banner:
[image]

WouterJD, 2020-04-18:
Rouvy Workouts is the one that I use.

closed 2020-04-18 by WouterJD

mikeherriman, 2020-04-18:
Thanks fellas!

mikeherriman, 2020-04-18:
Just a quick note.  I just finished my first ride using Rouvy Workouts and FortiusANT.  Everything synched up and displayed correctly between Rouvy GUI and FortiusANT GUI....power transferred into Rouvy correctly, incline % sent to FortiusANT from Rouvy.....however, the actual brake resistance did not seem to change up or down with the changes received....seemed pretty steady at a zeroish resistance.  

Is this normal operation?  Did I miss something in the setup?

WouterJD, 2020-04-20:
If it does not work correctly, please raise an issue