WouterJD/FortiusANT #53: A little help please...

RodeTheAlpe, 2020-04-18:
Went through these steps a couple of times.  Windows 10 running two ANT devices with Tacx Fortius 1942 blue control box
- Uninstalled device (Jungo showing Fortius) Delete driver software checked
- Unplugged/Replugged in Fortius, showed up as unknown
- Right click and select "update driver software"
- Select libusb-win32 devices
- Select ANT USB Stick 2

Fortius shows up in Device Manager as new ANT USB Stick 2 device (one of three)

- Ran FortiusANT.exe

Fortius shows up again each time in Device Manager as Jungo device with Tacx Driver and can't be discovered by Zwift or similar.

Please tell me I'm missing something easy here or is there another (Tacx?) file I need to delete?

Thanks!

WouterJD, 2020-04-18:
This is what my device manager shows:
[image]
Two ANT+ dongles and one VR=virtual reality device.

As you see; there is no Fortius in the device manager; it may be that you have installed Tacx software (drivers) which should not be the case.

RodeTheAlpe, 2020-04-19:
Thanks, but I seem to be going in circles.  Each time I uninstall the Tacx drivers, unplug the Fortius USB, plug in the Fortius USB.  It comes up as "Unknown Device", I update driver choosing  Select libusb-win32 devices, ANT USB Stick 2.  It shows under  libusb-win32 devices along with both ANT sticks.  As soon as I run FortiusANT the driver updates and goes back to a Jungo Device.
I really feel like I'm missing a step but not sure what it could be 

[image]

[image]

[image]

[image]

WouterJD, 2020-04-19:
as if the first step in https://github.com/WouterJD/FortiusANT/wiki#windows is not complete.
Do not just remove the device; remove (all) tacx software so that plug&play does not find it again.

On my FortiusANT system I cannot run any tacx SW to guarantee this will not happen. I use tts4 on another system and have to unplug usb-cable t the other to make it work.

Hope this helps

WouterJD, 2020-04-19:
By the way, it is not related to the USB dongle software; the tacx must be a libusb device.

The instructions are copies from:
https://github.com/john-38787364/antifier/blob/master/README.txt

The last line is strange though

RodeTheAlpe, 2020-04-20:
Up and running!!!  Not sure why it chose today to work but it seems great.  I'll play around with it some more and get my Zwift account up and running.  Thanks for your help and for the awsome app!

WouterJD, 2020-04-20:
Hi @RodeTheAlpe great that it works now.

Please confirm that the last step in https://github.com/WouterJD/FortiusANT/wiki#windows is correct (especially the "**Select ANT USB Stick 2**" part)
Also check how it shows in device manager and show image to improve installation instructions.

By the way, did we introduce ourselves? My name is WOuter from the Netherlands.

RodeTheAlpe, 2020-04-21:
Hi Wouter, Jeff (RodeTheAlpe) from Los Angeles, nice to e-meet you too.

The instructions were pretty clear although I never had "VR-Interface" show in Device Manager, mine went from "Unknown Device" and became "ANT USB Stick 2" once I updated the driver per the directions.
As shown in my image I wound up with three ANT USB Stick 2 showing in Device Manager.  I had to check the Properties of each to determine which device was added (Windows makes it a pain to rename the devices).  In your Wiki you mention a User Interface with four buttons but I have yet to see that show up.

For some reason in my previous attempts FortiusANT did not continue running when it was launched and the fact that the device kept pushing back to Jungo Fortius makes me think that some Tacx driver was stuck in memory or cache someplace and finally cleared.
No problem running Zwift, I'm going to go through Rouvy tonight.  Resistance felt comparable to similar rides through TTS4 which was a nice surprise.

I haven't tried using my old school Sigma Heartrate Monitor but was wondering if it would work still through Zwift or if I need to upgrade to an ANT+ version.  I see the results in your readout but wasn't wearing it during my test last night.

Here is a screen grab of the Device Manager with the Fortius plugged in
[image]

Thanks again for all of your hard work!

WouterJD, 2020-04-21:
Well, of course this triggered me to check the installation procedure and I've done it on a fresh laptop. I have updasted https://github.com/WouterJD/FortiusANT/wiki#windows with the experiences, please see there.

Did we introduce ourselves? My name is Wouter from the Netherlands and I'm anxious to see your results on [Strava](https://www.strava.com/athletes/2885978)!

closed 2020-04-21 by WouterJD