WouterJD/FortiusANT #55: Support for Data Page 50 - Wind Resistance

mattipee, 2020-04-20:
To allow wind resistance coefficient and head/tail wind to be provided and incorporated into calculation of GradeToPower().

No rush, statement of intent, as I may get round to implementing this myself.

Motivation - I enjoy re-riding .gpx files and would like to simulate approximate wind conditions.

WouterJD, 2020-04-20:
No rush, point noted.
What software supports this; I assume the message is to be received by the trainer?
Please attach logfile where message is available.

mattipee, 2020-04-20:
I don't know if any of the packages I may download and use already issue this, but ultimately I may modify GoldenCheetah to issue this information. Given lat/long during a .gpx replay for example, a compass direction could be calculated and head/tailwind value then calculated from some specified average wind speed/direction, and issued to the FortiusANT. Perhaps I may implement some variability for gusts, or swirl. :)

WouterJD, 2020-05-13:
Implemented in https://github.com/WouterJD/FortiusANT/tree/54-usbTrainer

WouterJD, 2020-05-14:
Implemented in 3.0

closed 2020-05-14 by WouterJD