WouterJD/FortiusANT #56: Support for Data Page 51 - Coefficient of Rolling Resistance

mattipee, 2020-04-20:
Already parsed in page 51, but would like to be able to actually use a non-default received value in the physical calculation of required power in Grade2Power().
Suppose need to check that "use default value" 0xFF is provided by sending system (Zwift/etc..) if unused, or use if present?

WouterJD, 2020-04-20:
Well, perhaps lazy programming - but I did not see the message coming in and hence ignored the possible existance.

mattipee, 2020-04-20:
It's a very subtle effect I imagine. I just like the idea of simulating where parameters are provided. I ride on country roads or through the woods, rarely in a velodrome. Again if the software I use doesn't provide it, I may modify, or at the very least specify as a parameter to FortiusANT if not available on ANT+.

WouterJD, 2020-05-13:
Implemented in https://github.com/WouterJD/FortiusANT/tree/54-usbTrainer

WouterJD, 2020-05-14:
Implemented in 3.0

closed 2020-05-14 by WouterJD