WouterJD/FortiusANT #57: Using FortiusANT directly on a RaspberryPi without a head-unit

Gunzo182, 2020-04-21:
I obtained a non functioning Fortius Trainer (T1932 blue-ring, T1941 motor brake) a good while back. At that stage I tried to get it running through GoldenCheetah and some old Tacx software with no luck. 

The current "Lock-down" rekindled my interest to get it running . A few searches later I found FortiusANT, installed it and connected the trainer. When I "start to pedal" the motor starts running forever, but I get no sensor feedback to FortiusANT. My suspicion is that there is an issue with the rx signal. I did open the head-unit and found a lot of corrosion and oxidation. 
*Note that I don't have the cadence sensor, is this required?  

Upon your reference to "totalreverse/ttyT1941" it got me thinking to get FortiusANT running without a head-unit, using serial communication. 
Now here comes my questions:
- Will this be possible?
- How could I achieve this technically?
- Will an old RPi model B be sufficient? (Processing power, single core, ram ect...)

My end goal is to have FortiusANT running on the Pi, and to transmit and receive an ANT+ signal that I can use with Zwift.

I am more mechanically inclined, but I can help myself around electronics and coding with the help of google. If you can give me some guidance it will be greatly appreciated.

I did note that @Alanimus uses a RPi Zero. I'm not sure about how he's using it on his system.

Thank you

WouterJD, 2020-04-21:
Let's answer some questions
- If Zwift works in ERGmode (fixed power) the resistance is calculated using the wheelspeed. Cadence is used to detect that the users stops pedalling. If you add a simple ANT+ cadence sensor on your system, we could pair with that one and use it. That's an hour work.
- I do NOT know whether the Fortius hardware use the cadense sensor itself.

- FortiusANT communicates through USB cable to Fortius; if you have a module to address the serial communication it could be integrated. @totalreverse has more knowledge in that area than I do. It would require a revision of usbTrainer.py.

- I have no experience whatsoever with "RPi"; please send a link to what device you use. I did not hear from @Alanimus since the comms you found.

In this lockdown period, I still like to introduce myself. I'm Wouter from the Netherlands and like to know who I chat with; you also find me on [Strava](https://www.strava.com/athletes/2885978).

Gunzo182, 2020-04-21:
Hi Wouter, 

Of course. Excuse my perceived rudeness. I appreciate a proper introduction myself.
I'm Richard from South Africa. [Strava](https://www.strava.com/athletes/266024)

Thanks for your reply. And for all the hard work you've put into FortiusANT.

I do have an ant+ cadence sensor. I just wanted to be sure that the missing sensor is not the cause of FortuisANT not receiving sensor data.

I think as a start I'm going to try @totalreverse's "ttyT1941.py" and a usb-ttl module to communicate to my Fortius T1941. If i'm able to get instructions to; and messages back from the trainer I'll start bothering you again about the required changes to "usbTrainer.py".

The Rasberry Pi model B [RPi](https://www.sparkfun.com/products/retired/11546) is an older Raspberry Pi computer. I don't have use for mine right now, and this can be perfect. Again I'll test drive it and report on the performance. 

Thank you.

WouterJD, 2020-04-21:
Hi Richard, no offense at all, great to be able to make people sport in this lockdown period!

As soon as you're ready for it we will work together to get this working. Currently pretty occupied integrating Tacx i-Vortex. As soon as that's done, some code cleanup work to be done.
Adding a cadense sensor would be a small job.

Have a lot of fun, hardware, software and cycling - it will all be there!

mattipee, 2020-04-21:
@Gunzo182 I have a hunch, which is worth ruling out.

What command line options are you providing on the Raspberry Pi? Specifically, are you running  the GUI with -g? I'm at the very early stages of playing with FortiusANT on Linux, and I believe the GUI is unstable. I was seeing the calibration routine, after I find HW, click Start and then kick the pedal to initiate calibration, it starts to run the motor at 20km/h, but no values were appearing in the GUI and the motor would continue to run.

You could try "-d 1" which should enable some debug on the console, either with or without "-g", and watch the output. What I'm saying is there's a chance that you could in fact have a functioning head unit and be able to use FortiusANT with USB mode.

Can you let us know what you find out trying the above?

Gunzo182, 2020-04-25:
Hi @mattipee , 

I am yet to run it on my RPi. I see you are also trying to get a headless (or rather head replaced by a Raspberry Pi) system running. Like I've stated earlier, I'm no software or electronic specialist, so hardware hacking is very much just a hobby.

Even though I did not run FortiusAnt on a RPi, I am running it on a Linux system. I did get it to run in the terminal and gui. But like I said, I could only get my wheel turning. 
I split my serial cable to get a logic analyzer attached and I tried to clean al of the oxidation off the head unit's circuit board. Now I cant access my head unit anymore, so I'm depending on a headless system.

I'm following your progress with great interest, and I am still trying to get something running from my side. I am able to pick up communication using ttyT1941.py and a CP2102 Module. But I'm not getting the communication I'm suppose to have.

Excited about this!

Alanimus, 2020-04-26:
I did get it working nicely with a Raspberry Pi Zero. Sorry  been very
busy with school but I use it every day.   Basically I made my own smart
trainer using a Pi Zero and some sensors and servo. It works well but since
I am using a mag trainer the lack of gears means you basically run into the
same issues as if you were using Zwift Power (like no ability to change
gears and ramp up your difficulty manually). I don't have any familiarity
with the actual Fortius trainer funnily enough as I am not using that code
at all but I would imagine it's no different than using something like GPIO
Zero to control a servo with Zwift resistance. The reason I found this
program so useful is I had no idea at the time how to interface with ANT+
but have since then learned a lot.

So let me try to make sure I understand, are you trying to connect the
Fortius trainer motor to a Raspberry PI and then use Fortius Ant program to
connect to Zwift and control the trainer's motor? Where does the Fortius
part come in?

mattipee, 2020-04-26:
> I am yet to run it on my RPi. I see you are also trying to get a headless (or rather head replaced by a Raspberry Pi) system running.

@Gunzo182  I did write in another issue about intending to run headless on a Raspberry PI, t which I assume you are referring. Unfortunately, I think there are two definitions of the word "headless" - running without the "head" unit of the Fortius (the handlebar mounted blue/green/white unit) and interfacing with the motor controller directly, and running without a monitor or FortiusANT GUI.

I just meant the latter, sorry. Until my head unit dies, I'll probably not be doing much otherwise. It's definitely possible, and would be happy enough to assist code-wise.

Gunzo182, 2020-04-26:
> I did get it working nicely with a Raspberry Pi Zero. Sorry been very busy with school but I use it every day. Basically I made my own smart trainer using a Pi Zero and some sensors and servo. It [...]

Just for clarity, The Fortius system I have has a brush-less motor to generate resistance.

The Fortius trainer consists out of effectively 4 parts. The Pc that runs the virtualization software and drivers. This is connected to the head unit/controller (attached to the bicycle handlebar) through USB. On the head unit you have 4 buttons to control the system and a microprocessor to communicate. The head unit is connected in turn to the "Power Back" motor controller through a serial interface. The "Power Back" controller control's the resistance (or assistance) the motor generates.  The motor is just a brushless DC motor.

*BTW the "Power Back" feature as far as I understand it, takes the excess energy created by pedaling against resistance and pushes it back into the power grid. This is a nice feature!

So to answer your last question: I want to use a Pi (or any computer) to talk to my Fortius motor controller without the head unit, and then use the ANT+ part to connect to Zwift.

Gunzo182, 2020-04-26:
> @Gunzo182 I did write in another issue about intending to run headless on a Raspberry PI, t which I assume you are referring. Unfortunately, I think there are two definitions of the word "headless" [...]

No Problem, I understand now what you're trying to achieve. 
Thanks for the offer of help. I'll see where I can get at, and then might press on your button.

Thanks!

WouterJD, 2020-04-26:
> The Fortius trainer consists out of effectively 4 parts. 
+ Cadence sensor
+ ANT heart rate monitor
Both devices are passed on the USB but functionally not used by the headunit, I think.

Alanimus, 2020-04-26:
So what you are trying to do seems pretty complex. @WouterJD does Fortius Ant completely take over the duties of the head unit or does it just send it signals to the firmware still on the head unit? 
I'm guessing you would have to simulate the firmware on the Pi to use it in this way. 
The other option would be to ignore all that and just connect the actual sensors to the Pi and re-code them. 

Anyway, I can help with getting Fortius Ant itself running on Pi and using ANT+ on the Pi instead of a USB interface along with the Zwift data to control something, but the rest I would probably have to have a trainer to mess with.

WouterJD, 2020-04-26:
FortiusANT communicates thru USB with the headunit (or thru ANT+ with iVortex, but that's offtopic here).

WouterJD, 2020-05-01:
Hey guys ... any news on this implementation?

StFalagar, 2021-06-20:
Hi @WouterJD and everyone else...

I have been looking all over but not sure if anyone got this working as yet...

I have my trainer connected to my Raspberry Pi 3 and it is working using @totalreverse 's ttyT1941...  
I get readings coming from the T1941 trainer into the Raspberry Pi and can set erg and grade and send through info... 
Is there a way to take these values and connect to Fortius Ant (possibly on the same Raspberry Pi) and create a BLE signal from there to Zwift?

[image]

Thanks a lot...

Sarel

WouterJD, 2021-06-21:
Since there is more interest, I reopen.
Will be a next-winter-time topic to see to integrate @TotalReverse option into FortiusAnt.

StFalagar, 2021-06-21:
Thanks Wouter.... I will try and see if I can get it working.... doesn't look too complicated...
Basically use @totalreverse's code in new SendToTrainer and ReceiveFromTrainer Methods
I am not a Python developer but will try... :)

WouterJD, 2021-06-23:
@StFalagar a new class should be created, implementing the new SendToTrainer and ReceiveFromTrainer methods.

StFalagar, 2021-06-23:
Thanks Wouter... I am still working through all the code to get my head around how FortiusAnt is structured etc.
Will do once I start writing code.... unfortunately.... I can't focus on this full time...

WouterJD, 2021-06-24:
Just a quick question: what physical/logical interface is used between PC and brake?
(I did not study @TotalReverse 's material yet)

StFalagar, 2021-06-24:
Either a Raspberry Pi GPIO connection or a TTL to USB Converter.  In my case I have connected the brake to a Pi3b's GPIO and ran @totalreverse's program.  It caters for calibration and returns cadence, speed, power etc. from the brake in much the same way as when you run FortiusANT without GUI.  

I suppose in the case of a PC it would be the same scenario where you use the TTL to USB Converter to connect the brake to USB and then run the Python Program to read the info from the brake.

It only works with the Motorbrake though and not with Eddy Current Brakes.

StFalagar, 2021-07-05:
Wow this is more challenging than I thought it would be hehe.... Massive learning curve for me here....
Just a quick update... 

1. I get the readings from the T1941 into the RPi and into FortiusAnt.  
2. Created a new Class specifically for Direct Serial Connections as you said and call that from usbTrainer.py's GetTrainer using clv's.
3. According to @totalreverse's github I had to disable onboard bluetooth on the Pi for GPIO to use the serial connection
4. So... readings coming into FortiusAnt... Calibration working (sort of) and able to get Power, Cadence Etc. coming into FortiusAnt when pedalling.
5. Using a Bluetooth Dongle I was able to connect to Zwift although Zwift only picks up the  Power device and no Cadence or Controllable devices as yet...
6. Also...  Zwift states "No Signal" so still some work to do there...  This happens for the T1941 Serial connection as well as in Simulation Mode so not sure if anything else might be the problem...
7. If I can get this working, I will have to clean up everything....  I still use @totalreverse's decryption etc. and suppose I'll have to learn how struct work to do that...

"Quick Change" now turning into a project lol

Cheers :)

planetSammy, 2021-10-06:
Hello, I try to install FortiusAnt on a RPi Zero WH and follow the instructions from the current manual. Step 1 and 2 (BLE-Support) works fine. I want to install the Package for the Ant+ Dongle (for Genius), but in the destination folder "raspberry" on my RPi 0 is no 02_InstallPackagesANT.sh?

Did i something wrong, must I insert the Dongle during the Download process?

Best regards, Torsten

WouterJD, 2021-10-06:
I will respond later (>16th), not at home nowSomebody else can help? Met vriendelijke groetWouter Dubbeldam

planetSammy, 2021-11-02:
Hi, short Update...when I start the "server.js" under ~/FortiusANT/node with the following lines:
[image]
I can establish a stable connection!
[image]

But when I start the with "sh FortiusANT.sh" i can't get a connection with bleno...any ideas, help?

planetSammy, 2021-11-07:
Hello, for everybody who runs into the errors above!
It is a bug in the last build of bluetooth-hci-socket@0.5.3-8, but you can fix it with the last stable version.

Run this command in the ~/FortiusANT/node folder, after that it works!
--> npm install @abandonware/bluetooth-hci-socket@0.5.3-7
 
See also the original post:
https://github.com/abandonware/bleno/issues/30#issuecomment-960316742

Best regards and nice ride  
Sammy

MeanHat, 2021-11-10:
Thanks @planetSammy  - I also suddenly experienced BLE disconnects when running Zwift with FortiusANT. Your fix has worked - though I used both commands in the link you posted:
`npm install bleno@npm:@abandonware/bleno`
`npm install @abandonware/bluetooth-hci-socket@0.5.3-7`

planetSammy, 2021-11-10:
Hi @Meanhat, there is another point, not really an issue. When I start FortiusAnt.sh, the first step is to delete the old log-files in the "trashcan", but the script cannot find the folder:$"/home/pi/.local/share/Trash".

I installed manually the trash-cli over "apt-get", but no default trash folder appears...do you know why or have any idea?

Best regards Sammy

planetSammy, 2021-11-12:
Fixed it 😁
[image]

MeanHat, 2021-11-12:
Hi @planetSammy - sorry not to have replied sooner but really pleased you have fixed your problem. Happy riding!

planetSammy, 2021-11-14:
Hi there, next issue 🤔 when I am riding on RGT_Cycling the Applikation (FortiusAnt) sometimes does not switch to power mode and remains permanently at 100W...I use the following settings:
[image]

Is there a way to use permamently the PowerMode, i never want to use the Grade Mode?!?

Best regards Sammy

WouterJD, 2021-11-15:
It seems this issue is solved, I assume can be closed.
If not happy to reopen

WouterJD, 2021-11-16:
Note: the issue "Using FortiusANT directly on a RaspberryPi without a head-unit" has diverted in another discussion.
@StFalagar how far are you  and is the development still ongoing?

StFalagar, 2021-11-16:
Hi Wouter

I got it to a point where I can't get the bluetooth working and I think the problem is that the trainer's connection to the pins uses the same ports as the bluetooth.  This means I have to now start playing with Ant+ and I have become very busy with work etc.  I have put this on hold for now.  I would like to pursue it again later but for the time being cannot get to it...

WouterJD, 2021-11-16:
OK @StFalagar as soon as you have time, I will see the reopen. Let's keep closed for now, that's easier to keep track on current issues. Succes!

closed 2021-11-16 by WouterJD

StFalagar, 2021-11-17:
No problem... thanks Wouter

erik-groen, 2022-04-27:
Sorry to barge in on this discussion, but it seems almost to adres the answers i am looking for. Is it  possible to use the fortius just as an adjustable generator? I am verry much into generating electricity. https://groeninnovaties.nl/groene-stroom/ Preferably i would connect the T1941 motor brake directly  to an arduino with potentiometer. Has somebody the pin layout of the connector and the brake commands for the T1941? It would be great, thanks and i'll keep you posted.

StFalagar, 2022-04-28:
Hi... Yes... You can find the information here https://github.com/totalreverse/ttyT1941