WouterJD/FortiusANT #59: FortiusANT and GoldenCheetah

WouterJD, 2020-04-21:
Hi, this is a great facility for us who don't have or can't get smart trainers. (Cost!) Thank you for you efforts. I've connected my Fortius (0x1932) and can see output data for all parameters, however in 'Run mode' the Target stays fixed at 100W. I'm using GoldenCheetah (GC) as the trainer app, and it seems that either GC is not sending data to Fortius, or Fortius is not accepting. Even if I use manual ERG mode and increase or decrease load, there is no change on the trainer.

1. Where does the Target come from?
2. Does FortiAnt send info back to the trainer?

Thanks again for this utility.

B.

_Originally posted by @brianrow in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-617202533_

WouterJD, 2020-04-21:
Hi @brianrow

I have created this separate issue for you.

This sounds like issue #49 so I would be curious what version of FortiusANT you use.

GC interrogates what capabilities are supported, something other software don't do and therefore believes ErgMode and GradeMode are not supported. This should be resolved since 17-4-2020 and [confirmed](https://github.com/WouterJD/FortiusANT/issues/49#issuecomment-615454356).

Your questions:
1. The Target comes from the CTP (Trainer Road, ..., Golden Cheetah)
2. FortiusANT sends the required resistance to the trainer (Fortius 0x1932) which is based upon the Grade or TargetPower from Golden Cheetah.

Welcome onboard; where are you from?

WouterJD, 2020-04-21:
@M4M4R2  states that no info would be sent to the trainer; he has a Tacx Flow + 1932 head causing the software to thing that it's a Fortius, which needs other calculations. Sorry @M4M4R2 have to respond to you still

brianrow, 2020-04-21:
Hey, wow fast response - thanks.

As far as I can determine, the S/W is version 2.5, the most recent master
download.

I checked out #49, but need to digest what they are actually saying. I'll
also look for which device @mattipee used - perhaps my config is wrong?

I'm based in Johannesburg, South Africa.

Respectfully,

*Brian Rowbotham*
*Regional Technical Executive*

T:   + 27 87 820 0164 <+27878200164>  M: +27 82 652 9181 <+27826529181>
E:   brian@connection-telecom.com
W:  www.connection-telecom.com    www.telviva.co.za

brianrow, 2020-04-21:
> @brianrow <https://github.com/brianrow> Are you running the python
> version, or the Windows .exe?
>
> The fix for getting ERG/Grade mode to work with GoldenCheetah is only a
> few days old, and only fixed in the Python .py source files - it hasn't
> made it into the Windows .exe that @WouterJD <https://github.com/WouterJD>
> builds.
>
> If you are in fact not running the Windows .exe but running from Python,
> can you confirm in FortiusAnt.py, around line 759 or nearby that you have
> "if RequestedPageNumber == 54:"?

Hey, thanks for the reply.

I'm currently running the windows .exe file, but can run the python code.
I'll get it running and take a look.

TIA.

B.

Respectfully,

*Brian Rowbotham*
*Regional Technical Executive*

T:   + 27 87 820 0164 <+27878200164>  M: +27 82 652 9181 <+27826529181>
E:   brian@connection-telecom.com
W:  www.connection-telecom.com    www.telviva.co.za

WouterJD, 2020-04-24:
Hi @brianrow how are things going?

brianrow, 2020-04-24:
Hello,
I experimented quite a bit with the system, and although the FortiusANT app
works perfectly, there does seem to be an issue with resistance feedback to
the trainer from GC. I went back to the native GC app and tried the USB
interface again, but this still disconnects randomly.

In desperation I've installed CadenceVR (Personal license is free
currently) and will try that with FortiusANT later today.

Respectfully,

*Brian Rowbotham*
*Regional Technical Executive*

T:   + 27 87 820 0164 <+27878200164>  M: +27 82 652 9181 <+27826529181>
E:   brian@connection-telecom.com
W:  www.connection-telecom.com    www.telviva.co.za

WouterJD, 2020-04-24:
The resistance feedback is a challenge to do right and may need improvement.
Don't get desperate and let me know experience, I will certainly use your feedback to improve.

brianrow, 2020-04-24:
Sure thing, I will persist.

Respectfully,

*Brian Rowbotham*
*Regional Technical Executive*

T:   + 27 87 820 0164 <+27878200164>  M: +27 82 652 9181 <+27826529181>
E:   brian@connection-telecom.com
W:  www.connection-telecom.com    www.telviva.co.za

WouterJD, 2020-04-24:
I have reviewed calculations and I believe that it works as designed.
For "designed" refer to Wiki and check also command line flags to influence the calculations.

Default is ftp=200Watt, perhaps you want a higher value.
Also, check the option of changing gears while doing a grade-mode ride, using the headunit.

Succes, ready for tests!

WouterJD, 2020-05-01:
PLease follow #51 and other issues discussing the powercurve

closed 2020-05-01 by WouterJD