WouterJD/FortiusANT #60: FortiusANT.exe blinks for 2 seconds

WouterJD, 2020-04-24:
Hi, everebody

I try start with check this new App...
But no succes today.
I have good PC win8/х64 with Tacx Fortius and extra old Netbook with XP, х086 (sure),
and two Garmin Ant+ sticks.

FortiusANT.exe  blink for 2 seconds, ExplorANT.exe also ....
(At both pc)

https://www.strava.com/athletes/1160339

_Originally posted by @Barh007 in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-618660907_

WouterJD, 2020-04-24:
Good morning barh

Welcome in the FortiusANT Team!

Your message is quite short.

On one PC start FortiusANT from a .bat file:
```
    FortiusANT.exe -g -d127
    pause
```
so after ending you can see the messages on the console. -g causes to start the user-interface, -d creates a logfile. If thinks do not work, please attach the logfile.

On the other PC you can start your application (Zwift, Trainer Road, ExplorANT.exe ...) and the application should see FortiusANT as ANT+-trainer. If you start ExplorANT.exe with -d flag it also creates a logfile.

Success and please attach logfile when commenting on the issue.
Good luck and if I'm correct, you are located in Moscow?

Barh007, 2020-04-24:
Hi, dear WouterJD
Thanks for fast response!

Problem has been fixed today, App work with Rouvy perfectly.

The problem was the incorrectly installed trainer driver into XP laptop.
Probably if FortiusANT does not see the trainer in the system,
it does not start correctly.

I had to make a "crocodile" ::
Stick and trainer are connected ti Win8(x086) tablet via usb-splitter.

Yes, I"m from Moscow, Russia and old *Antifier Project* user.

https://www.strava.com/athletes/1160339

Barh007, 2020-04-24:
http://images.vfl.ru/ii/1587728190/64982d9c/30315967.jpg

http://images.vfl.ru/ii/1587728190/95c80072/30315966.jpg

WouterJD, 2020-04-24:
Show the results on strava! I'm curious.
So FortiusANT runs on Windows XP as well?

WouterJD, 2020-04-24:
PS. Since you are an old antifier user, please share experiences and differences!

closed 2020-04-24 by WouterJD