WouterJD/FortiusANT #61: What USB ANT+ dongles to use and where to buy?

WouterJD, 2020-04-24:
FortiusANT supports USB dongles with the ID's 4104 (Suunto), 4105 (Garmin) or 4100 (Older).

USB-dongles can be bought on miscellaneous shops, [Google](https://www.google.com/search?q=usb+ant%2B+dongle) for "usb ant+ dongle"...

Currently (April 2020) it appears there is a high demand and therefore some links to suppliers to give you suggestions - sorry for suggestions in the Dutch area. @darkpotpot, hope it helps.
- [Aliexpress](https://nl.aliexpress.com/popular/usb-ant-stick.html),
- [Amazon](https://www.amazon.co.uk/TAOPE-Adapter-PerfPRO-CycleOps-TrainerRoad/dp/B01N2S103O),
- [Bike components](https://www.bike-components.de/en/Rotor/USB-ANT-Dongle-p55397)
- [Bol](https://www.bol.com/nl/p/wahoo-fitness-ant-usb-adapter-met-kabel/9200000076122781/),
- [Elite IT](https://www.elite-it.com/fr/produits/accessoires/capteurs/cle-usb-ant),
- [Futurumshop](https://www.futurumshop.nl/xand-ant-dongle.phtml), 
- [Zwift](https://zwiftinsider.com/ant-dongles-for-zwift/)

It's just suggestions - use this page to post questions and others may help you.

totalreverse, 2020-04-24:
I have two budget ANT+ sticks labeled ANSELF. They work without problems. 

You can order them i.e. from 
[ebay](https://www.ebay.com/sch/i.html?_nkw=anself+ant%2B)
[amazon](https://www.amazon.com/dp/B01M3VQP6Z)

My sticks have ID 0x1008 (4104). Other reported that their sticks have ID 0x1009 (4105), which should work, too.

RodeTheAlpe, 2020-04-26:
I bought Anself from Amazon as well, no problems using with Zwift, Rouvy, etc...
https://www.amazon.com/gp/product/B01M3VQP6Z

aegisdesign, 2020-05-06:
@WouterJD Looking at the investigation on tacxfaqx.com, is it possible to detect the faulty cycplus dongles lack of continuous scanning mode in code and warn they won't work?

WouterJD, 2020-05-09:
Any code-suggestions welcome. Please follow #65

harveygf, 2020-05-11:
@WouteJD Great product. Managed to finally get my old machine connected to Zwift. I have the I-flow with the T1932 (white/Blue) head unit. My question is to do with the Ant+ dongles. I was able to have the Trainer connected my one Windows 10 Laptop with the USB that runs a DynaStream AntUSB-M dongle and has the 0x1009 ID. I then ran the FortiusANT.exe and everything worked 100%. Then on a 2nd windows 10 Laptop I had the same DynaStream AntUSB-M dongle and has the 0x1009 ID, and this had zwift loaded. I was able to pick up the Ant sensors and my HR strap (direct into Zwift) and do a ride.
The problem is I still required the old TACX software to work for my wife. So I converted the first Laptop back to the Tacx software and decided to run everything off the 2nd laptop. I was able to get the trainer to talk to the FortiusANT.exe and it picked up and calibrated the trainer, also picked up the DynaStream AntUSB-M dongle and has the 0x1009 ID. I pressed the start button and Speed, Cadence and power were working 100%. I then started the Zwift software, which it picked up the sensors, but when doing that  FortiusANT.exe looses connection to the ANT dongle. It seems Zwift is trying to use the same ANT dongle as  FortiusANT.exe instead of it using the 2nd dongle. I know both dongles are the same but have I done something wrong with the setup or installation of the ANT Dongles, or do I need to use 2 different dongles.
Again this is an awesome product you are creating. Thank you very much

WouterJD, 2020-05-11:
Hi @harveygf . I'm Wouter from the Netherlands and happy that you're thrilled about FortiusANT. I'm always curious where the product goes to and therefore like to know where you are located. If you like, ping me on [strava](https://www.strava.com/athletes/2885978)!

Basically what you try to do is OK and I have done it myself as well.
So we're talking about PC2 (#1 is for your wife).
Sequence is OK: first start FortiusANT, Start, Calibrate; then start Zwift. Note that during calibration, FortiusANT will not be seen by Zwift.

FortiusANT connects to the first dongle that is available and I assume also Zwift does so, then taking the second dongle.

You state that "FortiusANT.exe looses connection to the ANT dongle"; please explain - what happens?

harveygf, 2020-05-12:
Hi @wouterJD To keep you up to date late last night I managed to get all the drivers uninstalled for Libusb-win32 (Had to use an external program in the end). I then installed the libusb-win-32 "VR-Virtulal" driver. After that I created libusb-win-32 -  for the FortiusAnt to access the Ant+ dongle. This one I actually changed the name and the manufacture. Once this was done I tested the FotiusAnt.exe and all was working 100%.
Lastly I plugged in the 2nd Ant+ dongle, but then went to device manager and updated the driver but used the one supplied by Zwift. Once this was done, I was able to get FortiusAnt.exe to work with the trainer and swift to use the 2nd Ant+ dongle for connection. 
I will test it tonight with a ride.
Just a quick question for FortiusAnt.exe. I know you can add commands when starting it up like your FTP, Bike gears etc. Does it also use rider and bike weight in it's calculations for power?
Thank you again, great product.

WouterJD, 2020-05-12:
We are working on a new version where the power calculation is improved.
Refer to pullrequest https://github.com/WouterJD/FortiusANT/pull/84 where this development is discussed

harveygf, 2020-05-13:
@WouterJD Managed to do a zwift ride last night, Worked great. I did notice my ave watts was way lower than normal on both the fortiusAnt.exe and Zwift, but was still great to use it.
Went to do another ride today and unfortunately it has been blocked by Symantic Anti Virus for both the ExploreANT.exe and the FortiusANT.exe as Heur.AdvML.B Heuristic Virus. Even states is does not suspect it is a real virus but is blocked anyway by the "blooodhound and SONAR" in Symantic. As my office controls this part, I can't even into the ignore list :-( Not sure if anyone else has had this happen.
Trying to now see if the wife will let me use her mac, but doubt that ha ha

mruzzon, 2020-09-06:
> Any code-suggestions welcome. Please follow #65

I've been running the code on a Mac using Python successfully for the most part except for frequent dropouts while using the CYCPLUS ANT+ dongle. I've noticed that the frequency of dropouts improved considerable after setting the hear rate data to none, thou...

Any plans to add BLE support instead? Would be really neat to run directly from the Mac or somewhere else directly to my iPad... Thanks

WouterJD, 2020-09-07:
> > Any code-suggestions welcome. Please follow #65
> 
> ...on a Mac .. frequent dropouts while using the CYCPLUS ANT+ dongle.
> Any plans to add BLE support instead? 

- Sorry that CYCPLUS ANT+ dongle remains troublesome
- What do you mean with BLE-support

mruzzon, 2020-09-07:
It disconnects several times and reconnects, only issue that's causing for now is that it seems to hinder the calibration process...

BLE - what I meant is if it would be possible to broadcast the data using bluetooth instead of ANT+, no dongle needed in this case.

Thanks a lot for this!

WouterJD, 2020-10-26:
Issue closed, because I want to have a short issue-list to keep focus.
Questions and comments remain welcome!

kiwisibk, 2020-11-05:
So is the CYCPLUS dongle now working or not? Which one to recommend for best results? Thanks

WouterJD, 2020-11-05:
If you make the question binary; the answer is: no.
The reason is unclear.

Other dongles do not report similar problems

kiwisibk, 2020-11-06:
Great, thanks for the quick answer! 👍

Could you recommend me some good working dongles? Not too expensive if
possible?

Thanks once again!
Best regards
Arno

WouterJD, 2020-11-06:
Hi Arno how are you today?
Please tel me what Tacx you use; welcome in our community - you are the first athlete in Poland!

What dongle to buy is difficult to say; it depends very much on the country.
The header of this post provides links; Futurumshop-dongles work well and cost here around EUR 30,= (you need two). I don't know for you what the easiest shop is. If you have issues, please let me know.

kiwisibk, 2020-11-06:
Good morning :)

Thanks I'm fine, how are you?

I'm from South Tyrol in northern Italy, not Poland.. :)
It's a paradise for biking, unfortunately with lockdown we have to stay at
home... :(

I have an old i-Magic trainer, and unfortunately cannot use anymore the
Tacx TTS4 software.

So I found your excellent tutorial and hope now to be able to move to Zwift
or Rouvy... :)

So you mean in Italy another brand should be used?

Thanks
Regards
Arno

WouterJD, 2020-11-06:
Hi Arno, sorry for confusion.
I got a response on Strava at same time from Krzysztof Bejner and connected that to kiwisibik...

Please refer to https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-722936417
Amazon stick, 17 days delivery.

I hope it helps

PS South Tyrol I know a bit; tell me more? Are you on strava as well?

aegisdesign, 2020-11-06:
> I have an old i-Magic trainer, and unfortunately cannot use anymore the Tacx TTS4 software. 

Quick reminder that you can still use Fortius v2.02 with your i-Magic. It doesn't require registration unlike later software. See https://lunicus.org/tacx-trainers/fortius-software/ for a download.

kiwisibk, 2020-11-06:
Wow! Thank you so much for this great information!! :)

Will try asap this afternoon!

skrypt-nl, 2020-11-07:
Hi Wouter,
Do you have any idea if [this dongle from ADDFOO](https://www.amazon.nl/ADDFOO-CycleOps-TrainerRoad-Compatibel-Forerunner/dp/B089VM9XM9) is supported by your software?

WouterJD, 2020-11-07:
Hi @skrypt-nl welcome in the FortiusANT world. Please tell me some more, like name, where you are located and what Tacx you use.

The only dongle I had negative reactions for is the CYCPLUS, so you may give it a try. Good prince.

omedirk, 2020-11-09:
This one works just fine:
[https://hackerstore.nl/Artikel/1261]

WouterJD, 2020-11-09:
Thanks @omedirk I've added it to the issue description!

kiwisibk, 2020-11-11:
> > I have an old i-Magic trainer, and unfortunately cannot use anymore the Tacx TTS4 software.
> 
> Quick reminder that you can still use Fortius v2.02 with your i-Magic. It doesn't require registration unlike later software. See https://lunicus.org/tacx-trainers/fortius-software/ for a download.

I have problems running it on a Virtual Machine. Do you think Fortius Software will run as well on WIN 10? Thanks

WouterJD, 2020-11-11:
FortiusANT works well on WIndows 10; virtual machine I cannot judge

aegisdesign, 2020-11-12:
> I have problems running it on a Virtual Machine. Do you think Fortius Software will run as well on WIN 10? Thanks

Tacx's Fortius 2.x software requires 32bit Windows XP or Vista. I've not tried in on Windows 10. I run it on a Mac with a WindowsXP VM using VirtualBox 6.0. 6.1 does not support Direct3D which is required by Fortius. It has to be VirtualBox 6.0. You also have to set the amount of VRAM to 256MB. You have to do that on the commandline

`vboxmanage modifyvm "name of VM" --vram 256MB`

Or there's a trick to do it in the GUI by increasing the monitor count to 8, then you can select 256MB video ram. The 256MB setting persists even if you switch monitor count back to 1.

Also, don't install the Fortius 2.03 or 2.04 patches without adding tacxvr.com to C:\Windows\System32\drivers\etc\hosts 

`127.0.0.1 www.tacxvr.com`

That prevents the software trying to contact a nonexistent server. Otherwise you'll get a blue screen on startup.

kiwisibk, 2020-11-19:
Hello WouterJD,

I bought two ANT+ Sticks and your software seems to be recongnizing them.

I connected my i-Magic trainer interface and the USB driver install was
successful.

But your software isn't recognizing the interface:

[image: image.png]

What I'm doing wrong?

Many thanks
Best regards
Arno

WouterJD, 2020-11-19:
Hi @kiwisibk 
Welcome to the FortiusANT community
--
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
--
The image.png is not shown in github; did you read the manual and follow all steps? installing libusb is probably the most critical part.

kiwisibk, 2020-11-19:
Hello,

It's a Tacx i-Magic dyno with the white/green interface.

[image: image.png]
Is this enough information for you?

Otherwise please let me know.

Many thanks
BR
Arno

WouterJD, 2020-11-19:
Please respond on github; reply-mails do not transfer attachments.

lello72, 2020-12-11:
> I have two budget ANT+ sticks labeled ANSELF. They work without problems.
>
> You can order them i.e. from
> [ebay](https://www.ebay.com/sch/i.html?_nkw=anself+ant%2B) [...]

@WouterJD Wouter this is the reason why I bought these. I'll let you know if they work if and when I receive them. Thanks

lello72, 2020-12-17:
@WouterJD Aiuto Wouter!!!, after waiting for more than month, I got my Anself dongles and for sure they turned out to be Cyclplus. Damn!!! They are unusable with Swift as they keep disconnecting/reconnecting. I attach the log to see if I am missing something spectacular. DId anybody try Onlyesh Brand?

lello72, 2020-12-17:
[log fortiusant 17 dec 20 222100.txt](https://github.com/WouterJD/FortiusANT/files/5712310/log.fortiusant.17.dec.20.222100.txt)

WouterJD, 2020-12-17:
Looks like bad: CYCPLUS makes me nervous for sure.
It's the only ANTdongle reported with errors; I cannot tell anything about the others.

22:09:32,453: GetDongle - Try dongle: manufacturer=CYCPLUS, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)

22:12:39,785: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
22:12:40,691: AntDongle.Write exception (message lost): [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: Un dispositivo collegato al sistema non \xe8 in funzione.\r\n\n'
22:12:40,691: devAntDongle.read exception: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: Impossibile trovare il file specificato.\r\n'
22:12:40,691: ANT Dongle not available; try to reconnect after 1 second

mruzzon, 2020-12-18:
I have tested the ANSELF, Garmin and Magene dongles and I can recommend you try the Magene one instead... from my experience the Magene was identified as the same hardware as the Garmin dongle and both worked flawlessly; ANSELF was particularly bad when running the FortiusANT code on a Mac.

Google for more shopping options, but you can find them quite cheap in AliExpress:
https://a.aliexpress.com/_mMuZYfL

lello72, 2020-12-19:
> Looks like bad: CYCPLUS makes me nervous for sure.
> It's the only ANTdongle reported with errors; I cannot tell anything about the others.
>
> 22:09:32,453: GetDongle - Try dongle: manufacturer=CYCPLUS, [...]

Today I tried Onlyesh, recognized as Dynastream by Fortiusant and they work spotlessly with same exact configuration used with the unfunctioning Antself/Cyclopes dongles. Stay away from them. 

On another note, I have to remove calibration to make it work because the for some reason the calibration keeps running until I manually stop it. How do I fix that? I am using 1932 so I read calibration should work. Not a big deal but a nice to have. Thanks all for giving a second life to my trainer also a pleasure to play with CMD; I feel young again.

WouterJD, 2020-12-19:
@lello72 send logfile if calibration continues not workable

WouterJD, 2020-12-21:
> I have tested the ANSELF, Garmin and Magene dongles 

Thanks for info and Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

WouterJD, 2021-01-28:
Open for discussion; your posts will/may be reflected in the initial post for future reference

MeanHat, 2021-01-28:
hi @WouterJD - I've been using a pair Lifeline ANT dongles (https://www.wiggle.co.uk/lifeline-ant-usb) as well as a pair of Garmin ones successfully. Both are detected as Dynastream Innovations.

switchabl, 2021-01-28:
> Dongles from `manufacturer=CYCPLUS` (also advertised as Antself, sold by Amazon) are reported not working well with FortiusANT, which may be an installation and/or driver issue (#45 #65). See also [tacxfaqx.com](https://tacxfaqx.com/knowledge-base/cycplus-ant-stick/) This issue is "solved" from FortiusANT 3.0. (https://github.com/WouterJD/FortiusANT/tree/65-CYPLUS).

If this issue is the new reference on what to buy, this section should probably be updated as well. I think we know now that this is not an installation/driver issue and that it is not solved.

WouterJD, 2021-01-29:
> the offending dongles crossed off the list

Which ones should be crossed? I think the ones mentioned were mentioned working.

switchabl, 2021-01-29:
The amazon links are both "Anself". They probably used to be fine but both have recent reviews saying "doesn't work" so there is a good chance they are CYCPLUS now as well. The first hit for the aliexpress is a CYCPLUS too, but I guess there are others (and if you order from aliexpress you probably know it is a bit of a lottery).

RodeTheAlpe, 2021-01-30:
I haven't gone through all of these comments specifically but I've been using these Anself ANT+ dongles since April and have had zero problems (running Window 10);
https://www.amazon.com/gp/product/B01M3VQP6Z

I would have no problem recommending them to anyone.  Hope this helps

WouterJD, 2021-01-30:
@RodeTheAlpe thanks, will add to the list

switchabl, 2021-01-30:
> I've been using these Anself ANT+ dongles _since April_ [emphasis mine] and have had zero problems

I think the Anself dongles used to be fine until spring/summer 2020 maybe and newer ones often turned out to be broken CYCPLUS ones (unless you were lucky and got some old stock maybe) and several users here have reported problems. If you sort the Amazon reviews by date, the ones from October onwards are mostly negative.

At this point it is hard to recommended anything that looks similar to the CYCPLUS unless you can return it easily and are willing to take a gamble.

aegisdesign, 2021-01-30:
In the UK I noticed that SportsDirect (big high street sports company) are selling what appear to be Suunto Movestick ANT+ dongles for £12. The description is unfortunately terrible but the picture is a dongle. https://www.sportsdirect.com/suunto-suunto-012098

I've not ordered these from SportsDirect but the Suunto dongles work flawlessly (unlike the CYCPLUS I had before). The other place for a cheap dongle is eBay. Hang around for an obsolete Suunto M5 watch. It usually comes with an ANT+ dongle and a DUAL HRM belt which is dual old analogue and ANT (not ANT+!!!). The belt works with the Fortius head units. The M5 watch isn't bad either. You can often pick these up for less than the price of a new ANT+ dongle alone. Suunto have   ended the Movescount app & website the M series watches use hence the cheap prices.

WouterJD, 2021-01-30:
@aegisdesign thanks!

switchabl, 2021-01-30:
@aegisdesign That is a good point. I've bought several used ANT dongles for cheap myself.

As a general guide (whether buying new or used):

GOOD:
- ANTUSB-m: Most reliable and full-featured option (but also most expensive). Made by Dynastream and sold under a variety of brand names.  They are all identical so get whichever is cheaper. They can be ordered with custom branding so labeling may vary slightly. By far the most common are "Garmin" and neutral "ANT+":
[image]
Known brands: BKOOL, Dynastream, Elite, Garmin, hLine, LifeLine, Rotor, Saris, Quarq, Wahoo, XAND, Zycle
- ANTUSB2: Somewhat older model but well-supported. Chipset made by Dynamstream but available in a number of different shapes, including
  * Garmin ANT USB2: *Be careful, there is also an older USB1 model that looks almost the same but doesn't work with most software anymore, see below*
  https://3.bp.blogspot.com/-oHvK6G2z7Fs/VWLG9rI5vXI/AAAAAAAAHHA/GWI18V1zWdw/s160/FullSizeRender%25288%2529.jpg
  * Suunto Movestick Mini: https://www.dcrainmaker.com/2012/02/much-better-usb-ant-stick-for-your.html
  * Tacx T2018:
  [image]
  * Tacx T2028: I *think* this is also an ANTUSB2 in a bulky package and with an extension cable. If anyone can confirm that would be great.
  https://gearmashers.com/tacx-t2028-usb-ant-antenna-review-2018/

PROBLEMATIC:
- CYCPLUS: Known to cause problems/not work. Also sold under other names, like Anself. Anself reported to work fine previously but newly sold ones now can cause problems too. Anything that looks similar (with or without CYCPLUS label) has a significant chance of not working correctly.
https://images-na.ssl-images-amazon.com/images/I/51Q2TMVWlzL._AC_SY450_.jpg
- Gamin ANT USB1: older model that is not compatible with FortiusANT and other software. See https://support.trainerroad.com/hc/en-us/articles/206007776-USB1-vs-USB2-ANT-Sticks on how to tell the difference.

WouterJD, 2021-01-30:
@switchabl thanks; linked in the header message

ibooij, 2021-02-08:
To put some more nails on the Anself/Cycplus coffin: I recently tried to purchase an Anself dongle, but got shipped a Cycplus one. Having seen reports that Cycplus did not work at all using continuous scanning mode, I was surprised that this Cycplus dongle did recognize my devices and could receive all messages. However, my software was unable to send any messages using this dongle. It looks like ANT Extended messages might not be supported on it. 
I then tried using the same dongle with "normal" ANT point-to-point connections, which does work. 

My conclusion for now is that Cycplus has updated their dongles to make continuous scanning "kind of" work because they can now receive messages. They're still broken though. From the reports that I have been getting, the newer Anself dongles behave in the same way as these newer Cyclus dongles.

WouterJD, 2021-02-17:
@ibooij thanks for this explanation; 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

leovanr1, 2021-02-20:
It is a shame that I didn't read this carefully enough: now I bought a Garmin USB1 model that doesn't work. I can succesfully install the driver with the libusb inf_wizard, but it is still not found.

switchabl, 2021-02-20:
@leovanr1 That is unfortunate. In theory we could try to add support for the USB1 but it only has 4 channels so we would need to turn some features off. I think it is also pretty rare now so I am not sure it is worth it. Zwift, Rouvy, Trainerroad and RGT don't support it anymore either.

leovanr1, 2021-02-20:
I could have known if I had read your information more carefully. So don't worry about it.

yancart, 2021-03-17:
Hi, can anyone confirm if the Garmin USB ANT+ Stick 010-01058-00 will work ok with FortiusAnt?

[image]

Thanks

yancart, 2021-03-17:
Hopefully they are and about to buy as they are wanted to replace my CYCPLUS USB ones as I'm having alot of dropped comms...

BikeBeppe64, 2021-03-17:
[image]

Mine works without problem

[5 more replies]