WouterJD/FortiusANT #62: Feature suggestion: HTTP interface

mattipee, 2020-04-24:
I have just bought a Raspberry PI with the intention of running a headless FortiusANT on it.

One motivation is that it would be handy to be able to view/control FortiusANT over the network.

Indeed, some of my other ideas are around making visible the current state of the various internal variables (eg all the values provided in the ANT+ pages for a start), tweaking virtual speed curve, etc... and perhaps some live graphing - cadence/speed/power etc.

It would be possible to make it useful stand-alone in this way, eg manual grade/power controls, but the intention really is just to allow it to do it's job as an ANT->USB or ANT->ANT bridge, but to give visibility of internal state, and perhaps recent history, and to allow access to tunable parameters at runtime.

The python GUI may be more difficult to modify than it might be to present a few different pages/tables/graphs and/or API calls over HTTP. Also the GUI appears unstable in Linux due to threading issues - not unresolvable, but a separate issue.

I see an HTTP interface as being a good compromise - works both locally and remotely, and Python has good support for coding such things.

tobekas, 2020-04-24:
Hi. As an alternative or addition to http, MQTT would be really nice. It’s really lightweight and can be used better bidirectional.

This would make it possible to develop a GUI completely independent from the backend application, for example a Web application. I like the idea of ​​using an open standard (http API or mqtt). So everyone has even more freedom to access the data and do whatever, logging, database connection ...

WouterJD, 2020-04-24:
Hi guys, appreciate the thoughts and follow this with great interest.
FortiusANT is GUI-independant, which is clear that you can simply run without. 

My focus at this moment is to implement the miscellaneous tacx-variants as you see in the other threads. The funpart is to get the pedal-stroke analysis implemented. Enough to do and let's not forget to do a ride now and then.

Success and keep posting!

PS. @tobekas where are you located?

tobekas, 2020-04-24:
Hi Wouter,
I find it extremely sympathetic that you are interested in your discussion participants accordingly.
I‘m from the middle of Germany. I have been a racing bike hobbyist since I studied electrical engineering some years ago.  Lately less on the road and more active running. I also have an old iVortex trainer in the basement that I haven't used for a while.

I admire your ambition here for this project.  My job is in the field of software and control technology and so its not surprising that I am enthusiastic about some technical sides of this sport and equipments. I think it's great that projects like this bring hobby and work closely together. I haven't tested FortiusANT myself yet, but funnily enough, this is exactly what I missed 5 years ago in my training phase.

Therefore I am very excited to see what else is going to happen here. Such projects are great for realizing your own ideas and learning a lot yourself. I'm also a fan of clean code. @mattipee's attitude and suggestions are great. But yes... first of all everything should functionally run so far.  Later one can continue to work on "clean code" topics.

WouterJD, 2020-04-25:
Great, undust the iVortex and connect. I have the impression you are a valuable partner for the team doing some testing! #46
As you see on strava, I'm running and cycling; MTB and race. Give me a ping there if you like!

mattipee, 2020-05-01:
Another + for something like this is GoldenCheetah "train" interface has configurable charts, including any arbitrary web address. A wee interface to FortiusANT would fit there quite nicely.

WouterJD, 2020-05-01:
Well... I would like to see some kind of architecture here.

FortiusANT bridges between CTP's (Trainer Road, Zwift, Rouvy) at one side and non-standard Tacx trainers at the other side. It can run with and without user-interface, where the Console mode displays one line per second.

Especially with version 2.6 FortiusdANT.py shows three implementation of the user-interface: Console, GUI and Multi-processing. The console class `clsFortiusAntConsole` illustrates how another interface can be built

Please draw an image what the purpose is here...

mattipee, 2020-05-01:
The main point is if I want a GUI interface, but I don't have a keyboard/mouse/monitor attached to the device. I'd like to access the GUI over the network.

WouterJD, 2020-05-01:
Succes!

WouterJD, 2020-05-26:
Currently I consider an HTTP interface not in scope of FortiusANT ([ref](https://github.com/WouterJD/FortiusANT/issues/62#issuecomment-622437212)) so close the issue; can be reopened when tangible suggestion presented.

closed 2020-05-26 by WouterJD