WouterJD/FortiusANT #63: Support for ANT+ FE-C page 71

mattipee, 2020-04-24:
Yet to identify controller that uses this, but just as I go through the spec, seems like a sensible thing to cover off or at least record here. No immediate requirement.

WouterJD, 2020-05-01:
As unlucky as it may be, FortiusANT is derived from antifier and not build from spec.
Page 71 is in the spec and should be there; but no CTP (Trainer Road, Zwift, Rouvy, ...) use it and hence it would be unused code that cannot be tested...

As a product-developer I would close the request with "Left for future improvement"

mattipee, 2020-05-01:
Any messages can be tested using SimulANT from Garmin.

WouterJD, 2020-05-01:
The purpose is to use, testing is how to get there.

WouterJD, 2020-05-26:
msgPage71_CommandStatus implemented in i-Vortex pull request.
Please let me know whether it works.

closed 2020-05-26 by WouterJD