WouterJD/FortiusANT #64: Errors Using Rundown Test Through GUI

RodeTheAlpe, 2020-04-26:
All other aspects of GUI work (Locate, Start, Stop) but I'm getting the following errors when selecting Rundown (Python 3.8).  Paths to files below are correct, Thanks!

FortiusAnt.py -g

19:34:48,439: GetTrainer - Trainer found: 0x1942
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Users\Jeff\AppData\Local\Programs\Python\Python38\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Users\Jeff\AppData\Local\Programs\Python\Python38\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\temp\Downloads\FortiusAnt\FortiusANT-master\FortiusANT-master\pythoncode\FortiusAntGui.py", line 794, in OnClick_btnRunoff_Thread
    self.callRunoff()
  File "C:\temp\Downloads\FortiusAnt\FortiusANT-master\FortiusANT-master\pythoncode\FortiusAnt.py", line 162, in callRunoff
    return Runoff(self)
  File "C:\temp\Downloads\FortiusAnt\FortiusANT-master\FortiusANT-master\pythoncode\FortiusAnt.py", line 356, in Runoff
    PedalEcho, SpeedKmh, Cadence, Calibrate, False)
NameError: name 'Calibrate' is not defined

WouterJD, 2020-04-26:
please check that you have last .py files downloaded...

RodeTheAlpe, 2020-04-26:
That did it.  Thanks Again!

closed 2020-04-26 by WouterJD