WouterJD/FortiusANT #66: Error mid way through

iainhay, 2020-04-28:
Hey @WouterJD 

Adding this as a new issue, couldn't find anything similar and I have a couple of questions.

Now I'm connected (wahoo!) I was trying out my first proper ride (see strava [here](https://www.strava.com/activities/3364705078)). Mid way through I noticed power was constant at 135W regardless or stopping, or increasing my output. Checking terminal I got the following message:

```
06:52:31,903: GetTrainer - Trainer found: 0x1932
06:58:38,936: Processing longer than 0.25 seconds: 1.80
07:30:47,349: error: checksum incorrect checksum="01" expected="a5" data="a4 03 40 03 40 01 01"
07:30:47,349: ReadFromDongle error: checksum incorrect
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/iain/Downloads/FortiusANT-master 3/pythoncode/FortiusAntGui.py", line 830, in OnClick_btnStart_Thread
    self.callTacx2Dongle()
  File "FortiusAnt.py", line 165, in callTacx2Dongle
    return Tacx2Dongle(self)
  File "FortiusAnt.py", line 625, in Tacx2Dongle
    data = ant.SendToDongle(messages, devAntDongle, '', True, False)
  File "/Users/iain/Downloads/FortiusANT-master 3/pythoncode/antDongle.py", line 491, in SendToDongle
    data = ReadFromDongle(devAntDongle, drop)
  File "/Users/iain/Downloads/FortiusANT-master 3/pythoncode/antDongle.py", line 553, in ReadFromDongle
    while trv[skip] != 0xa4 and skip < len(trv):
IndexError: array index out of range
```
You can see this on the power in [strava](https://www.strava.com/activities/3364705078/analysis) too.]

So, 2 questions:

1. Is there something I need to do differently to avoid that error? After this error I FortuisANT showed no results so the error basically ended my ride

2. I tried to exit FortiusANT and restart, unplug the dongles but Zwift wouldn't pair when paused - I would need to end the ride and restart Zwift. Is there a way to get this reconnected to resume the rise?

Side note - I'm not convinced cadence is working properly as it doesn't seem to match my input - but this FortiusANT doesn't pick up Cadence so I think this is somthing with the trainer - will try and get that working of buy a new Cadence sensor to confirm.

WouterJD, 2020-04-28:
Sorry for this inconvenience, @iainhay , the Zwifting does not come easy for you.

- Crash was caused by a coding error, which is resolved in new version antDongle.py
- We got in this error, because of the incorrect buffer with an incorrect checksum
Question now is why this incorrect buffer was there in the first place.

After the crash, restarting is causing problems which makes me think that those problems caused the incorrect buffer.

All I can say from this distance is reboot the whole thing (power down dongle) and restart.
I will upload the new code.

Regarding Cadence sensor, check all connections first. Loose connections will not trigger an error but cause unexpected results.

iainhay, 2020-04-28:
@WouterJD thanks - I'll pull the latest and check back if it happens again 👍

iainhay, 2020-04-29:
Took a spin jut now and this no longer happens.

It did crash with the error:
 
`terminated by signal SIGSEGV (Address boundary error)`

But I just restarted and Zwift carried on - so that is fine for now!

Thanks again

WouterJD, 2020-04-29:
https://www.strava.com/activities/3369991133
I see your ride!
Software layers do not seem to be stable :-( 
I wanted to ask for the logfile but since I did not flush it in the past, it would miss the last part...
Does python stop execution or does it crash itself?

iainhay, 2020-04-29:
What do you mean don't seem to be stable?

It crashed itself - will check if it happens again though.

The cadence doesn't work on my trainer, so I think Zwift estimates it from the power? No other app picks up cadence at all - even FortiusANT, and my understanding is that Zwift uses this for the speed - which would make sense because when I hit a hill and I keep my power the same, Zwift says my cadence drops (which it doesn't) and so the speed drops. E.g. today I hit a hill and speed dropped to like 5kph and it took me ages to get up the hill, even though the cadence hadn't dropped at all.

WouterJD, 2020-04-29:
A ANT+ FE-C device sends power,cadence,speed and heart rate to Zwift.Zwift does not use heart rate because it expects a separate device.Zwift does not use speed but calculates the speed based upon the received power and the internal knowledge of slope, rider weight and internal formula's. Do realize that the wheel speed is irrelevant and direct drive trainers do not even have a wheel.
I do not know if there is no cadence what Zwift would do (never tried). 
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

iainhay, 2020-04-29:
Will grab a cadence and see how that goes, nothing is picking up cadence, not rouvy, tacx software, trainer road or FortiusANT - all the other results seem okay

WouterJD, 2020-04-29:
Good idea. Cadence comes in same call as other parts, so hardware is the likely cause.
Can you still get a trainer-wired cadence sensor, otherwise ANT+ part must be activated

iainhay, 2020-04-29:
Was just taking a look, the [pedal attachement](https://www.wiggle.co.uk/tacx-cadence-magnet/) is plastic, and open on one end so it looks like the magnet has fallen out...

Will order a new magnet just to see if thats the cause, if not a ANT+ one - someone is really making it difficult for me to ride!!

WouterJD, 2020-05-11:
@iainhay how are things going?

iainhay, 2020-05-11:
Hey @WouterJD 

My new cadence sensor arrived yesterday, it connects etc and will go for a ride today. 

Haven't got this error since either so happy for you to close. 

I saw the posts about the CYCPLUS dongles too, haven't had a chance to catch up but will have a look today too! 

Thanks again

closed 2020-05-11 by WouterJD