WouterJD/FortiusANT #67: Unable to communicate with i-Magic

martingeraghty, 2020-04-29:
Hello @WouterJD 

I hope you're well during these difficult times, you're doing a great job with this project.

I have installed the software on win10 (working on installing it on a Raspberry Pi, i'll let you know how it goes!) but i'm receiving errors from the communication to the FE. Please see the log below.

Any thoughts?

Thanks very much,

Marty

[FortiusANT_log.txt](https://github.com/WouterJD/FortiusANT/files/4554570/FortiusANT_log.txt)

WouterJD, 2020-04-29:
yep see #61 CYCPLUS dongle does not work well
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

WouterJD, 2020-04-29:
Welcome in the team; where are you from?

martingeraghty, 2020-04-29:
Thanks very much, I'm in the UK.

The issue is also apparent in manual mode with the simulated dongle.  The second half of the log file shows the only info coming from the FE is the r value. No speed etc... I can change the required power using the head unit but although it is displayed on the GUI and command line, it doesn't seem to be reaching the motor. Interestingly if I disconnect the motor from the head unit the r value drops to 0, I guess indicating that the 2 are communicating.

Hope this helps,

Marty

WouterJD, 2020-04-30:
Marty please join into issue #65 it deals the CYPLUS ANT subject, I will close this one?
I like to know about the spread of FortiusANT and if you can, give me a ping on Strava: https://www.strava.com/athletes/2885978

closed 2020-04-30 by WouterJD

WouterJD, 2020-05-13:
@martingeraghty  The powercurve is modified and ready for test.
I would appreciate you testing the iMagic, since I do not have that Tacx, so feedback from your side welcome

Beta-version available in
https://github.com/WouterJD/FortiusANT/tree/54-usbTrainer

Comments please add to
https://github.com/WouterJD/FortiusANT/pull/84

martingeraghty, 2020-05-13:
@WouterJD No probs, I'll test the power curve tomorrow (tired legs from alpe du zwift last night) from the beta version linked above. What feedback are you looking for? A subjective view or is there an output I can create for you?

WouterJD, 2020-05-13:
No specific output; a) does it work b) how does this feel?