WouterJD/FortiusANT #70: Feature suggestion: LCD display for headless Raspberry PI

mattipee, 2020-05-01:
A simple little 16x02 or 20x04 character LCD interface might be an easy addition for Raspberry PI users.

For example:
```
 --------------------
|Grade      -0.5 %   |
|Power     100   W   |
|Speed      46.5 km/h|
|Cadence    50   rpm |
 --------------------

 ----------------
| -0.5%  46.5km/h|
|  100W    50rpm |
 ----------------
```

mattipee, 2020-05-01:
https://github.com/mattipee/FortiusANT/tree/issue_70_rpi_lcd

WouterJD, 2020-05-01:
Somewhat in line with #61 
Why not make a clsFortiusAntLCD class?

And the idea is sympathetic, but the point is: "# TODO write to I2C LCD panel, for now, just Console output" and how many riders would have a RPI with LCD?

mattipee, 2020-05-01:
> how many riders would have a RPI with LCD?
Me, for one... ;)

WouterJD, 2020-05-01:
The best and most reliable answer! Let's go!

mattipee, 2020-05-01:
LCD arrives on Monday.

Idea was that a headless RPI-based FortiusANT is a nice, clean solution to getting Fortius and family connected to Zwift etc. Normal smart trainers have no visual interface.

But for a small embedded system like a Raspberry PI, small LCD panels are EVERYWHERE and cheap. Adding the very simplest interface with a few stats is therefore an easy thing to integrate into the codebase, such that any RPI user can plug in an LCD and it already works. I accept the number of users is likely vanishingly small.

I also thought that for the HTTP interface or whatever... actually, with the exception of perhaps some tuning parameters like virtual-speed curves and perhaps grade compensation or rider weight overrides, a simple thing to display remotely is a simple cycle-computer style display. That image could be served over HTTP or put onto an LCD. Same thing, kinda.

The 128x64 pixel LCD panels, or various OLED panels, colour/touch etc. would be nice, too. 128x64 pixels is plenty enough screen real-estate to populate with useful stats.

But I thought 20x4 character interface was a good deal easier to implement than a graphic panel, at least to start with.

WouterJD, 2020-05-01:
I heard of raspberry through thuisproject, @alanimus being the first. Give me some links to product & LCD to get some idea. Nice😉

mattipee, 2020-05-01:
The unit I have is:
[https://www.amazon.nl/Raspberry-Model-Mainboard-Polig-Microsd-Geheugenkaartsleuf/dp/B00LPESRUK/](https://www.amazon.nl/Raspberry-Model-Mainboard-Polig-Microsd-Geheugenkaartsleuf/dp/B00LPESRUK/)

It needs an SD card, HDMI cable, decent 5V 3A(?) power supply. Bundles and cases are available. Raspberry  PI 4 is more powerful, I linked the 3 B+ because it's older, cheaper, slower, and the one I have.

The LCD panel I bought is similar to this one:

[https://www.amazon.nl/GeeekPi-Character-Display-Protocol-Raspberry/dp/B07QNKCLJM](https://www.amazon.nl/GeeekPi-Character-Display-Protocol-Raspberry/dp/B07QNKCLJM)

Nothing special about that particular link.

WouterJD, 2020-05-01:
And let's work on the power-curve

mattipee, 2020-05-01:
Actually no - that first link I think is an even older model.

mattipee, 2020-05-01:
Anyway, you get the idea. It's a little ARM-based single board computer, with onboard Wifi, plus several other ports. It runs Linux.

mattipee, 2020-05-01:
And has GPIO pins for controlling several other things, like servos, I2C LCD panels, switches, sensors, etc... A little embedded system development platform.

WouterJD, 2020-05-01:
nice stuff. I have some laptops (Tacx, Zwift, develop&work) so will not add a Raspberry to the stack.
If you develop, I'll be happy to integrate.
Succes!

aegisdesign, 2020-05-02:
I was thinking about this earlier. I've a spare Pi 1 model B somewhere though eventually I think the Pi Zero might be a smaller cheaper component. They're £4-5 IIRC. Add a small LCD and a tiny case for a nice self contained addon to the trainer rather than using a USB port on a laptop.

WouterJD, 2020-05-02:
Fun!

mattipee, 2020-05-03:
My 2004 display arrived today.

https://www.circuitbasics.com/raspberry-pi-i2c-lcd-set-up-and-programming

- sudo raspi-config # find setting and enable I2C
- sudo apt-get install i2c-tools
- sudo apt-get install python-smbus
- i2cdetect -y 1

Still reviewing I2C LCD Python libraries and their licenses.

WouterJD, 2020-05-26:
Hi how is the clsFortiusANT_LCD development proceeding?
If any.

mattipee, 2020-05-26:
One for the winter. ;)

WouterJD, 2020-05-26:
OK I will leave that open then.

omedirk, 2020-11-09:
I just started working with the software, but I want to set up the RPi standalone as wifi hotspot.
Then connect with my phone to the RPi and have a website running as interface. No need for any wiring.

WouterJD, 2021-02-22:
I have a second Fortius and ordered a RPI4 for that. Learning case & a Fortius to implement.
So ... going through the learning curve :-)

WouterJD, 2021-02-27:
See https://github.com/WouterJD/FortiusANT/issues/192#issuecomment-787122844

WouterJD, 2021-03-04:
Hi @mattipee please refer to #237 where others have proceeded with the small display. I close this issue to avoid duplicates.
Please join in and share some pictures!

Also look at  https://github.com/WouterJD/FortiusANT/wiki/Raspberry-page-of-glory

closed 2021-03-04 by WouterJD