WouterJD/FortiusANT #71: FORTIUS IMAGIC T1902 AND ZWIFT

BMIL534, 2020-05-01:
System :WIN XP running on partition on IMAC using Virtual box
Have Fortius I-magic with T1902 head and TACX TTS3.
Cannot get RLV rides to load as system keeps crashing.
Want to use Swift if possible.
Any ideas?

WouterJD, 2020-05-01:
Hi BMIL, let's introduce ourselves first to avoid to be too virtual these days.
My name is Wouter and located in the Netherlands and I like to know who I work with; please drop a note on #14 or ping me on strava please.

Interesting question, Windows XP and virtual. 
What are RLV rides?

First step would be to install python & files and see whether you can run ExplorANT and FortiusANT at all. ExplorANT is the easiest to start with since you can then check whether you see ANT-dongles.

Then install tacx on USB and ANT dongle, refer to wiki how. I have already seen Windows XP but not in a virtual box yet!

Like to read your experiences.

aegisdesign, 2020-05-02:
I don't think this is related to FortiusANT. I had similar on one of my Macs. Make sure you have the VM set up with Direct3D support and 256MB of VRAM. If you're using VirtualBox then you have to use v6.0 as v6.1 was now removed Direct3D support for XP. I'm using v6.0.20 and RLV works in the Fortius 2.04 software. You also have to add '127.0.0.1    www.tacxvr.com' to C:\WINDOWS\SYSTEM32\DRIVERS\ETC\HOSTS otherwise it will bluescreen while trying to contact a non-existent Tacx server.

RLV are 'Real Life Video' btw @WouterJD. It's real video that speeds up or down depending on how fast you pedal. Kind of like Rouvy instead of Zwift.

@BMIL534 If you're running the python, be aware MacOS comes with Python 2.7. You need to install v3+. I would suggest using brew and pyenv to do so. See https://medium.com/python-every-day/python-development-on-macos-with-pyenv-2509c694a808

Rbarcan, 2020-05-08:
Hi. I’m interested in this too but need to say at outset I’m not very tech savvy! I have Tacx Fortius set up on an old Windows XP laptop which I don’t use for anything else. It runs the Tacx RLV DVDs fine. I recently bought some downloads from a guy in Germany trading as Real Life Videos after several emails back and forth. He advised me to try his demo first and that worked fine. Unfortunately the ones I bought don’t work! He says it’s to do with Codecs and advised upgrading to Windows 7. I’ve looked into that but it’s beyond my abilities. I’d appreciate any advice! 
Was also interested to see that the Fortius will work with Zwift but don’t understand quite how. If I buy a dongle for the laptop will it then connect with my IPhone 6S to run Zwift on the laptop? 
Please forgive my lack of expertise! Thanks

WouterJD, 2020-05-09:
Hi @Rbarcan welcome on the FortiusANT team!
Since I like to know who I talk with, please tell us where you are from, perhaps even ping me on [strava](https://www.strava.com/athletes/2885978)! I'm from the Netherlands.

Installation info can be found on https://github.com/WouterJD/FortiusANT/wiki#windows

FortiusANT is reported to run on Windows XP, so that should not be an issue.

How does it work?
- You connect your Tacx trainer to the windows PC
- The windows PC has one dongle and you run FortiusANT there (use -g flag to get the user interface)
- FortiusANT then connects to Zwift "by air"
- If your iPhone has ANT-support (not bluetooth!) then Zwift on your phone will detect a Tacx Neo trainer and you can go.

Success and if you have any questions, come back here!

Rbarcan, 2020-05-09:
Hi WouterJD

I’m in the UK. I’ve checked and it doesn’t seem that the iPhone has ANT- support. I do have a PC but that is in the house and my Fortius set up in the the garage. Do you know how close together the 2 devices have to be for it to work? Also am I right in thinking that I’d need 2 dongles, one in laptop and the other in the PC?

Many thanks for your help

WouterJD, 2020-05-09:
How does it work?

- You connect your Tacx trainer to the windows PC
- The windows PC has one dongle and you run FortiusANT there (use -g flag to get the user interface)
- FortiusANT then connects to Zwift "by air"
- RuN Zwift on the same PC and use a second dongle and will connect to FortiusANT"by air"

Two dongles needed
Please check wiki

WouterJD, 2020-05-10:
Also check https://github.com/WouterJD/FortiusANT/wiki/FortiusANT-Configurations---ANT--Master-and-Slave-devices

Rbarcan, 2020-05-10:
Thanks very much. Very useful information. 
Regarding the earlier suggestion of trying Rouvy, I’ve looked at that but website says Windows 10 minimum and on the laptop I’m using I only have XP.

WouterJD, 2020-05-10:
I run on windows 7, check the older version with ANTsupport. Not online now so cannot provide link

WouterJD, 2020-05-10:
Check #52

WouterJD, 2020-05-15:
@Rbarcan how are things going?

Rbarcan, 2020-05-15:
No progress I’m afraid, combination of uncertainty of my ability to cope with the tech aspects and relaxing of the rules so we can ride outside for longer. I’ll get there! Thanks for your help and interest.

WouterJD, 2020-05-15:
Succes!

closed 2020-05-15 by WouterJD