WouterJD/FortiusANT #72: Fortiusant Problems

JordiCostaLo, 2020-05-01:
Hello Everybody.
First of all, I would like to thank you for all efforts you are making to allow tacx fortiu users to ride. :-)

Well, I am absolutely sure, that I am doing something wrong when I try install fortiusant.
I have the following things to complete the installation without problems:
 - 2 dongles ANSELF USB ANT+
 - Tacx Fortius T-1942
 - TTS4 (software)
 - Windows 10

I have tried to follow your instructions, but every time I am getting lost.

I plug, USB from my TACT Fortius T1942 and my 2 dongles.. I try to run Zwift, but I am unsucessful.
Can you help me?

THANK YOU VERY MUCH EVERYBODY!!

WouterJD, 2020-05-01:
Hi Jordi, welcome in the team.

How are you these days and where are you from?

On the windows 10 PC where you connect the Tacx trainer **you must deinstall the TTS**, because you need other drivers. 
You probably already read https://github.com/WouterJD/FortiusANT/wiki#installation, check the steps under Windows/uninstall.

If you do not manage, just give another ping!

JordiCostaLo, 2020-05-01:
Wowwww!!!
As fast as the wind
Thank you very much for you prompt reply, and for your kindess.
I am from Barcelona - Spain. Nice to meet you. :-)

I have uninstalled TTS as your instructions, and Yes!! You are right, I printed your FORTIUSANT intructions (https://github.com/WouterJD/FortiusANT/wiki#installation)  and I have tried to follow the 2nd. Option.
Find attached drivers pictures after following instructions. Are drivers Ok??
[image]
After that  I've have executed the exe file ExplorANT, and I have got the following error (also attached as screenshot).

[image]

What I have done wrong, that time?? :-(

Thank you very much!!

WouterJD, 2020-05-01:
what dongle do you use?
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

JordiCostaLo, 2020-05-01:
Hi Wouter, 
this is Jordi, from Barcelona.

My dongles are ANSELF. I bought both of them in Amazon.

https://www.amazon.com/Forerunner-405-410-610-910-compatible-Entrenador-trainerroad/dp/B01M3VQP6Z/ref=sr_1_1?dchild=1&keywords=anself+ant&qid=1588364540&sr=8-1

How can I supply you more information about then?

Heel hartelijk bedankt

WouterJD, 2020-05-01:
Run the program with -d 127 flag which creates logfile and provide logfile please

JordiCostaLo, 2020-05-02:
Hi Wouter.
How are you? I hope you are fine!!
Run de program with -d 127 have to be done via Python?? or can be done at windows enviroment?

thanks!!

WouterJD, 2020-05-02:
commandline FortiusANT.py -d127 ...Or FortiusANT.exe -d127See wiki for more info
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

JordiCostaLo, 2020-05-02:
hello.
Find attached log, you requested.

thank you very much again

[FortiusANT.2020-05-02 19-24-06.log](https://github.com/WouterJD/FortiusANT/files/4568628/FortiusANT.2020-05-02.19-24-06.log)

WouterJD, 2020-05-02:
manufacturer=CYCPLUS, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)

Please join #65 which is investigating the subject
I will close this issue

closed 2020-05-02 by WouterJD

JordiCostaLo, 2020-05-02:
Thank you very much

I will join to this issue

Best regards!!