WouterJD/FortiusANT #73: FortiusANT with T1932 head unit and Tacx Flow i-Magic T1901 trainer

SpunkyOZ, 2020-05-02:
Hi, Marcus from Melbourne here,

I have been tinkering for a couple of weeks now to get my old **Tacx Flow i-Magic T1901** trainer with the **T1932 head unit** working with Zwift, and finally had some success. I've been using Antifier with the i-Magic power curve, however since the software is no longer in development, I wanted to switch to FortiusANT.

When I start the program, it correctly recognises the head unit, and I am able to start cycling in Zwift (ERG mode appears to be working fine). However what I have noticed is, that compared to Antifier, my power output seems to be much lower (as an example, around 54 W compared to closer to 100 W in Antifier).

This seems to indicate to me, that it could possibly be a problem with the power curve that is used. Hence I wanted to ask whether the combination of my head unit and trainer is currently supported in FortiusAnt, or whether I need to generate my own Power Curve as per one of the previous issues (or change any of the other settings).

If you could please let me know, that would be very much appreciated. On another note, it's great to see the Antifier project getting revived - especially in times of lockdown. Thank you for that!

Cheers,
Marcus.

WouterJD, 2020-05-02:
hi Marcus, you are right. I'm working on this topic with @mattipee refer to #51 an #54 please join those developments we might ask you to test.

SpunkyOZ, 2020-05-02:
Thanks so much, I will make sure to contribute to the tests. Really appreciate your quick response.

closed 2020-05-02 by SpunkyOZ

torqueNoFriction, 2020-10-30:
@SpunkyOZ hey I recently received a T1932 and was trying to find a way to hook it all up with Zwift. Do you have a set of rough instructions I can follow ?

WouterJD, 2020-10-31:
@torqueNoFriction I would suggest to start with the manual; see https://github.com/WouterJD/FortiusANT/blob/master/supportfiles/FortiusANTUserManual.pdf