WouterJD/FortiusANT #74: Clarification on HRM and SCS

mattipee, 2020-05-02:
@WouterJD 

Is the idea the FortiusANT can be asked to pair with an existing ANT+ HRM or SCS device, in order that it can use those results in preference to those from the Fortius?

I think I understand that in the case of HRM, where the Fortius can provide it, Zwift etc expect a different ANT+ device, rather than the value in page 16?

In which case FortiusANT needs to emulate an ANT+ HRM device too. And in the absence of either, it makes sense to be able to suppress HRM from the Fortius completely.

SCS, I guess you would just overwrite the values from the Fortius with the values from a separate ANT+ device.

Is this understanding correct? Or can you clarify?

I guess my question is: WHY? Why is FortiusANT pairing with other devices to then pass the values onwards? Does it enable functionality that you can't get just by selecting in the client software which device(s) you want to pair with? At the moment, it looks like unnecessary code but I'd like to understand why I might be wrong.

WouterJD, 2020-05-02:
**Let's start with SCS (Fortius ANT = slave = listening to Speed and Cadence Sensor)**
@alanimus started with Raspberry and wanted SCS support; I built the basic hooks but got no further feedback. So developments unknown. @iainhay has a faulty cadense sensor and if he does not manage to repair an ANT+ cadence sensor might be an idea.
For now: "dead code". Because of lack of channel numbers (must be 0...7) I parked them at -1 in the last version.

**Then HRM (Fortius ANT = master = broadcasting heartrate)**
Old trainers pair with non-coded HRM and transfer the heartrate on the USB-dongle. Therefore antifier built two master devices: not only FE-C but HRM as well. (Even though the FE-C interface communicates the heartrate)

Modern trainers do not listen and/or do anything with HRM but provide a master FE-C on ANT, where heartrate=0. CTP's (Zwift, Trainer Road and alike), being ANT+ slave devices, simply pair with ANT+ devices: FE-C and HRM. CTP's simply ignore the heartrate from the FE-C interface, which is sloopy in my opinion, but realistic regarding the actuality of the trainers.

This is my (the?) explanation for the existance of the master HRM device in antifier and FortiusANT bridging the connected HRM to the CTP.

**And finally the HRM slave (Fortius ANT = slave = listening to HRM)**
FortiusANT GUI displays the heartrate as one on screen.

But if your HRM (paired to the trainer) isn't there anymore, the heartrate on the FortiusANT screen stops. And that's a pity and it can be brought alive again with the -H0 flag. 0 pairs with any HRM, a number pairs with a known HRM. 

Technically not necessary, just meant to keep the heartrate blinking AND the experiment to build an ANT+ slave (the other two are masters).

---
I guess this answers the question, if not I will hear from you - closed for now.

closed 2020-05-02 by WouterJD

mattipee, 2020-05-02:
Thanks for taking the time to explain @WouterJD ! I'll have a play - this was prompted by looking at refactor steps and in particular, the ReceiveFromTrainer() return tuple being pushed inside a trainer object... the HeartRateT, SpeedKmhT etc... (T suffix being Trainer, I assume), and the way in which the loop works and overwrites values for the next time round the loop... just trying to unpick it and make a sensible proposal for refactor.

WouterJD, 2020-05-02:
Yep. T=trainer and Temp :-)