WouterJD/FortiusANT #75: TACX iMagic T1900 Support

TrevorHunter, 2020-05-02:
Hi Woulter, I have an old T1900 which I still use with TTS 3.13.0 using a Win10 32 Bit VM.

Do you know if this model could be supported? I've tried it, but doesn't seem to be find it? Is there anything this end I can do?

Thanks Trev...
[image]

mattipee, 2020-05-02:
If TACX software has control of the device, then FortiusANT won't find it. FortiusANT can't use the driver that TTS has installed. There are instructions on the wiki to disable the TACX driver and use libusb driver instead. It can be easier just to uninstall TTS if you still have the installer to put it back if/when you want it, but it is possible without uninstalling.

See here: https://github.com/WouterJD/FortiusANT/wiki

TrevorHunter, 2020-05-02:
That's great thank you :). I can now see my T1900 with my x64 Windows10 installation and go back my 32Bit VM, should I wish to use it with the TTS3 software again.

mattipee, 2020-05-02:
@TrevorHunter good result, having two separate setups sounds convenient.

WouterJD, 2020-05-02:
@mattipee look at the dongle that is used here: cycplus. Is this the first one working?

mattipee, 2020-05-02:
@TrevorHunter 
Have you actually ridden with Zwift or similar using FortiusANT? There is a known misbehaviour of CYCPLUS dongles that we are working on. Interesting to know if you a) have actually ridden it without issue, or b) if you experience the same crashing that other CYCPLUS users do.

https://github.com/WouterJD/FortiusANT/issues/65 for more details

TrevorHunter, 2020-05-02:
Sorry, not got that far yet, so far just seen that it can talk to the imagic and turning the pedal by hand shows comms. 

I am getting the following errors in the console though.

J:\FortiusANT-master\WindowsExecutable>FortiusANT.exe -g
Load firmware C:\Users\WILLOT~1\AppData\Local\Temp\_MEI170402\tacximagic_1902_firmware.hex for DEVICE ID 3561:1902 on Bus 000 Adresse 001
....Stop CPU
....Start CPU
17:35:41,455: GetTrainer - iMagic head unit initialised
17:35:44,975: devAntDongle.write exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: A device attached to the system is not functioning.\r\n\n'
17:35:44,977: devAntDongle.read exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: A device attached to the system is not functioning.\r\n\n'
17:35:44,978: devAntDongle.write exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] reaping request failed, win error: The I/O operation has been aborted because of either a thread exit or an application request.\r\n\n'
17:35:44,978: devAntDongle.read exception: [Errno None] b'libusb0-dll:err [submit_async] submitting request failed, win error: A device which does not exist was specified.\r\n'

I saw briefly about the issue this morning and tried the branch suggest in the link 'FortiusANT-issue_65_cycplus.zip', but still got the same error.  Ill try a proper test over the next few days
[image]

mattipee, 2020-05-02:
@TrevorHunter  please see https://github.com/WouterJD/FortiusANT/issues/65

Basically, my cyclplus branch was a statement of intent - a work in progress.

Elie found that suppressing the dongle reset improves things but all of us who have tried get intermittent resets after that.

Probably close this thread as known issue re: TTS driver, and move CYCPLUS-related testing to #65

WouterJD, 2020-05-02:
Hi trever, I agree wait @mattipee comments, we are working on the issue.
In the meantime: where are you from? I like to know who we work with and how FortiusANT spreads... If you like, ping me on strava!
https://www.strava.com/athletes/2885978

closed 2020-05-02 by WouterJD

WouterJD, 2020-05-13:
@TrevorHunter  The powercurve is modified and ready for test.

Beta-version available in
https://github.com/WouterJD/FortiusANT/tree/54-usbTrainer

Comments please add to
https://github.com/WouterJD/FortiusANT/pull/84