WouterJD/FortiusANT #79: Transmit PWR profile separately

tobekas, 2020-05-09:
Hi Wouter,
is it possible to transmit Ant+ Device Profile PWR (and maybe S&C) in addition to FE-C?
I tested FortiusANT in simulation mode and tried to pair it with my Edge 800. It could find the HRM channel but unfortunately no power sensor. I think this is because the edge 800 does not support FE-C. However it would be great, if it could be used to track the workout with power information.

WouterJD, 2020-05-09:
Will do after current developments

WouterJD, 2020-05-18:
Check in on #87

tobekas, 2020-05-20:
Do you mean, this feature suggestion will be tracked in #87 or that is should already be implemented in the lated version?
With Version 3.0 a cannot find a separate "ANT+ PWR" channel with my Edge 800 by now. At least not in simulation mode (-s flag).
My original feature suggestion was to additionally broadcast a plain power channel (ANT+ PWR profile) in addition to the FE-E channel, which already includes power. Or even beyond that to additionally broadcast the S&C channel (speed and cadence), if possible.

WouterJD, 2020-05-20:
Hi @tobekas Thanks for responding.

I sure understood your request and from #87 I understand that Edge pairs with FE-C channel.
First step please check what's going on in #87.
I have no edge so cannot test that.

If edge does not pair, I'm happy to create PWR profile; in that case please reopen this issue.

tobekas, 2020-05-20:
Edge 820 does pair with FE-C but Edge 800 and some other old generation devices don’t. In #87 it’s about pairing with an Edge 820. He also mentions an 800, but this is used as a reference measurement paired with an independent power meter and speed sensor for comparison.

My suggestion is to add PWR profile, to enable pairing with older devices. As far as I know, new generation Smart Trainers also provide PWR in addition to FE-C. 

In addition to using Swift..., this would also make it possible to simply record power data with the Edge (for workout history in garminConnect, Strava...) during boring training only with the head unit, in case you don't have a power meter on your bike.

WouterJD, 2020-05-20:
ok

WouterJD, 2020-06-12:
@tobekas SCS and PWR profile are ready for test; refer to https://github.com/WouterJD/FortiusANT/pull/95.

tobekas, 2020-06-13:
Hi. Just tested your branch (in simulation mode) and it works like a charm. SCS and PWR Sensors now appear and can be paired with my Edge 800. Great!

WouterJD, 2020-06-13:
Great, thanks for testing👍👍

WouterJD, 2020-06-14:
Implemented

closed 2020-06-14 by WouterJD

BikeBeppe64, 2020-06-17:
> Hi. Just tested your branch (in simulation mode) and it works like a charm. SCS and PWR Sensors now appear and can be paired with my Edge 800. Great!

I did It in a simulation mode too but had problem in normal mode.