WouterJD/FortiusANT #80: Windows executable, don't see interface

Dedacciai, 2020-05-09:
Hello folks, congrats for the good work ! So nice to use again my old I-Magic 1901 Trainer :)
When I open windows executable fortiusant.exe     I only have a blackscreen with info of trainer.
But how can I get the interface with buttons ? Need Help on that.

Thanks in advance

From Quebec, Canada :)

WouterJD, 2020-05-09:
hi @Dedacciai 

Welcome in the FortiusANT world.
You found the software, not the wiki😉

sorry for perhaps all-in-one-page info, to be improved! Check https://github.com/WouterJD/FortiusANT/wiki#command-line for options.

-g gives the GUI!
Succes!

RodeTheAlpe, 2020-05-10:
The windows executable didn't bring up the gui for me.  I needed to run it from the Python script with the -g tag -> FortiusANT-master\pythoncode\FortiusAnt.py -g
If you don't have Python on your machine it's an easy and free install.  I created a shortcut with the above command and it worked perfectly (once I reinstalled the latest version of FortiusANT).

Dedacciai, 2020-05-10:
I m so newbie at this new thing... i ve been able to load the GUI. But it creates each time a 60mo temp folder.  Ibalso downloaded the python  3.8. But i m not able  to use it. Don't know how. Grrrrrrr. I ll need a how to i think hahahaha

WouterJD, 2020-05-10:
The WindowsExecutable extracts itself into a folder and then executes; this takes time and space but avoids installing python. When extracted it works as good as python.

What is the problem with the temp folder?
Unless it poses a real problem, I would suggest to use the executable.

Success and just let us know your experiences!

Dedacciai, 2020-05-10:
Thanks guys for your help :)
I'm trying to get it works to let my I-magic to my father. who is completely zero in computer. Fact is that each time I open executable, it makes a new folder in temp. So only for yesterday with all the test I made, I had 2go of open executable hihi. For me not a big deal to erase that temp, but for him, it's gonna be a pain in the a** hahaha :) I'll try to make him a shortcut with a temp eraser.

Very nice software :) ... Personaly I switch my I Magic 3 years ago to get the first Neo generation. Love it :) Will try to install on a windows 10 the tack desktop app for my dad if he doesn't want to try a shot at Zwift ;0

Sorry for my english too folks. French Quebecois here

WouterJD, 2020-05-11:
Hi @Dedacciai 

Good day! English is OK and great that you make your dad zwifting!

I think that the temp-files are removed automatically by the executable.
One thing: do not "kill" the program by closing the consol (black screen) but close the program either with the cancel button on the console or ALT-F4 on the keyboard or the close-button (righttop X) on the FortiusANT console.

Great to see that the program made it to Quebec! If you like, give me a ping on [strava](https://www.strava.com/athletes/2885978)!

WouterJD, 2020-05-14:
Hi is your issue solved?

Dedacciai, 2020-05-14:
I Didn t have time to check it out. Will try again this week-end. Thanks
for asking 😊  back to work this week ....

antmonkey, 2020-05-17:
I am also a noob at this, and I can't seem to get the graphic interface.
I'm using the FortiusAnt windows executable, but I have no idea where I'm supposed to be entering the commands listed in the wiki (such as -g to start the GUI). 
When the FortiusAnt text interface opens it enters the calibration mode for my fortius trainer but does not seem to complete the calibration. It does not allow me to type any text, and only seems to recognize the ctrl+c command to cancel the calibration.

Any guidance you could provide would be much appreciated.

Dedacciai, 2020-05-18:
You have to open from search in Windows command CMD.EXE

In cmd mode. Write

Fortiusant.exe - g

It should open the interface 😊

antmonkey, 2020-05-18:
Thank you for your help.

It actually took a few more steps/commands but I would not have figured that out without your suggestion. (had to type in "cd" and then the file path to the folder containing the .exe file, and then "start FortiusANT.exe -g").

Once I figured that out, I was also able to figure out how to make a shortcut with the following "target" so I don't have to type in the CMD every time I want to open FortiusANT: C:\Windows\System32\cmd.exe /c cd C:\Program Files\FortiusANT-master\WindowsExecutable & start FortiusANT.exe -g

Hopefully this will help me get my trainer calibrated. I will post here if I have any more issues.

Thank you so much for providing this resource!

WouterJD, 2020-05-18:
Sorry that starting-up is a bit complicated. Will be resolved in version 3.0 which will be released soon.

WouterJD, 2020-05-18:
@antmonkey welcome on the FortiusANT team; always curious who I am talking with! Where are you from and what Tacx do you use? If you like, ping me on [strava](https://www.strava.com/athletes/2885978)!

antmonkey, 2020-05-18:
I'm guessing this is unrelated to the previous topic, but I am having trouble calibrating my Fortius. When I select start, it enters calibration mode and asks me to start pedaling but in FortiusAnt it says:
23:33:19,483: Connected to Tacx Trainer T1932
23:33:52,641: devAntDongle.write exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
23:33:54,142: devAntDongle.write exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
23:33:55,643: devAntDongle.write exception: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'

And if I start pedaling, the motor kicks in and gets the wheel up to 20 km/h but the calibration never ends and FortiusAnt lists a bunch of errors.

Any ideas on what I'm doing wrong?

antmonkey, 2020-05-19:
> @antmonkey welcome on the FortiusANT team; always curious who I am talking with! Where are you from and what Tacx do you use? If you like, ping me on [strava](https://www.strava.com/athletes/2885978)!

I am from Toronto, Canada, and I am using an old Tacx T1932 that I am trying to get operational.

WouterJD, 2020-05-19:
Based upon the error I woudl expect you have CYCPLUS ANT-dongles, frequently showing this behaviour. See #65.
[Version 3.0](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-now-being-tested) will recover from that situation (as good as possible) and I expect to release tonight after some final tests.

To verify: what dongles do you have and -in case of errors- always attach a logfile (-d127 flag on command line) for inspection.

Succes!

antmonkey, 2020-05-19:
I am using Dynastream Innovations Dongles, if that makes a difference. 
I re-attached my cadence monitor because I read that not having it attached can lead to motor stoppage (even though I can't find that part that is supposed to attach to the crank arm) and now I get a difference result.

After starting the calibration, it now stops saying:

18:35:29,625: FortiusANT started
18:35:29,626: -b 2.096,50/34,15/25
18:35:29,626: -d 127 (0b1111111)
18:35:29,627: -f 200
18:35:29,627: -g
18:35:29,628: -H None
18:35:29,628: -p 1.0
18:35:29,628: -r 150/100
18:35:29,629: ------------------
18:35:30,542: FortiusAnt GUI started in child-process
18:35:36,077: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
18:35:36,085: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
18:35:36,651: Connected to Tacx Trainer T1932
18:35:54,320: Calibration stopped because of unexpected resistance value.

I would add the log file, but I'm not sure where it was saved.

antmonkey, 2020-05-19:
I tried it again, and now it's continuing calibration perpetually again...

I found the log files, here are the most recent ones.
[FortiusANT.2020-05-19 20-31-59.log](https://github.com/WouterJD/FortiusANT/files/4652981/FortiusANT.2020-05-19.20-31-59.log)
[FortiusAntGUI.2020-05-19 20-32-00.log](https://github.com/WouterJD/FortiusANT/files/4652982/FortiusAntGUI.2020-05-19.20-32-00.log)

WouterJD, 2020-05-19:
[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020) and starting the executable should be easier for windows users.
Of course of there are any issues, let me know.

WouterJD, 2020-05-19:
Perhaps there is something wrong with calibration, start with `-n flag` to skip that step.
Will have a look at logfile tomorrow.

antmonkey, 2020-05-24:
Hello again, and sorry for the delayed response. I've been busy with other things this week.

I tried to calibrate the trainer again, but ran into the same "unexpected resistance" error. 
(I attached the two log files that were generated, including the one for the GUI)

I also ran the trainer without calibration, and it brakes after a few seconds. My online research tells me that the braking issue could be an issue with the cadence monitor (I cannot find the piece that attaches to the crank arm). This issue may also be what is affecting the calibration. I will look into replacing that piece (unless you think it is something else). 
(I also attached the two log files that were generated for this attempt. They have the (no cal) indicator)
[FortiusANT.2020-05-24 01-31-27.log](https://github.com/WouterJD/FortiusANT/files/4672820/FortiusANT.2020-05-24.01-31-27.log)
[FortiusANT.2020-05-24 01-36-06 (no cal).log](https://github.com/WouterJD/FortiusANT/files/4672821/FortiusANT.2020-05-24.01-36-06.no.cal.log)
[FortiusAntGUI.2020-05-24 01-31-28.log](https://github.com/WouterJD/FortiusANT/files/4672822/FortiusAntGUI.2020-05-24.01-31-28.log)
[FortiusAntGUI.2020-05-24 01-36-07 (no cal).log](https://github.com/WouterJD/FortiusANT/files/4672823/FortiusAntGUI.2020-05-24.01-36-07.no.cal.log)

antmonkey, 2020-05-24:
I attached a magnet from one of my other bike's sensor, and tried again but it still fails (still possibly an error with the cadence sensor though). I have attached the logs.
[FortiusAntGUI.2020-05-24 03-15-10.log](https://github.com/WouterJD/FortiusANT/files/4672894/FortiusAntGUI.2020-05-24.03-15-10.log)
[FortiusANT.2020-05-24 03-15-08.log](https://github.com/WouterJD/FortiusANT/files/4672895/FortiusANT.2020-05-24.03-15-08.log)

WouterJD, 2020-05-24:
Hi @antmonkey 

I see in FortiusANT.2020-05-24 03-15-08.log (last one) that when you start calibration, the speed is returned but no cadence. It sems that calibration does not work the same way as for Fortius. So skipping is a good idea at this moment.

What I think now is:
- Until now I saw iMagic with headunit 1902 and therefore I know it's an iMagic
- AND I saw Fortius with headunit 1932

iMagic / 1902 has the so-called legacy interface with one set of formula's (setA)
Fortius / 1932 has the so-called newer interface with another set of formula's (setB)

It may be that iMagic/1932 has the newer interface but that the setA formula's must be used.

Question:
1. do you use the windows executable or do you use python native version?
2. would you be able to modify the python code/

antmonkey, 2020-05-25:
The trainer has a Fortius sticker on it, but who knows what the tech is?

WouterJD, 2020-05-25:
I think confusion is caused by the combination of @Dedacciai and @antmonkey writing on the same issue; where @Dedacciai reported having an iMagic.

So: let's get back to the basics: please describe what configuration you have, if possible include some photo's.

Dedacciai, 2020-05-25:
Mine is a i-magic with motor 1901 written.  But controller is 1902 green

WouterJD, 2020-05-25:
@Dedacciai thanks: 1902 controller (for me) implies iMagic with related interface and formula's.
1932 (for me) implies Fortius with other interface and formula's.

Untill now, I have nothing else to know what Tacx I talk to as the headunit ID (1902, 1932).

antmonkey, 2020-05-25:
The software recognizes mine as a T1932 and the label says Fortius. I have attached pictures of my setup, hopefully that helps. 

[image]

[image]

[image]

WouterJD, 2020-05-25:
That's great, it is the same devices as I have! [See this image](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAnt.jpg)
Apart from minor details, like 110V where I have a 220V :-)

So we are talking Tacx Fortius 1932.
Unless you have a hardware error, we should be able to fix.

I was busy with Tacx i-Vortex (which is now also supported, #46)
I will do a quick test without cadence sensor, to see whether that influences the environment.
Have to do some other business now, will get back to it asap.

WouterJD, 2020-05-25:
I have done a test and disconnected the cadence sensor from the main unit.
Calibration and riding works, just that no cadence is returned.
So it seems is informative only and not used by the Tacx Fortius nor by FortiusANT (other than broadcast, display and pedal stroke analysis).

Check all connections carefully, though. Loose connections do not (always) cause an error but misbehaviour.

Ciao and succes!

WouterJD, 2020-05-25:
I have looked at your FortiusANT.2020-05-24.03-15-08.log again.

Active trace starts at `03:15:54,595: Calibration stopped because of unexpected resistance value`

Then  you see lines like
`clsTacxUsbTrainer.SendToTrainer(T=0, M=1, P=100, G=0, R=2142, W=85, PE=0, S=20.0, C=0)`
This means that TargetPower=100Watt, Resistance=2142 and Speed=20 km/hr.

`SendToTrainer(T=2, M=1, P=100, G=0, R=2923, W=85, PE=4, S=14.6, C=0)` resistance goes up because speed goes down. That is as expected.

`SendToTrainer(T=2, M=1, P=100, G=0, R=1984, W=85, PE=0, S=21.6, C=0)` resistance goes down because speed goes up.

`SendToTrainer(T=2, M=1, P=100, G=0, R=5115, W=85, PE=4, S=8.4, C=0)` resistance goes up because of low speed (8.4 km/hr).

Below 10km/hr the resistance goes down from 6000 --> 1500; below 10km/hr the Fortius does not perform well at higher loads.

Somewhat later, the speed drops to zero and hence the resistance is also set to zero.

If I understand well, no Cycling Training program is yet connected, so the TargetPower is fixed at 100Watt.

From the logfile, it all seems quite normal.
What is you FTP if I may ask?

Keep up the good work!

antmonkey, 2020-05-26:
I guess the Tacx program was not detecting the trainer properly bc zwift seems to be working so far (just an initial try, haven't done any workouts to test the resistance but I will do that tmrw). 

Thank you for all of your help!

WouterJD, 2020-05-26:
👍

WouterJD, 2020-06-08:
No follow-up in two weeks.
If issue needs to be reopened, feel free to do so.

closed 2020-06-08 by WouterJD