WouterJD/FortiusANT #81: Road feel - ANT+ page 252

mattipee, 2020-05-10:
Not to distract you from more important work...

Looks like page 252 from Zwift conveys road surface information:

Eg.
"00 fc 00 00 64 00 00 XX YY 00"

XX is surface, YY looks somehow related to speed?? frequency?? (looks clipped at max val 0x64, see it lower for lower speeds). The other 0x64 seen at bytes[4] could be intensity, just a guess, and just that Zwift always sets to 100?? Watopia Flat Route after the tunnels there's boardwalk (0x09) and gravel (0x07). Cobbles are 0x05. If I remembered those values the right way round.

Now that you have the ability to increase the frequency of polling - I wonder whether we could get any road feel coded as a resistance modification function eg... pulsing, or oscillating. Obviously frequency is limited, but interesting idea, do you agree?

WouterJD, 2020-05-10:
I have been thinking of that , would be a real bommer.
How would the Tacx Neo have implement this feature?
Either setting resistance every 0.0xx seconds (we could do such a thing, question is whether Fortius reacts quick enough) or by implementing roadfeel in the trainer itself.

As you say; first things first but fun to try!

You could do some experiments; use the runoff code with PSAon and vary the resistance and see what happens. If you ride over wooden planches, x km/hr, 20cm wide, it can be calculated what frequency the resistance must bump.

Have fun, curious to hear!

mattipee, 2020-05-10:
Even modifying resistance to +/- 32k (ie max swing), there's not a sharp enough 'edge' - somewhat lumpy at best, even with virtual flywheel at 0. Non-starter.

closed 2020-05-10 by mattipee