WouterJD/FortiusANT #82: Device T2220 : error reaping request failed

emarinet, 2020-05-11:
Hi Wouter,
I have a tacx VR flow Multiplayer T2220. I made the configurations for libusb-win32-bin-1.2.6.0. When I try to launch executable I get the following error: "reaping request failed, win error: A device attached to the system is not functioning." what could be the problem?

Thanks.
Enrico.

WouterJD, 2020-05-11:
hi Enrico; I'm Wouter from the Netherlands welcome to the FortiusANT community! I'm always curious where the product goes to and therefore like to know where you are located. If you like, ping me on [strava](https://www.strava.com/athletes/2885978)!

I have not heard about this error before in the FortiusANT context, so google must be your first friend here.

Please explain in more detail what you have connected, did Tacx Training Software work?
This is difficult to advice from distance!

Succes, perhaps somebody on github has an idea?

mattipee, 2020-05-11:
@emarinet Hi, Enrico.

"A device attached to the system is not functioning" can be one of the errors observed when using certain CYCPLUS ANT+ dongles. Can you first confirm what dongle you are using? And what FortiusANT displays either in the GUI or the logfile for the dongle which it connects to?

Matt.

emarinet, 2020-05-11:
Hi Matt and Wouter,
I have tacx interface T1932 with T2220 trainer, I use them with the TTS software 3.5 windows7.  I use two CYCPLUS ANT + dongles.
My goal is to use zwift software. I tried with Antifier, it worked  correctly in zwift for 5/10 minutes then it gave, in the cmd window Antifier, the following error:
USB WRITE ERROR [errno None] libusb0-dll: err [_usb_read_async] reaping request failed, win error: A device attached to the system is not functioning.
Then I tried with FortiusANT but the problem is the same.
Do you think the problem is in the CYCPLUS ANT+ dongles.?

Thanks.
Enrico.

mattipee, 2020-05-11:
@emarinet It's a good bet. See here: https://github.com/WouterJD/FortiusANT/issues/65

There is a best-effort workaround coming in the next release which may help (simply, it will reconnect the dongle if/when it drops), but results may be mixed.

WouterJD, 2020-05-12:
@emarinet you could use the CYCPLUS branch which contains the being-tested code for the CYCPLUS. The code tries to compensate the erroneous behaviour of the dongle, please let us know whether it relieves your issue.
Others get new dongles to replace the erroneous ones, but that does not help you at this moment.

https://github.com/WouterJD/FortiusANT/tree/65-CYPLUS

WouterJD, 2020-05-12:
Please comment on #65 so I close this issue

closed 2020-05-12 by WouterJD