WouterJD/FortiusANT #83: Mal installer

regis65, 2020-05-11:
Bonjour,
J ai essaye d installé FortiusAnt sur mon pc mais je n y arrive pas j ai lu les instruction et sa fonctionne pas 
Mon home trainer Tacx Fortius
Mon pc windows 10
Hp i5 
6Go ram
Carte Graphique 2go
Pouvez m aider merci

WouterJD, 2020-05-11:
"J ai essaye d installé FortiusAnt sur mon pc mais je n y arrive pas j ai lu les instruction et sa fonctionne pas"

Please provide more information; what happens, what error-code is given?

regis65, 2020-05-11:
Salut,
en fait j essaye d installer le nouveau driver sur Vr interface il veux ce change

WouterJD, 2020-05-12:
Regis
Pour vous pouvoir aider j'ai besoin de plus informations.
- Quel actions vous faites
- Quel reponses, message du système

Voyez le page wiki pour informations sur installer é desinstaller les drivers de Tacx: https://github.com/WouterJD/FortiusANT/wiki#windows

regis65, 2020-05-12:
bonjour 
j ai reçu a l installer sur un pc de 32 bits sa a l air de fonctionner mais le souci c est que j ai les kilomètres,cadence........ directement sur la page noire et il normal ?

WouterJD, 2020-05-12:
Oui c'est normal, vous n'avez pas specifié des paramètres.

Si vous lancer FortiusANT avec -g, l'interface d'utilisateur est ouvert: `FortiusANT.exe -a -g `

regarder le wiki: https://github.com/WouterJD/FortiusANT/wiki#command-line

Succes!

regis65, 2020-05-12:
Bonjour,
Comment mettons les "arguments facultatif ".
Merci

WouterJD, 2020-05-12:
Il faux creer un petit fichier FortiusANT.bat
https://github.com/WouterJD/FortiusANT/blob/master/supportfiles/FortiusAnt.bat

avec les regles
```
FortiusAnt.py -g -a
pause

```
et puis on peux executer cette .bat

regis65, 2020-05-13:
En fait le programme se lance dans bios mais j aimerais avoir l interface
avec les commande du guidon

WouterJD, 2020-05-13:
Version 3 supportera lancer la programme sans ligne de commande.
Pour ce moment il faux creer un .bat

regis65, 2020-05-13:
Apres avoir fait le fichier .bat dans quel dossiers faut il l installe .

WouterJD, 2020-05-14:
Hi @regis65 

Le fichier `FortiusAnt.bat` avec les regles
```
FortiusAnt.py -g -a
pause
```
peut-être installé dans la meme dossier comme les fichier *.py ses mêmes.

Succes!

regis65, 2020-05-14:
Le fichier .bat je l es installé dans le dossier Pythoncode je lance
fortiusant dans le dossier windowsExecutable et toujours pareil l ecran
bios avec les renseignement du hometrainer avec la vitesse cadence mais je
voudrais installé l interface.
Tres beau programme mais compliqué a installé et paramétré 😢😭
Merci

WouterJD, 2020-05-14:
I disagree that the program is difficult to install; a bat-file with a commandline should not be complicated.

As said, add the -g flag so that the gui is displayed.

Version 3.0 will be easier to use.

regis65, 2020-05-15:
je voudrais savoir si sans mettre l interface mon home trainer peux fonctionner parfaitement sous sufferfest, binringvr, zwift, tacx desktop.... ou y aurais t il d autre paramètre a rajoute car je n y arrive pas a faire fonctionner l interface.
merci

WouterJD, 2020-05-15:
Pour faire le functionner c'est peutetre une idee de demander in Ami avex qc experience avex windows.
Quando ça marche ça marche avec Trainer Road, Zwift, Rouvy, ... tous programmes sopportants ANT FE-C.

regis65, 2020-05-15:
D'accord merci

WouterJD, 2020-05-19:
[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020) and starting the executable should be easier for windows users.
Of course of there are any issues, let me know.

regis65, 2020-05-20:
Merci mais comment fait on pour le téléchargé Sur le pc

WouterJD, 2020-05-20:
Vous pouvez remplacer le FortiusANT.exe sur vôtre système avec:
https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe

Le fichier FortiusANT.bat est toujours valable:

```
FortiusAnt.exe
pause
```
suffit; je l'ai fait plus simple - especialement pour vous! :-)

regis65, 2020-05-20:
Je vous remercie.
Sportivement Régis

regis65, 2020-05-22:
Salut,
L'Interface fonctionne il n y a plus qu'attendre l arrivée de mes clés Ant+
que j ai commandé sur le net.
Je te tiendrais au courant si sa fonctionne correctement.
Et encore merci bon boulot
Sportivement
Régis

WouterJD, 2020-05-22:
👍

WouterJD, 2020-06-08:
No follow-up in two weeks.
If issue needs to be reopened, feel free to do so.

closed 2020-06-08 by WouterJD

regis65, 2020-06-08:
Daccord merci,je n es toujours pas reçu les clés ANT+

WouterJD, 2020-06-08:
Cést bien dommage!
Tu les as commandé deja depuis combien des temps?
En cas je pourrais les commander et envoyer!

regis65, 2020-06-08:
Vu que j ai commandé en plein confinement,c est normal .
Merci mais je vais attendre sa va pas tardé.
Sportivement
Régis

regis65, 2020-06-15:
Salut
J ai reçu les clés ANT+ de la marque Anself aujourd'hui quand je lance
explorerAnt sa me marque des erreurs (voir photo ci-joint) mais je lance l
aplication sa fonctionne mais sa se decalibre peux tu m aidé
Merci
Regis

WouterJD, 2020-06-15:
1. Vous n'avez pas besoin de ExplorANT; FortiusANT devrais sufficer.
2. les erreurs je ne vois pas; phot's ne sont pas transferé quand on repond par email (au site github seulement)

Salutr en attendant ta reponse

regis65, 2023-01-26:
Salut,
En ce moment mon logiciel Fortiusant délire je m ensert sur kinomap et le
résistance des dénivelé fonctionne plus je voudrais connaître le lien du
dernier Fortiusant car le mien c est le 6.2 et je n arrive pas a trouvé le
6.5
Sportivement
Regis

MeanHat, 2023-01-26:
Salut @regis65 - when I have had similar patterns of uneven resistance it has been caused by the magnet or sensor connected to the Tacx motor/brake unit. No doubt you have already tried this but I thought I would mention it just in case as it is simple to fix.

WouterJD, 2023-01-26:
==

Salut @ regis65 - lorsque j'ai eu des schémas similaires de résistance inégale, cela a été causé par l'aimant ou le capteur connecté à l'unité moteur/frein Tacx.  Vous avez sans doute déjà essayé cela, mais j'ai pensé que je le mentionnerais au cas où car il est simple à réparer.

regis65, 2023-01-27:
[image]
Salut,Je t envoie la photo j ai fais ce que tu m as il c est toujours pareil

regis65, 2023-01-28:
Salut,
J ai téléchargé le fortius ant  6. 64 sur Githu c est toujours pareil j
arrive pas a trouvé la panne pour simulé les dénivelé.
Je t envoi la photo.
Sportivement Regis