WouterJD/FortiusANT #85: Information about tacx  fortius antir

ringhio66, 2020-05-12:
Hello everybody I'm interested  at this project. 
My question  is :for use fortius ANT  with pc windows 8 and tablet window 10 with built in Bluetooth  i need   1 or 2 a bluetooth   dongle transmitter like SUUNTO?? 😳
Thank you very much Roby

WouterJD, 2020-05-12:
Hi Roby

I assume Windows 8 will work like Windows XP, Windows 7 and Windows 10; windows 8 was not mentioned before.

I assume you want to run FortiusANT on the windows 7 PC, using one dongle and connecting the Fortius by USB to that same PC

Then run Zwift on the tablet, using the second dongle to communicate with the Windows PC/FortiusANT.

Question is whether you can put an ANTdongle in the tablet, which would be a test for you to complete: put dongle in tablet, start Zwift and check whether Zwift recognizes the dongle.

* I assume you want to use Zwift, could be Trainer Road or Rouvy, etc as well.

PS. I always like to know where users come from. I am from the Netherlands, where are you located?

ringhio66, 2020-05-12:
Thank you  WouterJD for answer. I'm from Italy Varese.
I thinking  to use the built in Bluetooth  of tablet  and a suunto dongle connect to pc window 8 with fortius USB. Zwift on tablet  see USB  ANT  on Pc  but don't found  fortius sensor. I'm Sorry for my bad English.

WouterJD, 2020-05-12:
Ciao Roby

Non e un problema, chatteriamo in Italiano.

Io spero che il wiki spiaga bene, ma c'è da migliorare.

1- Lancia FortiusANT, passi per il calibrazione cose il trainer sia pronto
2- Lancia Zwift, dovrebbe trovare FortiusANT

Auguri!

ringhio66, 2020-05-12:
Grazie WouterJD, per l'italiano, quando apro FortiusAnt  il pulsante " Locate HW" e sempre  attivo, quindi  secondo la guida non riconosce Fortius e se clicco sul pulsante non cambia niente.. Grazie

WouterJD, 2020-05-12:
Aggiungi -d127 alla linea di command.
FortiusANT produce un logfile, attacca quello al prossimo message, grazie.

ringhio66, 2020-05-12:
Ciao usando il file eseguibile di windows non riesco ad aggiungere, - d127 alla  linea di comando be non riesco a tenere aperta la finestra nera Dos per leggere i messaggi grazie!

ringhio66, 2020-05-12:
Ciao WouterJD ho scaricato il log di FortiusANT ma come faccio a inviartelo? grazie

WouterJD, 2020-05-13:
Sotto il testo comment che scrivo adesso (pagina internet) vedo "Attach files by dragging...."; li si puo drag/drop il logfile.

ringhio66, 2020-05-13:
[FortiusANT.2020-05-12 22-45-23.log](https://github.com/WouterJD/FortiusANT/files/4620199/FortiusANT.2020-05-12.22-45-23.log)

ringhio66, 2020-05-13:
Grazie

WouterJD, 2020-05-13:
22:45:23,626: GetDongle - Set configuration
22:45:23,626: GetDongle - Send reset string to dongle
22:45:23,626: GetDongle - Exception: [Errno None] b'libusb0-dll:err [claim_interface] could not claim interface 0, win error: **La risorsa richiesta \xe8 in uso**.\r\n'
22:45:23,627: GetDongle - Check for dongle 4105 Garmin
22:45:23,635: GetDongle() returns: No (free) ANT-dongle found
22:45:23,635: Dongle - No (free) ANT-dongle found

C`e un altro programma che usa il ANTdongle? Zwift, Garmin Express, Rouvy?

ringhio66, 2020-05-13:
Si forse potrebbe essere il programma  di comunicazione del dongle  scaricato dal sito del produttore stasera lo disinstallo e provo. Poi ti farò sapere grazie e buona giornata

WouterJD, 2020-05-13:
Auguri!

ringhio66, 2020-05-13:
Bene WouterJD ho eliminato il software del costruttore del dongle  e FortiusANT sembra funzionare ma purtroppo  zwift non vede i sensori forse ci vuole un altro dongle?? Ti allego il log

WouterJD, 2020-05-14:
Quando utilizzi lo stesso PC per FortiusANT e Zwift servano due dongle.

ringhio66, 2020-05-15:
Ciao WouterJD ho installato  Rouvy  su cellulare Android 9 Huawei mate 10 pro con bluetooth 4.2 ma anche Rouvy non trova i sensori. Forse sbaglio la procedura?? 1- lancio FortiusANT SUL PC  e appare messaggio start calibrazione poi  sul cellulare lancio Rouvy  che cerca i sensori via Bluetooth  ma non li trova. Ho letto anche in altri forum e hai ragione tu basterebbe solo un dongle.. forse mi manca qualche drive da installare??
Grazie

WouterJD, 2020-05-15:
PC con FortiusANT = ANT
cellulare Android 9 Huawei mate 10 pro con bluetooth 4.2 = BLUETOOTH

To consiglierei di comprare un secondo ANT-dongle e girare Rouvy sul stesso PC

ringhio66, 2020-05-15:
Grazie mille l'ordinato   quando arriva ti farò sapere

WouterJD, 2020-05-15:
Auguri!

WouterJD, 2020-05-19:
[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020) and starting the executable should be easier for windows users.
Of course of there are any issues, let me know.

ringhio66, 2020-06-02:
Ciao WouterJD,
è arrivato il secondo dongle .
Non ho potuto Testare  Ziwfit  perchè  i miei  pc e tablet sono a 32 bit e ziwift no li supporta più.
Rouvy AR invece sembra non supportare le interfaccia ANT+   almeno per ora.
Ho provato KINOMAP   per Android e questo funziona   vengono visti tutti i sensori disponibili nel fortius compresa la fascia sensore cardiofrequenzimetro  polar; quindi posso utilizzare questo programma.
Ho notato poi che durante la calibrazione la velocita con FortiuntAnt+   indicata dal computer della bici è 21 km/h  mentre facendola con il  software Tacx  TTS2  la velocità del computer della bici  risulta essere  20 km/h .
Quindi ti chiedo  come  è possibile  modificare in FortiusAnt+   qualche parametro come la circonferenza della ruota.
Grazie

WouterJD, 2020-06-02:
Ciao @ringhio66 

Leggi #52 per Rouvy/ANT

La calibrazione Tacx Fortius funziona cosi: il software dice il motore di andare a 20km/ora, poi il systema si calibra fino che la resistenza e stabile e finice.

La calculazione velocitá e fatta dalle routine Wheel2Speed() and Speed2Wheel() in usbTrainer.py con la constanta SpeedScale=301. 286.86 dovrebbe resultare in 20km/hr. No credo che sia tanto importante, ma di sicuro e piu bello quando sia preciso. No é paramiterizzabile.

Sono contento capire che adesso il systema funzioni per te.

WouterJD, 2020-06-08:
No follow-up in a week.
If issue needs to be reopened, feel free to do so.

closed 2020-06-08 by WouterJD