WouterJD/FortiusANT #86: Failing to start FortinusANT

JuanRMarques, 2020-05-14:
Hi WouterJD,

This is Mon269 from UK.  I have an I-Magic running with the T1902 headset on a Windows Vista 32bit PC.  I found FortiusAnt when looking for a tool that would broadcast the power from the trainer so that my Garmin watch can pick it up and include this info in my workout files.  FortiusAnt looks like the ticket for what I want to do.  Unfortunately, it fails to run for me.  I don't have Python so I went the Windows executable route, downloaded FortiusAnt.exe and run it from the command line.

C:\Program Files\FortiusAnt\FortiusAnt.exe

gives me a dialog screen with title "FortiusAnt.exe - Entry Point Not Found", and the box has the text "The procedure entry point GetActiveProcessorCount could not be located in the dynamic link library KERNEL32.dll".

Any pointers on what the problem may be?

I'm really hoping I can get this running as I really need power for Garmin to give me the FastBeats data.

Much appreciated.
Mon269

WouterJD, 2020-05-14:
Hi Juan,

I am Wouter from the Netherlands; welcom on the FortiusANT team.

If I recall correctly, Vista was the OS the world would have liked to skip. That does not help you - you would like to use your system to connect to the Tacx.

A quick google searchs directly gives a [hint](https://bugs.python.org/issue38704) and without further investigation that does not sound good. It's likely that the problem will also occur for the python version.

I know that FortiusANT works under Windows XP or Windows 7.

So ... I'm sorry to say ... with this configuration it's unlikely that it will get to work.
How is your hardware, any chance to upgrade to Windows 10? (You could go for Windows XP or 7 - but they are all outdated).

Success - if I can help you just pop a message

JuanRMarques, 2020-05-15:
Hi Wouter,

Good morning and thanks for the very quick response.  I did have a quick search on the net after I logged my message and came to the same conclusion as you.  With Vista I won't be able to get it working.  That is a real shame.

The HW will run Windows 10, but this is a machine that I only use to run my Trainer and play videos whilst on the bike.  I don't fancy the idea of paying close to £200 for the Windows 10 license for this use case.  In fact, for another £200 or even less I can have a proper power meter and then I don't need to broadcast the power from Trainerroad.  

Regards,
Juan

mattipee, 2020-05-15:
@JuanRMarques You might be able to run a virtualmachine with VirtualBox or similar, and given cost of Windows, probably run Linux inside it, and pass the USB through to it, and make it run that way. It's convoluted, I don't know your background, just thinking out loud and tying to conceive of a free technical solution.

WouterJD, 2020-05-15:
Yeah. Installing Windows XP ... if you have the installers. But it's old...
A legal windows 10 installer should not be that expensive, search for a **legal** internet provider who sells licence keys.

JuanRMarques, 2020-05-15:
Both are good suggestions, guys.  Thanks.  I will have a look.

WouterJD, 2020-05-19:
How are  things going?

WouterJD, 2020-05-24:
If there is any new info, please reopen issue.

closed 2020-05-24 by WouterJD