WouterJD/FortiusANT #87: Edge 820 paired

BikeBeppe64, 2020-05-17:
Hi WD, I have a Tacx I-Flow T2250, (1932 & 1901) 2 USBs -  Antifier & Garmin  and an Edge 820.

After running FortiusAnt  and then turning on the Garmin, I trayed searching for something and It paired to the roller, identifying it as in the attached photo.  after heating the rear tire, it started showing power values ​​lower than the strain felt by my legs. 
Inside  Garmin I was able to increase the effort of the roller but it immediantly sent the application off.  

Also Lower speed but No hearbeat & cadence showed. 

Maybe I have been lucky and in the next days I will get further tests  in order to obtain uselful datas. 
Please let me have  commands for Windows (10 x64). 

[image]

WouterJD, 2020-05-18:
Hi @BikeBeppe64. Interesting experiment, see also #79.

I understand you want to use your Edgo to display power/cadence/heartrate and speed; is that correct?

Note that FortiusANT broadcasts two devices on ANT: FE-C (Controllable Fitness Equipment = Tacx Trainer) and HRM (heartrate monitor).

Do you use a HRM connected to your trainer? Only in that case FortiusANT can broadcast a heartrate?

What does FortiusANT display on the graphical user interface?

In other words: tell me some more please and we'll see.

BikeBeppe64, 2020-05-18:
New Test
- Edge 800 (dark display) paired Power Meter, Speed Sensore, Hrm belt
- Edge 820 (clear display) paired tacx 1932, Speed Sensore, Hrm belt
PC: win 10 x64 FortiusAnt 2.6 (win)

Note: PC & Edge 820  real close values
(Power - Speed)  Hrm beat High value
Take a look to my picture Just to have datas

My intention Is to use my roller as It is to follow my train on edges or Watching TV during my workout

[image]

WouterJD, 2020-05-18:
The last case seems as if you pressed Ctrl+C to interrupt the program. What happened?

Funny idea. What program do you use: Trainer Road, Zwift, Rouvy, FulGaz???
What is the advantage of following training on edge instead of the FortiusAnt GUI (other than that it's fun to do)?

BikeBeppe64, 2020-05-18:
> The last case seems as if you pressed Ctrl+C to interrupt the program. What happened?
I don't Remember exactly. 
Any help ti close in the right way is appreciate.
> 
> Funny idea. What program do you use: Trainer Road, Zwift, Rouvy, FulGaz???
- I'm trying (trial) all those for different situation in the screen , (real or virtual ?) But at the end of story I'll chiose GC but at the moment I don't know how to connect T1932 to it having slope & pwr

> What is the advantage of following training on edge instead of the FortiusAnt GUI (other than that it's fun to do)?
How to connect my old 1932 to other programs in order to have Pwr and slope in the raining days or small time biking ?

WouterJD, 2020-05-18:
I do not know GC myself, but during your first tests you can use -m (manual power) or -M (manual grade) to set slope and power. After that: Zwift, Rouvy, ...

So for the technical experiment:
- FortiusANT -g -a -n -M -H0
- Connect Edge to check power/speed/cadence from FortiusANT
- Start cycling and add/reduce grade with headunit buttons UP/DOWN

Succes!

BikeBeppe64, 2020-05-18:
> I do not know GC myself, but during your first tests you can use -m (manual power) or -M (manual grade) to set slope and power. After that: Zwift, Rouvy, ...
> 
> So for the technical experiment:

> * FortiusANT -g -a -n -M -H0
> * Connect Edge to check power/speed/cadence from FortiusANT
> * Start cycling and add/reduce grade with headunit buttons UP/DOWN
> 
> Succes!

> I do not know GC myself, but during your first tests you can use -m (manual power) or -M (manual grade) to set slope and power. After that: Zwift, Rouvy, ...
> 
> So for the technical experiment:
> 
> * FortiusANT -g -a -n -M -H0

> * Connect Edge to check power/speed/cadence from FortiusANT
> * Start cycling and add/reduce grade with headunit buttons UP/DOWN
> 
> Succes!

> I do not know GC myself, but during your first tests you can use -m (manual power) or -M (manual grade) to set slope and power. After that: Zwift, Rouvy, ...
> 
> So for the technical experiment:
> 
> * FortiusANT -g -a -n -M -H0

(1) no LocateHwd ?
(2) no RunOff ? 40kn/h 
(3) no -d127  (logfile) ?

> * Connect Edge to check power/speed/cadence from FortiusANT

(4) I can connect/pair  my Edge 820 Just pressing "Start" button

> * Start cycling and add/reduce grade with headunit buttons UP/DOWN
> 
> Succes!

Are you interesting on my two Edge files in order to compare values with log file ? 
How long does It have to be and on which format ?

As soon as possible  another  2cent

WouterJD, 2020-05-18:
(1) (2)
-a means: autostart so you proceed immediatly into operational mode.
If you think that doing Runoff is better then do not specify -a.

(3) -d127 is only in case you have problems, so I can check the logfile to analyse your problem.
So if you have no problem, no need to make a logfile.

Auguri con le vostre prove!
Avete provato usare -m o -M per attivare il modo manuale (-m = manual power, -M = manual grade)?

BikeBeppe64, 2020-05-18:
> (1) (2)
> -a means: autostart so you proceed immediatly into operational mode.

Ok
> If you think that doing Runoff is better then do not specify -a.

Ok
> 
> (3) -d127 is only in case you have problems, so I can check the logfile to analyse your problem.
> So if you have no problem, no need to make a logfile.
> 
Ok

> Auguri con le vostre prove!
Big thanks for your prompt replies and help

> Avete provato usare -m o -M per attivare il modo manuale (-m = manual power, -M = manual grade)?

Not yey

BikeBeppe64, 2020-05-19:
> So for the technical experiment:
> 

> * FortiusANT -g -a -n -M -H0
> * Connect Edge to check power/speed/cadence from FortiusANT

[image]

- No paired Edge 820, no Cadence, with above commands, so decided to use 
Gpwm (Garmin Pwr Mt) and take a look at FApwr (FortiusAnt) to watch values.

> * Start cycling and add/reduce grade with headunit buttons UP/DOWN
> 
I had been able just to use Cancel button to exit the program, for other buttons no results

> Succes!

Paired T1932 to 820 Just exit and enter again in FortiusAnt -g -n and Press "Start" 

My 2cents :
- From Speed 12-14 km/h Gpwm 112w  FApwr 100w +/-
At Speed 30 km/h 300w  FApwr 200 +/-

Close program CNTR + F4 had same black screen 

[image]

Right close way ?

WouterJD, 2020-05-19:
You can stop FortiusANT with ALT+F$ or press the X at the right-top.
An exit-button would have been nicer, sorry that I did not make that.

WouterJD, 2020-05-19:
[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020); please download newest version since error-handling is improved in many places.
Of course of there are any issues, let me know.

BikeBeppe64, 2020-05-19:
> [Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020); please download newest version since error-handling is improved in many places.
> Of course of there are any issues, let me know.

Thanks in advance , we do appreciate your big ed exausting job.  I'll keep you informed. Tomorrow new workout

BikeBeppe64, 2020-05-22:
Hi WJD & All ,
Got really small test with FTAnt 3.0 It seem better then before, hartbeat showed immediatly, GC reconized and paired trainer T1932 on set up/configuration, as model as in the Edge 820. (First picture)
FTant+ Power force/values is lower than Power Meter in my bike

[image]

WouterJD, 2020-05-22:
if the relation between the "real" and transmitted power is (almost) constant, you can use the -p commandline parameter. Default is one. Do some tests...

WouterJD, 2020-05-24:
Hi BikeBeppe. You are working with Edgo820. Does the Edge800 also see FortiusANT?

BikeBeppe64, 2020-05-26:
Ho WouterJD, tried this mornig to pair FortiunsAnt & Trainer by Edge 800 but no success

[image]

In the Big One (800) Just Power Meter paired

WouterJD, 2020-05-26:
Hi Beppe, thanks.

I do not see size-difference: the top one is 800 and the bottom one 820?

BikeBeppe64, 2020-05-26:
> Hi Beppe, thanks.
> 
> I do not see size-difference: the top one is 800 and the bottom one 820?

The above one (longer) Is the Edge 800 where I can't get the trainer Tacx nr  57591  as paired as in the small one 820

WouterJD, 2020-05-26:
Thanks. I will create a PWR profile for the Edge 800 as asked by #79.
Ciao e a presto!

PS. Is there something I can help you with?

BikeBeppe64, 2020-05-26:
> Thanks. I will create a PWR profile for the Edge 800 as asked by #79.
> Ciao e a presto!
> 
> PS. Is there something I can help you with?

Thanks in advance for #79 and what can i do for helping you ?

WouterJD, 2020-05-26:
Ha! I have been stupid!

When you run in manual mode, commands do not come from Zwift/GC/Rouvy but from the headunit-buttons, so I disconnected ANT at all.

I did not consider the possibility that somebody would display the results from FortiusANT on an Edge and command from the headunit. So that's why (when using -m or -M) FortiusANT is not found by your edge.

I ran into this aspect when building the iVortex version; as soon as that version is released - ANT will remain active in manual mode. 

Thanks for joining in on #79.
Are there any open questions from your side?

BikeBeppe64, 2020-06-04:
Oggi, non riuscendo a connettere Golden Cheetah al trainer, ho eseguito il workout  direttamente dal Garmin Edge 820, dovevo fare 4 volte 10 x  40" a 220/250 e 20" a 60/120w, recupero 5' a 60/120w, notavo che nei cambi di tempo la resistenza aumentava e diminuiva automaticamente, e questo non lo avevo mai notato.

Comunque i watt erano differenti tra il powermeter e il T1932.

Cadenza ancora assente nel fortiusAnt

WouterJD, 2020-06-04:
Usando `FortiusANT -d127` alla linea di commando, un logfile é creato; to potresti allegare questo logfile, cosí posso vedere se la dadenza sia ricevuto dal T1932 e trasmesso al Edge.

La differenza tra Fortius e powermeter non dovrebbe essere troppo grande.
In caso di si, fai un test, sia utilizzando `FortiusANT -m` o Edge 820.

Scrivi cosa dice FortiusANT, Edge e Powermeter:
30 secondi per test sarebbe sufficiente; basta che il sistema sia stabile (invece di stancarsi troppo).

| test | target | FortiusANT TargetPower | FortiusANT CurrentPower | Edge CurrentPower  |
| --- | --- | --- | --- | --- |
| 1 | 50 W | ... | ... | ... |
| 1 | 100 W | ... | ... | ... |
| 1 | 150 W | ... | ... | ... |
| 1 | 200 W | ... | ... | ... |
| 1 | 250 W | ... | ... | ... |

WouterJD, 2020-06-05:
It's a little bit short.

I would like to know:
- the target you set
- the target that is displayed by FortiusANT
- the actual value that is displayed by FortiusANT
- the actual value of the power meter

Add the comment in PC browser; you can attach logfile in box under the text where you type

BikeBeppe64, 2020-06-07:
Hi WJD
First of all it was not possible to pair T1932 without GUI  command.

I did a test twice, First following watt by Power Meter (Edge 800) and then by FortiusAnt+ (Edge 820) , and got these results: 

[image]

NP stay for Normalized Power

Hoping It Is usefull , waiting for your reply

WouterJD, 2020-06-07:
Wauw that's serious work. Will check later!
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

BikeBeppe64, 2020-06-09:
I got a new one, professional of course, with less steps & longer time Just in case 

[image]

Let me know how to send you a bere ...

WouterJD, 2020-06-11:
Hi @BikeBeppe64 please check [pull request](https://github.com/WouterJD/FortiusANT/pull/95)
Speed and Cadence Sensor is implemented, PWR profile is next step

WouterJD, 2020-06-15:
> Hi WJD
> First of all it was not possible to pair T1932 without GUI command.
> 
> I did a test twice, First following watt by Power Meter (Edge 800) and then by FortiusAnt+ (Edge 820) , and got these results:
> 
> [image]

I did not forget this test!
No what is the conclusion to be drawn
- Speed/speed shows that FortiusANT speed is a bit too low (-8%)
- Med/Med is the ratio between the power measurements, but I do not know whether this is what you wanted to say

In all cases, please realize: A Fortius / i-Magic / i-Vortex is not as accurate and/or advanced as a Tacx Neo. Converting USB to ANT+ does not change that. The Tacx Neo is many years development ahead AND is a direct drive system, with it's related price.

If there is something I can do for you, please let me know.

||Edge 800||Power Meter|||Edge 820||Fortius ANT||med / med|speed / speed|
|---|---|---|---|---|---|---|---|---|---|---|---|
|00:00|Watt||||km/hr|||||67%|92%|
|Time|||med|max|Speed|med|max|np|speed|||
|03:00|50|80|47|61|10,3|65|82|65|9,43|138%|92%|
|00:30|47|55|56|76|11,3|34|62|52|10,6|61%|94%|
|00:30|97|105|165|208|22,2|65|88|52|20,5|39%|92%|
|00:30|145|155|172|255|18,1|158|182|122|16,76|92%|93%|
|00:30|195|210|170|217|17,9|180|201|165|16,73|106%|93%|
|00:30|245|255|243|344|21,1|217|241|204|20,12|89%|95%|
|00:30|295|305|378|446|27,5|276|294|249|25,21|73%|92%|
|05:00|40|110|90|298|14,5|54|229|113|13,61|60%|94%|
|---|---|---|---|---|---|---|---|---|---|---|---|
|03:00|50|80|133|197|19,1|43|46|43|17,61|32%|92%|
|00:30|47|55|135|167|18,1|42|45|43|17,5|31%|97%|
|00:30|97|105|223|291|28,6|61|81|53|24,7|27%|86%|
|00:30|145|155|257|296|28,8|124|139|97|26,6|48%|92%|
|00:30|195|210|235|267|20,7|202|239|171|19,2|86%|93%|
|00:30|245|255|272|312|22,2|223|238|163|20,5|82%|92%|
|00:30|295|305|307|357|24,4|244|265|230|20,2|79%|83%|
|05:00|40|110|122|237|17,3|42|137|101|16,3|34%|94%|

BikeBeppe64, 2020-06-17:
@WouterJD numbers are numbers, you're right

BikeBeppe64, 2020-06-17:
Paired by New Edge 820

[image]

[image]

[image]

In order to have same/similar watt (+/- 5w) for trainer and Power Meter , I fixed my Gear in 34 (front)  - 30 (rear) and Cadence 80 rpm.

I got problem when pairing by old Edge 800, because It connected various ant+ but during the workout those elements confused the old Edge going in and out quickly 

Tomorrow new tests

WouterJD, 2020-06-17:
Download the newest version 3.2.1; two crashes resolved

WouterJD, 2020-06-17:
> @WouterJD numbers are numbers, you're right

Yep - Most important is that the feeling is OK.
My ride with Rouvy yesterday was a bit bumpy; to ofast switching between +5% and -3%. For the rest it worked.
Ciao!

BikeBeppe64, 2020-06-19:
Ciao Wouter, con l'ultima versione 3.2.1 tutto va benissimo. Meriti una grossa Birra quando vieni in Italia. keep in touch!!

BikeBeppe64, 2020-06-19:
Puoi chiudere.....

BikeBeppe64, 2020-06-19:
Please close , everything work perfectly

WouterJD, 2020-06-19:
> Meriti una grossa Birra quando vieni in Italia. keep in touch!!

Grazie per la risposta; speriamo di incontrarci!

closed 2020-06-19 by WouterJD