WouterJD/FortiusANT #88: Mathematical solution to the Fortius low-speed-high-power limitation

mattipee, 2020-05-18:
The Fortius does not appear to handle high-power at low wheel speeds.

The symptom is that if you send a large target resistance to the Fortius, it result in motor-stuttering when the wheel speed is low.

There are a couple of consequence to this:
- in grade mode, simulating steep gradients using a straightforward combination of Grade2Power() plus Power2Resistance() can result in too high a target resistance value at low speed
- in power mode, setting a high power target and setting off from zero results in too high a target resistance value at low speed

Wouter has an approach to limit resistance at low speeds according to a formula which implements a sliding scale between resistance values of 1500 and 6000 for wheel speeds between 0 and 15khm.

```
 if self.SpeedKmh <= 15 and Resistance >= 6000:
            Resistance = int(max(1500, self.SpeedKmh / 15 * 6000))
```

[image]

There's also a slightly different approach to avoiding wheel stuttering for grade mode whereby we implement virtual gearing, to make virtual speed (and therefore targetpower and resistance) disconnected from actual wheel speed - "make the wheel spin faster and pretend it's actually running more slowly". However, it requires manual headunit button pushes and can still make the motor stutter, without the additional <15kmh trick above, or something like it.

I'm trying to work out how one might combine the two with one mathematical solution.

Previous attempts at thinking this through have taken place on the following issues:
- https://github.com/WouterJD/FortiusANT/pull/84
- https://github.com/WouterJD/FortiusANT/issues/54
- https://github.com/WouterJD/FortiusANT/issues/51

Raising this as a separate issue because my thinking and posting on this subject on other threads are somewhat long and unhelpful. Apologies.

WouterJD, 2020-05-18:
Practical test done today:
R=6000 WheelSpeed=10km/hr ==> motor brake stutters
R=4500 WheelSpeed=10km/hr ==> motor brake runs smoothly
I have set the cutoff point down to 10km/hr and R=1500 + SpeedKmh * 300.

Conclusion: real-life test to do highpower/lowspeed, like MTB-ride against a 20% slope cannot be done with ANT Fortius.

Alternative suggestions welcome.

WouterJD, 2020-05-19:
PS. The red line has it's own challenges since if you get below the calibration value the engine will start to spin instead of brake, something you do not expect probable.

[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020) with CoD as described.
Of course of there are any issues, let me know.

closed 2020-05-19 by WouterJD